mdhiggins/ESP8266-HTTP-IR-Blaster #187: Node MCU Goes down -stops responding after 3 -4 min .

sujitrp, 2018-06-27:
Hi Project is really very rich and simple to use . I am using nodeMCU v3 to replicate the project .

Need help  to overcome following 

- NodeMCU goes down after 4- 5 min (240 - 250 pings) and does not come up till i press hard reset on board . Post up same cycle .

- S3 - Grnd wifi setting reset not working as a result 

- as result on every boot it is considering the last config file . mot allowing me to update any of the paramter like amazon account etc

[image]

- 

[image]

serial monitor log ;

ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"IRSERVER","passcode":"","port_str":"80","user_id":"mygamil@gmail.com","ip":"192.168.1.24","gw":"192.168.1.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.24
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.24
WiFi connected! User chose hostname 'IRSERVER' passcode '' and port '80'
WiFi configuration complete
Local IP: 192.168.1.24
URL to send commands: http://IRSERVER.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to IRSERVER.local:80
HTTP Server started on port 80
Turning off the LED to save power.
Error retrieving external IP
HTTP Code: -1
connection refused
External IP address request took 5020 ms
Warning, user_id appears to be in the wrong format, security check will most likely fail. Should start with amzn1.account.***
Epoch time from timeServer is unexpectedly old, probably failed connection to the time server. Check your network settings
14400
Invalid EPOCH time, security checks may fail if unable to sync with NTP server
Ready to send and receive IR signals
Connection received
External IP: <my external Ip>
External IP address request took 342 ms
Connection received
Connection received
Connection received
Connection received
Connection received
Connection received
Connection received
Connection received

Ping Stat.

Ping statistics for 192.168.1.24:
    Packets: Sent = 277, Received = 241, Lost = 36 (12% loss),
Approximate round trip times in milli-seconds:
    Minimum = 3ms, Maximum = 140ms, Average = 7ms

Reply from 192.168.1.24: bytes=32 time=9ms TTL=128
Reply from 192.168.1.24: bytes=32 time=6ms TTL=128
Reply from 192.168.1.24: bytes=32 time=12ms TTL=128
Reply from 192.168.1.24: bytes=32 time=10ms TTL=128
Reply from 192.168.1.24: bytes=32 time=4ms TTL=128
Reply from 192.168.1.24: bytes=32 time=6ms TTL=128
Reply from 192.168.1.24: bytes=32 time=6ms TTL=128
Reply from 192.168.1.24: bytes=32 time=4ms TTL=128
Reply from 192.168.1.24: bytes=32 time=4ms TTL=128
Reply from 192.168.1.24: bytes=32 time=11ms TTL=128
Reply from 192.168.1.24: bytes=32 time=6ms TTL=128
Reply from 192.168.1.24: bytes=32 time=3ms TTL=128
Reply from 192.168.1.24: bytes=32 time=8ms TTL=128
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 192.168.1.24:
    Packets: Sent = 277, Received = 241, Lost = 36 (12% loss),
Approximate round trip times in milli-seconds:
    Minimum = 3ms, Maximum = 140ms, Average = 7ms
Control-C

teohell, 2018-06-30:
Same problem here with WeMos D1 mini

using ESP8266 2.4.0 and lwIP 1.4 prebuilt, but the micro just hangs after 4-5 seconds after reset or powerup. It also stops responding the ping, as it was loosing its IP address. It shows without IP address in the router page in fact.

Tried with different combinations of core/lwIP but without success. Downgrading everything to a previous version of the project (downgrading also libraries and core), it is working fine so the micro is ok.

when I get to load the webpage, it gives two errors, EPOC time too low and unable to fetch external IP address.

bhattvishal, 2018-07-01:
I am also facing the same issue, but not sure how to get this resolved. If anyone found the solution, please let me know.

Niflou, 2018-07-01:
Same behaviour here. I started this project yesterday and I experience the same issue.

I'm using :
- NodeMCU v3
- Arduino IDE 1.8.5
- ESP8266 Arduino Core 2.4.1
- lwIP "1.4 compile from source"
- ESP8266WIFI 1.0.0
- ESP8266Webserver 1.0.0
- EasyNTP : 1.1.0
- IRRemoteESP8266 2.4.2
- CryptosuiteMaster : last git commit
- ArduinoJSON 5.13.2

I'm not experimented in arduino so I'm really unable to debug this behaviour :(

@teohell can you provide more informations when you said "Downgrading everything to a previous version" please ? I'm really interessted in getting this working.

Niflou, 2018-07-01:
Ok thanks for the information.

On my side, I disabled the external IP, NTP and MDNS and it's working (I mean, not crashing because I didn't tried all the IR part yet) for 20 minutes now. But maybe I'm just in a lucky run, I'll tell you if the device remains stable.

```
const bool getExternalIP = false;
const bool getTime = false;
const bool enableMDNSServices = false;
```

mdhiggins, 2018-07-01:
That will disable all the security features this is likely being caused by a memory leak in one of the dependency updates, working on it

mdhiggins, 2018-07-01:
Unable to reproduce, how long does it take to freeze for you guys?

I'm using NodeMCU V2

Here's the version of everything I'm using:
- NodeMCU v2
- Arduino IDE 1.8.5
- ESP8266 Arduino Core **2.4.0** (@DB-CL try downgrading)
- lwIP v1.4 Prebuilt
- EasyNTP : 1.1.0
- IRRemoteESP8266 2.4.2
- CryptosuiteMaster : latest (8d691213f23b87fa2231f2f00285799c5b9f6546)
- ArduinoJSON 5.13.2 (beta 6 changes things significantly and will require rewriting, don't use, won't compile)
- WiFiManager 0.12.0
- ArduinoOTA 1.0.0

My advice would be to sync these exactly
From what I know, lwIP 2.0 has a memory leak and 2.4.1 has a memory leak, using either of these will create the exact problem you guys are experiencing

The other possibility is that when using different hardware you'll need to adjust the pins for transmission, receiving, and reset. 

@DB-CL I would advise against what you're doing, disabling these services reduces functionality and is very insecure and is just masking the network memory leak because you're using the network less, it will probably eventually freeze. Try and downgrade to 2.4.0 and use lwIP 1.4 prebuilt (both of these things you weren't doing according to your post

@sujitrp if you can't reset using whatever the default pin is (again this was designed for the NodeMCU v2) just change to a different configuration pin that you can set to ground

A sign that there are networking troubles is that its unable to retrieve the external IP address. If that's the case then there are either bad libraries when compiling or bad networking settings when you've configured

**If you are still having trouble please post all of the above information I have in the outline including your versions, your hardware that you're using, and your compile settings (the string at the bottom right of the Arduino IDE). Additionally post your configuration settings including your local IP address, gateway, subnet, and your routers IP address (local address). Do NOT post your amazon user-id, your password, or your external IP address for security reasons**

DB-CL, 2018-07-01:
Hi @mdhiggins, 
Thanks for your update, I'll try the exact same versions you use and tell you if it works.
I have no interest in security (I'm behind nginx and managing security with it) but if you say I'm just delaying the leak, that's not good for me and I will try to get it to work with the services enabled

DB-CL, 2018-07-01:
I have a good stability with the versions like you. Thanks !!

mdhiggins, 2018-07-01:
Glad to hear it, waiting on @sujitrp to close

teohell, 2018-07-02:
@mdhiggins 
I Michael, I checked all libraries versions, I compiled with the same as you but still got the same problems
here's details

-     **Wemos D1 mini** (cna't figure out the version, maybe 2016?)
-     Arduino IDE 1.8.5
-     ESP8266 Arduino Core 2.4.0
-     lwIP v1.4 Prebuilt
-     EasyNTP : 1.1.0
-     IRRemoteESP8266 2.4.2
-     CryptosuiteMaster : 8d691213f23b87fa2231f2f00285799c5b9f6546
-     ArduinoJSON 5.13.2
-     WiFiManager 0.12.0
-     ArduinoOTA 1.0.0

here's the only changes I made to the code:
```
// WEMOS users may need to adjust pins for compatability
//const int pinr1 = 14;                                         // Receiving pin
//const int pins1 = 4;                                          // Transmitting preset 1
const int pins2 = 5;                                          // Transmitting preset 2
const int pins3 = 12;                                         // Transmitting preset 3
const int pins4 = 13;                                         // Transmitting preset 4
//const int configpin = 10;                                     // Reset Pin
#define pinr1 D5
#define pins1 D2
#define configpin D7
```

here's compile settings
`Wemos D1R2 & mini, 80 MHz, 4M (1M SPIFFS), v1.4 Prebuilt, Disabled, None, 115200 on COM5`

tested on two different routers, got the same behavior

teohell, 2018-07-02:
this is the output I get on the webpage

``100395ms uptime; EPOCH 14400 / 1530516617 ( 1530502220 )
Error - unable to retrieve external IP address, this may be due to bad network settings. There is currently a bug with the latest versions of ESP8266 for Arduino, please use version 2.4.0 along with lwIP v1.4 Prebuilt to resolve this
Error - EPOCH time is inappropraitely low, likely connection to external time server has failed, check your network settings``

the external IP does not show

the ping loses normally 1-2 packages, the reload of the webpage is extremely low, until a point were it eventually stops to respond completely

mdhiggins, 2018-07-02:
@teohell 
If you run the software on the bare board (no additional connections to transistors, LEDs, etc) does that change anything?

Additionally can you post your networking settings that you setup in WiFiManager as well as post your routers local IP address?

teohell, 2018-07-02:
this is a bareboard; last week I was testing on a fully connected board, and the difference is that it crashed sooner. The bareboard lasts longer but in the end the result is the same

static IP: 192.168.49.150
gateway: 192.168.49.254
SM: 255.255.255.0

mdhiggins, 2018-07-02:
And your routers local IP is 192.168.49.254 ?

I think I have a WeMos D1 lying around somewhere I’ll try and find it and test

teohell, 2018-07-02:
''And your routers local IP is 192.168.49.254 ?''

As funny as it sounds :-)

teohell, 2018-07-02:
Bingo!
The DNS is external, and it wasn't forwarding the request for external ip for some reason!
Bypassed and now it is working like a charm

sujitrp, 2018-07-02:
Sorry Guys was in home shifting for last few days. Let me try this with  Michael lib versions .

mdhiggins, 2018-07-02:
Nice and that has fixed your stability issues? @teohell

teohell, 2018-07-02:
Yep, I also had to set a different ntp in the code because for some reason it seems that the firewall wasn't allowing pool.ntp.org
Testing with bareboard now anyway, will let you know tomorrow with full board

mdhiggins, 2018-07-02:
So did you change your router to forward the DNS or did you just resolve all the domains to static IPs?

teohell, 2018-07-04:
made some test
first of all, by resolving the DNS correctly to let the libraries get the external IP, it's working good also with full board.
I don't know how the resoultion of external IP has changed from last versions, but when flashing with a sept'17 version of the project, it doesn't hang even if it's not able to resolve the external IP, while with up to date libraries and blueprint, it shows stability problems as above.
I also noted that with wemos D1 it's better to comment out the unused ir ouputs, it seems to me that it's working faster

sujitrp, 2018-07-09:
Hi Friends It started working when i flashed using esptools.py 
using IDE I generated the bin file and flashed using esptools.py write_flash 
last 15 -20 Min it is working without break .  Will connect and load tomorrow and update the progress .

Kindly suggest  reference for examples /use cases . I am planning to use openhab http binding.  

esptool.py --port COM5 write_flash 0x00000 C:\OS\arduino-1.8.5-Set2\arduino-1.8.5\portable\sketchbook\ESP8266-HTTP-IR-Blaster\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino.nodemcu.bin