mdhiggins/ESP8266-HTTP-IR-Blaster #191: Do not have Amazon Alexa device

chiql, 2018-07-10:
Hi Mr. Higgins,

I do not have the Alexa device (to understand my voice), is it possible to get the circuit (ESP, Tx, Rx IR) work with a web page/smart phone app to control the TV or the air conditioner?

mdhiggins, 2018-07-10:
Here's an example using a shell script

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/wiki/Shell-Command-Testing

chiql, 2018-07-13:
Thanks a lots, but I cannot get the irblaster.sh script, would you please send me?

mdhiggins, 2018-07-13:
Its in the link I posted

chiql, 2018-07-15:
Thanks, I can use the script to send the command.
However, the IR TX LED did not flash (maybe it flashed with very week light).
I used transistor KSP2222A (2N2222), R1 = 4.7K, R2 = 0.
I then used normal LED to replace IR TX LED for debug, but the normal LED did not flash also.
I think my TX circuit is correct, but in my imagination, the TX LED should flash very strong.

mdhiggins, 2018-07-15:
I would check your circuit its possible something isn't wired up right
Make sure you're using the right data pin as well, are you running off a NodeMCU v2?

Feel free to post a picture of your build

chiql, 2018-07-15:
I think my data pins are similar to yours:
[image]
GPIO14 = D5 = RX:
[image]
GPIO4 = D2 = TX:
[image]

My ESP chip is ESP-12N (written in the chip). I don't know it is NodeMCU v2 or not.
Board name: NodeMCU 1.0 (ESP-12E Module) on IDE.
I use the normal red LED for debugging.
My sending command is: bash -x irblaster.sh EA525C2F:UNKNOWN:100
Below is my screenshot of sending and receiving commands:
[image]

mdhiggins, 2018-07-15:
Ah I see the problem
You can't transmit unknown commands, unknown commands are errors and trying to send them doesn't actually do anything
If the received codes read as unknown you need to send them in the raw format

chiql, 2018-07-16:
Thank you very much @mdhiggins, I tried to transmit the raw data for UNKNOWN command like below, but it not work. Do you have any suggetst?
wget -o - 'http://192.168.1.200:80/msg?plain=[{'data':[3338,1710,370,1308,370,492,346,520,316,468,370,492,372,468,346,466,370,472,370,462,372,522,338,448,392,456,354,1336,392,468,318,494,344,496,368,442,368,468,394,444,368,466,370,470,370,470,368,610,224,474,368,490,344,492,344,468,368,466,372,466,368,470,394,1280,398,444,370,1278,396,1306,368,1306,370,1356,318,1358,342,1282,370,468,370,1336,344,1328,344,1304,396,1280,370,1332,342,1330,370,1282,368,1308,392,1286,368,466,394,444,368,508,332,466,372,468,368,468,396,468,344,472,370,488,344,522,316,1304,370,1332,346,496,366,468,370,1306,344,1386,294,1274,394,1310,392,470,364,442,398,1282,390,1336,342,434,376,528,312,496,342,1332,368,494,318,518,320,1332,344,464,372,468,416,1264,368,1302,370,470,368,1306,368,1304,372,494,368,1280,372,1320,354,496,320,1326,418,1258,370,492,372,468,342,1306,370,468,368,468,370,468,372,462,372,466,396],'type':'raw','khz':38}]&pass=pass&auth=d91af3d72f125ac145ec92fea9f7fb0258c773d5d0a51ab3299534a9605bcb2b&mid=aabbccdd-1234-abcd-5678-aabbccddeeff&epid=1&time=1531753846&simple=1'

mdhiggins, 2018-07-16:
You have to use /json not /msg to send the RAW stuff

```
wget -o - 'http://192.168.1.200:80/json?plain=[{'data':[3338,1710,370,1308,370,492,346,520,316,468,370,492,372,468,346,466,370,472,370,462,372,522,338,448,392,456,354,1336,392,468,318,494,344,496,368,442,368,468,394,444,368,466,370,470,370,470,368,610,224,474,368,490,344,492,344,468,368,466,372,466,368,470,394,1280,398,444,370,1278,396,1306,368,1306,370,1356,318,1358,342,1282,370,468,370,1336,344,1328,344,1304,396,1280,370,1332,342,1330,370,1282,368,1308,392,1286,368,466,394,444,368,508,332,466,372,468,368,468,396,468,344,472,370,488,344,522,316,1304,370,1332,346,496,366,468,370,1306,344,1386,294,1274,394,1310,392,470,364,442,398,1282,390,1336,342,434,376,528,312,496,342,1332,368,494,318,518,320,1332,344,464,372,468,416,1264,368,1302,370,470,368,1306,368,1304,372,494,368,1280,372,1320,354,496,320,1326,418,1258,370,492,372,468,342,1306,370,468,368,468,370,468,372,462,372,466,396],'type':'raw','khz':38}]&pass=pass&auth=d91af3d72f125ac145ec92fea9f7fb0258c773d5d0a51ab3299534a9605bcb2b&mid=aabbccdd-1234-abcd-5678-aabbccddeeff&epid=1&time=1531753846&simple=1'
```

Also the time and auth commands are time sensitive and will not work after about 30 seconds, so you'll need to recreate those

chiql, 2018-07-16:
I replaced /msg by /json
```
wget -o - 'http://192.168.1.200:80/json?plain=[{'data':[3324,1728,374,1304,374,492,346,462,374,466,372,464,372,464,374,464,372,468,374,460,372,464,396,442,372,466,398,1256,414,442,372,468,396,446,370,464,370,464,374,468,394,466,346,464,374,464,370,468,370,494,348,460,372,466,396,438,374,466,370,492,346,464,372,1304,374,468,374,1296,374,1306,368,1328,348,1302,396,1280,372,1302,374,464,372,1284,396,1300,396,1278,372,1302,376,1302,372,1330,346,1278,422,1278,370,1310,372,460,370,468,370,470,368,440,418,442,372,468,374,462,370,470,372,462,372,466,370,1304,372,1302,398,468,344,492,370,1278,376,1306,372,1274,396,1304,372,464,372,466,374,1302,372,1304,370,466,398,442,376,460,374,1302,396,438,374,516,318,1306,370,466,374,462,374,1282,394,1302,398,442,370,1304,372,1304,372,440,398,1306,372,1300,372,446,396,1274,396,1304,374,488,348,466,370,1306,368,492,348,464,372,468,374,462,398,438,376],'type':'raw','khz':38}]&pass=pass&auth=08d29f058a239dcd5e697380541b31e415fc4086eac119cc19528e8c8cdaed96&mid=aabbccdd-1234-abcd-5678-aabbccddeeff&epid=1&time=1531784452&simple=1'
```
Here is the error:
**The name is too long**, 1025 chars total.
Trying to shorten...
```
New name is json?plain=[{data:[3324,1728,374,1304,374,492,346,462,374,466,372,464,372,464,374,464,372,468,374,460,372,464,396,442,372,466,398,1256,414,442,372,468,396,446,370,464,370,464,374,468,394,466,346,464,374,464,370,468,370,494,348,460,372,4.
--2018-07-17 06:41:38--  http://192.168.1.200/json?plain=[%7Bdata:[3324,1728,374,1304,374,492,346,462,374,466,372,464,372,464,374,464,372,468,374,460,372,464,396,442,372,466,398,1256,414,442,372,468,396,446,370,464,370,464,374,468,394,466,346,464,374,464,370,468,370,494,348,460,372,466,396,438,374,466,370,492,346,464,372,1304,374,468,374,1296,374,1306,368,1328,348,1302,396,1280,372,1302,374,464,372,1284,396,1300,396,1278,372,1302,376,1302,372,1330,346,1278,422,1278,370,1310,372,460,370,468,370,470,368,440,418,442,372,468,374,462,370,470,372,462,372,466,370,1304,372,1302,398,468,344,492,370,1278,376,1306,372,1274,396,1304,372,464,372,466,374,1302,372,1304,370,466,398,442,376,460,374,1302,396,438,374,516,318,1306,370,466,374,462,374,1282,394,1302,398,442,370,1304,372,1304,372,440,398,1306,372,1300,372,446,396,1274,396,1304,374,488,348,466,370,1306,368,492,348,464,372,468,374,462,398,438,376],type:raw,khz:38%7D]&pass=pass&auth=08d29f058a239dcd5e697380541b31e415fc4086eac119cc19528e8c8cdaed96&mid=aabbccdd-1234-abcd-5678-aabbccddeeff&epid=1&time=1531784452&simple=1
```
Connecting to 192.168.1.200:80... connected.
HTTP request sent, awaiting response... 401 Unauthorized

Username/Password Authentication Failed.
Is the code too long?

mdhiggins, 2018-07-17:
Honestly I would just disable the security features (you can leave the passcode) by deleting or blanking out the user_id when you do the network settings until you get your commands working

This will allow you to just paste the URLs in the browser and not need to use the shell file, should probably help getting things started

chiql, 2018-07-25:
Thank you for a great project & kindly reply.
[Screenshot of sending some RAW codes with no security features](https://user-images.githubusercontent.com/39230815/43214210-d7f68a06-9062-11e8-81a9-c4c2663475c8.png) shows some codes are sent successfully, the IR LED are flashed. But the device seem unable to receive as it do nothing. It may show that the software does not work for unknown codes, does it?

mdhiggins, 2018-07-25:
What type of device are you trying to control?

The library it uses for IR commands does not work well with various AC units but should be good for most other devices

From: Chuong Le
Sent: Wednesday, July 25, 2018 12:40 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Mention
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Do not have Amazon Alexadevice (#191)

Thank you for a great project & kindly reply.
Screenshot of sending some RAW codes with no security features shows some codes are sent successfully, the IR LED are flashed. But the device seem unable to receive as it do nothing. It may show that the software does not work for unknown codes, does it?

chiql, 2018-07-25:
>What type of device are you trying to control?

The Hitachi AC.

The buffer is increased as the code is so long:
`const unsigned int captureBufSize = 300; // Size of the IR capture buffer.`

>The library it uses for IR commands does not work well with various AC units but should be good for most other devices

I will test with Daikin AC later.

Yes, below libs seem not work with Hitachi AC unit:
<IRremoteESP8266.h>
<IRsend.h>
<IRrecv.h>
<IRutils.h>