mdhiggins/ESP8266-HTTP-IR-Blaster #193: SHA256 error

schroepfer, 2018-07-13:
Hello,
i get errors compiling your code. Which library "sha256.h" do you exactly use?

_IRController:201: error: 'Sha256' was not declared in this scope
     Sha256.initHmac((uint8_t*)key.c_str(), key.length()); // key, and length of key in bytes
     ^
IRController:206: error: 'HASH_LENGTH' was not declared in this scope
     String computedSignature = bin2hex(hash, HASH_LENGTH);
                                             ^
exit status 1
'Sha256' was not declared in this scope_

chiql, 2018-07-13:
Did you installed [Cryptosuite](https://github.com/jjssoftware/Cryptosuite) library on Arduino IDE?

schroepfer, 2018-07-13:
No.
Is this bad, or shall I install this?

chiql, 2018-07-13:
I met similar (but not exact) error like yours described in [fatal-error-sha256-h-no-such-file-or-directory](https://stackoverflow.com/questions/50951813/fatal-error-sha256-h-no-such-file-or-directory).

After many trials to installed some libraries, I failed.
Lastly, I installed Cryptosuite as guided in the [README](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster), it worked.

So I think you should install Cryptosuite.

schroepfer, 2018-07-16:
Installing Cryprosuite doesn't help!

_C:\Users\schrjude\Downloads\IRController\IRController.ino: In function 'bool validateHMAC(String, String, String, String)':
IRController:201: error: 'Sha256' was not declared in this scope
     Sha256.initHmac((uint8_t*)key.c_str(), key.length()); // key, and length of key in bytes
     ^
IRController:206: error: 'HASH_LENGTH' was not declared in this scope
     String computedSignature = bin2hex(hash, HASH_LENGTH);
                                              ^_
_exit status 1
'Sha256' was not declared in this scope_

chiql, 2018-07-16:
Your issue is similar to [issue 169](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/169).
Did you install Cryptosuite by "**Add .ZIP Library...**"? It cannot be installed directly from Library Manager.
Go to `Sketch > Include Library` and scroll down to see if "Cryptosuite" library is added or not. If Cryptosuite is **not** added, your install was **unsuccessful**.

schroepfer, 2018-07-16:
Yes, i did it so.

closed 2018-07-17 by mdhiggins