mdhiggins/ESP8266-HTTP-IR-Blaster #194: Phantom unknown codes received?

westerncanada, 2018-07-13:
Hello, first a big thank you for writing this, and to raptordemon for the PCB's.

I have 5 of these throughout the house and they all seem to receive phantom unknown IR.

for example:
```
Received | Command | Type | Length | Address
09:59:53 | FDAC0153 | UNKNOWN | 3 | 0x0
09:59:47 | FDAC0150 | UNKNOWN | 3 | 0x0
09:58:22 | 22AE7A2A | UNKNOWN | 3 | 0x0
09:58:22 | 8D1D8837 | UNKNOWN | 13 | 0x0
09:58:05 | 4AB0F7B6 | UNKNOWN | 3 | 0x0
```
More a curiosity than anything I only notice them because the LED goes blue.

I'm using the TSOP38238 as the IR receiver, and the v1.5 mini board, but it also did this on breadboard.

chiql, 2018-07-14:
I also receive the weird UNKNOWN codes for my Hitachi and Daikin air conditioner.
Received | Command | Type | Length | Address
00:00:00 | 60599430 | UNKNOWN | 42 | 0x0
00:00:00 | EA525C2F | UNKNOWN | 100 | 0x0
00:00:00 | 50203E79 | UNKNOWN | 100 | 0x0
00:00:00 | FDFD7DBA | UNKNOWN | 100 | 0x0

mdhiggins, 2018-07-25:
Yeah, this is a known limitation of the ESP8266 IRRemote Library and is related to the legnth and encoding of AC units

Phantom codes can happen from sunlight or certain lights that give off some IR signal

closed 2018-07-25 by mdhiggins

ralexsander, 2020-11-24:
> I also received the weird UNKNOWN codes for the Hitachi and Daikin air conditioner remote control:
> 
> ```
> Received | Command | Type | Length | Address
> 00:00:00 | 60599430 | UNKNOWN | 42 | 0x0
> 00:00:00 | EA525C2F | UNKNOWN | 100 | 0x0
> 00:00:00 | 50203E79 | UNKNOWN | 100 | 0x0
> 00:00:00 | FDFD7DBA | UNKNOWN | 100 | 0x0
> ```
> 
> However, I could receive the normal codes for the Sony TV remote control:
> 
> ```
> Received | Command | Type | Length | Address
> 00:00:00 | 210 | SONY | 12 | 0x1
> 00:00:00 | 210 | SONY | 12 | 0x1
> 00:00:00 | C10 | SONY | 12 | 0x1
> 00:00:00 | C10 | SONY | 12 | 0x1
> 00:00:00 | 610 | SONY | 12 | 0x1
> ```
> 
> The received time column is all zero as:
> `Error - EPOCH time is inappropriately low, likely connection to external time server has failed, check your network settings`
> And the zero-received-time error was fixed by downgrading ESP8266 to 2.4.0 with lwIP 1.4 PreBuild as guided in [issues 158](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/158)

Here you can see how to control your Daikin unit:
https://github.com/ralexsander/ESP8266-HTTP-IR-Blaster