mdhiggins/ESP8266-HTTP-IR-Blaster #198: Can't compile with current ver of ArduinoJSON

sledzik1984, 2018-07-22:
Arduino:1.8.3 (Windows 10), Płytka:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

IRController:54: error: 'DynamicJsonBuffer' does not name a type

 DynamicJsonBuffer jsonBuffer;

 ^

IRController:55: error: 'jsonBuffer' was not declared in this scope

 JsonObject& deviceState = jsonBuffer.createObject();

                           ^

C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino: In function 'String getUserID(String)':

IRController:243: error: 'DynamicJsonBuffer' was not declared in this scope

     DynamicJsonBuffer jsonBuffer;

     ^

C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:243:5: note: suggested alternative:

In file included from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonDocument.hpp:10:0,

                 from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:11:

C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'

     DynamicJsonBuffer;

     ^

IRController:243: error: expected ';' before 'jsonBuffer'

     DynamicJsonBuffer jsonBuffer;

                       ^

IRController:244: error: 'jsonBuffer' was not declared in this scope

     JsonObject& json = jsonBuffer.parseObject(payload);

                        ^

C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino: In function 'bool setupWifi(bool)':

IRController:388: error: 'DynamicJsonBuffer' was not declared in this scope

         DynamicJsonBuffer jsonBuffer;

         ^

C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:388:9: note: suggested alternative:

In file included from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonDocument.hpp:10:0,

                 from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:11:

C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'

     DynamicJsonBuffer;

     ^

IRController:388: error: expected ';' before 'jsonBuffer'

         DynamicJsonBuffer jsonBuffer;

                           ^

IRController:389: error: 'jsonBuffer' was not declared in this scope

         JsonObject& json = jsonBuffer.parseObject(buf.get());

                            ^

IRController:390: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'

         json.printTo(Serial);

              ^

IRController:391: error: 'class ArduinoJson::JsonObject' has no member named 'success'

         if (json.success()) {

                  ^

IRController:459: error: 'DynamicJsonBuffer' was not declared in this scope

     DynamicJsonBuffer jsonBuffer;

     ^

C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:459:5: note: suggested alternative:

In file included from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonDocument.hpp:10:0,

                 from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:11:

C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'

     DynamicJsonBuffer;

     ^

IRController:459: error: expected ';' before 'jsonBuffer'

     DynamicJsonBuffer jsonBuffer;

                       ^

IRController:460: error: 'jsonBuffer' was not declared in this scope

     JsonObject& json = jsonBuffer.createObject();

                        ^

IRController:474: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'

     json.printTo(Serial);

          ^

IRController:477: error: 'class ArduinoJson::JsonObject' has no member named 'printTo'

     json.printTo(configFile);

          ^

C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino: In lambda function:

IRController:565: error: 'DynamicJsonBuffer' was not declared in this scope

     DynamicJsonBuffer jsonBuffer;

     ^

C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:565:5: note: suggested alternative:

In file included from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonDocument.hpp:10:0,

                 from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:9,

                 from C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\Piotrek\Downloads\ESP8266-HTTP-IR-Blaster-master\ESP8266-HTTP-IR-Blaster-master\IRController\IRController.ino:11:

C:\Users\Piotrek\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/DynamicJsonBuffer.hpp:159:5: note:   'ArduinoJson::Internals::DynamicJsonBuffer'

     DynamicJsonBuffer;

     ^

IRController:565: error: expected ';' before 'jsonBuffer'

     DynamicJsonBuffer jsonBuffer;

                       ^

IRController:566: error: 'jsonBuffer' was not declared in this scope

     JsonArray& root = jsonBuffer.parseArray(server->arg("plain"));

                       ^

IRController:576: error: 'class ArduinoJson::JsonArray' has no member named 'success'

     if (!root.success()) {

               ^

IRController:674: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonArray&' from an rvalue of type 'ArduinoJson::Internals::enable_if<true, ArduinoJson::Internals::JsonObjectSubscript<const char*> >::type {aka ArduinoJson::Internals::JsonObjectSubscript<const char*>}'

           JsonArray &raw = root[x]["data"]; // Array of unsigned int values for the raw signal

                                          ^

exit status 1
'DynamicJsonBuffer' does not name a type

mdhiggins, 2018-07-22:
Install Arduino JSON, not the beta version

closed 2018-07-22 by mdhiggins

dachshund-digital, 2019-03-01:
Same issue with NON-BETA version 6.9

mdhiggins, 2019-03-03:
Use older version until I have time to migrate to the latest changes; they've restructured quite a bit

Hsmartrobot, 2019-07-31:
Hi! hepl me.......code error  ArduinoJson Ver 6.9 NON-BETA
json = jsData.get("commands").as<String>();
'ArduinoJson::JsonObject' has no member named 'get'
Thank you!