mdhiggins/ESP8266-HTTP-IR-Blaster #199: Can't access ESP8266 via browser on local network even though it's connected to router

chrisp79, 2018-07-25:
Hi. I'm pretty much a newbie when it comes to all this so apologies if I'm missing something really obvious.

With much research, self-teaching, trial and error I've managed to get as far as 6:50 in your brilliant YouTube video (https://www.youtube.com/watch?v=JysKXOdrOlM&feature=youtu.be) but no matter what I try (and I've tried everything I can think of / see mentioned online) I cannot access my ESP board.

I've found my ESP board's MAC address and I can see that it's connecting to my router. I've tried accessing the board by typing **irblaster.local:80** into my browser with no success. I do realise this is dependent on having MDNS and I'm not really sure if I do or don't, so have gone have been focussing on the safer option of entering it's IP address i.e. 192.168.0.99 which should be guaranteed to work from what I can see. However, no success with that either, I just get a "Page cannot be reached" page. I have tried many different IP addresses and can see on my router's "Connected Devices" page that it is connecting and assigning the IP addresses I enter.

My issue seems to be very similar to **Config file and WiFi Manager #110 ** - albeit his had been working previously then stopped - I've never got mine working.

Is as if the device is connecting but it isn't starting it's web interface. I bought two boards and both act the same way, so I don't think it's down to a faulty board.

Any ideas / advice would be gratefully received - I'd love to get past 06:50 in your video!!

Thanks,
Chris

chrisp79, 2018-07-25:
Hi, thanks for getting back to me so quickly.

I abandoned the first of my two boards after a few hours at the weekend when I couldn't get it working and moved onto my second board. When I couldn't get that working either, I decided it wasn't a board issue and started playing around with library versions, etc - trying every version and combinations. I did all that really intense testing with the 2nd board and it was after this that I created this thread.

For some reason, I thought just now, I'd give the original board a try seeing as I gave up on that one quite quickly - just in case. I didn't expect anything different but, that one is now working!!

I can only conclude that I inadvertently stumbled upon the correct combination of library versions, etc whilst testing with my 2nd board, but that that board is somehow bust. How annoying!

I've now been able to learn a code, create an Alexa device, associate the code with that device, get Alexa to discover it and I can see the red LED on the board flash when I ask Alexa to control it - so I've clearly managed to get the port forwarding sorted too. Brilliant!

The signal isn't actually getting through to my A/C unit though for whatever reason, so I need to 'debug' that now. But it's a huge step forward. Hopefully I'll be able to figure that out myself in the near future, but I know now where to come if I get stuck.

Thanks again
Chris

mdhiggins, 2018-07-25:
AC units can be particularly challenging as their codes are often very long and not always going to work

Make sure you’re not sending an ‘UNKNOWN’ code too, if the code is read as unknown you have to send the RAW format

From: chrisp79
Sent: Wednesday, July 25, 2018 2:13 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Can't access ESP8266 viabrowser on local network even though it's connected to router (#199)

Hi, thanks for getting back to me so quickly.
I abandoned the first of my two boards after a few hours at the weekend when I couldn't get it working and moved onto my second board. When I couldn't get that working either, I decided it wasn't a board issue and started playing around with library versions, etc - trying every version and combinations. I did all that really intense testing with the 2nd board and it was after this that I created this thread.
For some reason, I thought just now, I'd give the original board a try seeing as I gave up on that one quite quickly - just in case. I didn't expect anything different but, that one is now working!!
I can only conclude that I inadvertently stumbled upon the correct combination of library versions, etc whilst testing with my 2nd board, but that that board is somehow bust. How annoying!
I've now been able to learn a code, create an Alexa device, associate the code with that device, get Alexa to discover it and I can see the red LED on the board flash when I ask Alexa to control it - so I've clearly managed to get the port forwarding sorted too. Brilliant!
The signal isn't actually getting through to my A/C unit though for whatever reason, so I need to 'debug' that now. But it's a huge step forward. Hopefully I'll be able to figure that out myself in the near future, but I know now where to come if I get stuck.
Thanks again
Chris

chrisp79, 2018-07-25:
Awesome, thanks for the tip. Maybe I'll try getting it working with a TV first then before moving on to my A/C unit

mdhiggins, 2018-07-25:
I would definitely advise that

#187 has a post I made going through the specifics of which latest library has been tested and should be working if you want to check it against your own setup

chrisp79, 2018-07-25:
Thanks again. I will try and compare that list and my own tomorrow and see if I can make any improvements. All seems pretty stable though how I have it now ;o)

Seem to be able to receive codes and I think I understand how I transmit them now via URL. Can't get my TV to respond though. Will have another go at the weekend.

Thanks again

SirJMD, 2018-07-28:
I don't have "IwIP Variant: v1.4 Prebuilt", I have "Higher bandwidth" and "Compile from source". Currently using "Higher bandwidth".

Gonna try:
ESP8266 2.4.0 (currently 2.4.1)
ArduinoJson v5.13.2 (currently v5.13.1)
EasyNTPClient v1.0.0 (currently v.1.1.0)
IRremoteESP8266 v2.4.0 (currently v2.4.3)

SirJMD, 2018-07-28:
Now it's completely unresponsive.. 2 steps back :D

war6000, 2018-07-31:
I’m getting the same error as the previous person. Any solutions or at least knowledge of what might be the next thing to diagnose

SirJMD, 2018-07-31:
I'm stuck. No matter what I try it dies after just a few seconds.

mdhiggins, 2018-07-31:
@SirJMD If you don't see lwIP Variant v1.4 Prebuilt its because you're using the wrong version of the Arduino ESP8266 Core (needs to be 2.4.0, 2.4.1 DOES NOT work) and possibly you have the wrong board selected from your board manager. Until you get the right combination including v1.4 Prebuilt the memory leak will cause freezing and it will not work

See #187 for all the specifics

SirJMD, 2018-07-31:
@mdhiggins I noticed that after downgrading. However, 2.4.0 just results in the ESP8266 being completely unresponsive. With 2.4.1 the AP mode works fine and I can setup wifi connection and everything, but as soon as it reboots its only alive for a few seconds. 2.4.0 doesn't even get to the AP mode.

mdhiggins, 2018-07-31:
What does the serial output say?

SirJMD, 2018-07-31:
Nothing at all. No communication at all.

mdhiggins, 2018-07-31:
That doesn't make any sense, you need to reflash or recheck your Arduino libraries and potentially reinstall 2.4.0 because something is broken then. Try deleting your 2.4.0 install and redownloading

SirJMD, 2018-07-31:
I'll give that a shot tonight.

SirJMD, 2018-07-31:
Just tested deleting the esp8266 board folder and reinstalling 2.4.0. The board is yet again completely unresponsive. Nothing on the serial monitor at all.

With 2.4.1 I'm getting this:
```
ESP8266 IR Controller
Config pin GPIO10 set to: 1
failed to mount FS
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.105
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
Entered config mode
192.168.4.1
IR Controller Configuration
*WM: 
*WM: Configuring access point... 
*WM: IR Controller Configuration
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Scan done
*WM: TP-Link_701F
*WM: -89
*WM: HP-Print-90-Photosmart 6520
*WM: -95
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: WiFi save
*WM: Parameter
*WM: hostname
*WM: Blaster
*WM: Parameter
*WM: passcode
*WM: 1234
*WM: Parameter
*WM: port_str
*WM: 80
*WM: Parameter
*WM: user_id
*WM: Testuser
*WM: static ip
*WM: 192.168.1.105
*WM: static gateway
*WM: 192.168.1.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.105
*WM: Already connected. Bailing out.
Should save config
WiFi connected! User chose hostname 'Blaster' passcode '1234' and port '80'
 config...
failed to open config file for writing
{"hostname":"Blaster","passcode":"1234","port_str":"80","user_id":"Testuser","ip":"192.168.1.105","gw":"192.168.1.1","sn":"255.255.255.0"}
Writing config file
Config written successfully
WiFi configuration complete
Local IP: 192.168.1.105
URL to send commands: http://Blaster.local:80
HTTP Server started on port 80
Warning, user_id appears to be in the wrong format, security check will most likely fail. Should start with amzn1.account.***
EPOCH time obtained for security checks
Ready to send and receive IR signals
Turning off the LED to save power.
```

Right after it turns off the LED, it stops responding.

mdhiggins, 2018-07-31:
The loss of response is expected with 2.4.1 - its a known bug due to a memory leak in changes made to the WiFi libraries. There's been countless people opening issues about this and a persistent one I opened myself. Unfortunately it is not yet fixed but its out of my hands to fix until they update the ESP8266 Arduino Core

Now why 2.4.0 isn't working is very strange and to get no response at all is even more strange but I can assure you that the entire project was developed with 2.4.0 and does work so it must be something unique on your end

What board do you have set?
What are all your compile and upload options? (Frequency, SPIFFS, baud etc)
What hardware are you using?

SirJMD, 2018-07-31:
I'm using a Wemos D1 mini clone, with a ESP8266MOD. It's been working fine in other projects, but for some reason not this time.

My settings follow the ones recommended for Tasmota:
https://github.com/arendst/arendst.github.io/raw/master/media/arduinoide2b.png

What are the other hardware specs are you asking for?

mdhiggins, 2018-07-31:
Try using NodeMCU 1.0 (ESP-12E Module) and 3M SPIFFS

SirJMD, 2018-07-31:
That seems to work! It hasn't crashed yet and the web ui seems to work (it won't read IR codes yet, probably a wiring error on my part).

Interesting.. what's your reasoning behind suggesting that solution?

SirJMD, 2018-08-01:
Interesting. I will do some more testing tonight. With the "Generic ESP8266" chosen I had tried many different settings for flash size, and tested it with the ESP8266 example that tests the flash configuration.  Based on that a setting of 4M with 1M SPIFF resulted in config OK.

For this project it made no difference what I chose.

closed 2018-08-17 by mdhiggins