mdhiggins/ESP8266-HTTP-IR-Blaster #206: Output to LED

RenderPrism, 2018-08-22:
Apologies if this has already been covered.

I've been tinkering around all evening trying to get this running but haven't had much luck.

What's working:
Connecting to wifi
Receiving codes from an IR remote
Sending requests to the ESP8266 - request shows up in the monitor, seems to process alright.

What's not:
The IR blasting.

A little bit of prodding around tells me that the GPIO pin (currently using D2, have tried others) is not outputting anything. I've had it connected up to a multimeter (pin->multi->ground) to check for a voltage coming from the pin while a code should be transmitting and there's no change from 0V.
I tried using a simple sketch to check if I could get something outputted and it worked! It was a modification of the basic Blink sketch:

`void setup() {
  pinMode(D2, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(D2, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is active low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(D2, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}`

pointing pinMode() and digitalWrite() to D2 rather than the inbuilt LED. Verified function using a camera to view the IR flashing.

My ESP8266 package is v2.4.0
IwIP variant is v1.4 (prebuilt)
Board is a NodeMCU ESP12-E (set as v1.0 in Arduino IDE)
IRremoteESP8266 v2.4.3

Layout is as-per the readme, including resistors (100 and 1K) and transistor (2N2222).

[image]
[image]

The IR LED itself is different, but should be within spec (although I am new to this so feel free to tell me I'm wrong) - I've attached the spec sheet for that too.
[Z0880.pdf](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/2311412/Z0880.pdf).

Any advice is much appreciated. I'm a little stumped as to where to take it from here. It seems to me like the code isn't delivering the instructions to the board but I'm too naive to know how to address it.

Cheers,

Blair

mdhiggins, 2018-08-22:
What command are you trying to send?

From: RenderPrism
Sent: Wednesday, August 22, 2018 1:40 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Subscribed
Subject: [mdhiggins/ESP8266-HTTP-IR-Blaster] Output to LED (#206)

Apologies if this has already been covered.
I've been tinkering around all evening trying to get this running but haven't had much luck.
What's working:
Connecting to wifi
Receiving codes from an IR remote
Sending requests to the ESP8266 - request shows up in the monitor, seems to process alright.
What's not:
The IR blasting.
A little bit of prodding around tells me that the GPIO pin (currently using D2, have tried others) is not outputting anything. I've had it connected up to a multimeter (pin->multi->ground) to check for a voltage coming from the pin while a code should be transmitting and there's no change from 0V.
I tried using a simple sketch to check if I could get something outputted and it worked! It was a modification of the basic Blink sketch:
`void setup() {
pinMode(D2, OUTPUT); // Initialize the LED_BUILTIN pin as an output
}
// the loop function runs over and over again forever
void loop() {
digitalWrite(D2, LOW); // Turn the LED on (Note that LOW is the voltage level
// but actually the LED is on; this is because
// it is active low on the ESP-01)
delay(1000); // Wait for a second
digitalWrite(D2, HIGH); // Turn the LED off by making the voltage HIGH
delay(2000); // Wait for two seconds (to demonstrate the active low LED)
}`
pointing pinMode() and digitalWrite() to D2 rather than the inbuilt LED. Verified function using a camera to view the IR flashing.
My ESP8266 package is v2.4.0
IwIP variant is v1.4 (prebuilt)
Board is a NodeMCU ESP12-E (set as v1.0 in Arduino IDE)
IRremoteESP8266 v2.4.3
Layout is as-per the readme, including resistors (100 and 1K) and transistor (2N2222).

The IR LED itself is different, but should be within spec (although I am new to this so feel free to tell me I'm wrong) - I've attached the spec sheet for that too.
Z0880.pdf.
Any advice is much appreciated. I'm a little stumped as to where to take it from here. It seems to me like the code isn't delivering the instructions to the board but I'm too naive to know how to address it.
Cheers,
Blair

RenderPrism, 2018-08-22:
I've tried a few different things, but mostly:

http://ip.address:port/msg?code=DDF9B038:UNKNOWN:51&pass=XXXXX&repeat=10
(The code is what came up when I pointed my IR remote at the receiver)

http://ip.address:port/json?pass=XXXXX&plain=[%20{%20"type":"raw",%20"data":[2450,600,%201300,600,%20700,550,%201300,600,%20700,550,%201300,550,%20700,550,%20700,600,%201300,600,%20700,550,%20700,550,%20700,550,%20700],%20"khz":38,%20"pulse":3%20}%20]&repeat=10

I'd been throwing repeats on to help debug so I didn't miss a pulse, however I can tell you nothing is different without it.

mdhiggins, 2018-08-22:
So unknown codes are like error codes, they don’t actually work. You can’t send an unknown code. The RAW variant should have worked though. Try for testing sending a real code

A90:SONY:12 is a good sample

See if that works

From: RenderPrism
Sent: Wednesday, August 22, 2018 1:49 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Output to LED (#206)

I've tried a few different things, but mostly:
http://ip.address:port/msg?code=DDF9B038:UNKNOWN:51&pass=XXXXX&repeat=10
(The code is what came up when I pointed my IR remote at the receiver)
http://ip.address:port/json?pass=XXXXX&plain=[%20{%20"type":"raw",%20"data":[2450,600,%201300,600,%20700,550,%201300,600,%20700,550,%201300,550,%20700,550,%20700,600,%201300,600,%20700,550,%20700,550,%20700,550,%20700],%20"khz":38,%20"pulse":3%20}%20]&repeat=10
I'd been throwing repeats on to help debug so I didn't miss a pulse, however I can tell you nothing is different without it.

RenderPrism, 2018-08-22:
AH!
That's worked!
I'm a little concerned that the raw/JSON isn't working though, considering that I'll likely need to send raw data to work with my device.
Any suggestions there?

Thank you for the help!

mdhiggins, 2018-08-22:
Its hard to tell what the original command looked like for the RAW option because its full of % marks and HTML issues

When you paste it on here wrap it in the code markup

```
So its like this
```

RenderPrism, 2018-08-22:
Right, sorry, not used to posting in these sorts of forums.

I did double-check that because I've been on a role for silly mistakes tonight and I realised I hadn't removed the spaces from the data to be sent (hence all the escape characters).

Here's how the url should look:

`http://ip.address:port/json?pass=XXXXX&plain=[{"type":"raw","data":[2450,600,1300,600,700,550,1300,600,700,550,1300,550,700,550,700,600,1300,600,700,550,700,550,700,550,700],"khz":38,"pulse":3}]&repeat=10`

Still no luck with it though.

mdhiggins, 2018-08-22:
Looks ok, the repeat option should be in the JSON though

```
http://ip.address:port/json?pass=XXXXX&plain=[{"type":"raw","data":[2450,600,1300,600,700,550,1300,600,700,550,1300,550,700,550,700,600,1300,600,700,550,700,550,700,550,700],"khz":38,"pulse":3,"repeat":10}]
```

What kind of device are you trying to operate? Its not an air conditioner is it

RenderPrism, 2018-08-22:
Not sure how many "sending code"s should be showing up with 10 repeats but there's more than 10:

`Connection received - JSON
Raw transmit
Blocking incoming IR signals
Sending code
Sending code
Sending code
Turning off the LED to save power.
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Sending code
Transmission complete
Reenabling receiving
`
And yeah, it will be an air conditioner. The RAW isn't specific to that though, I pulled that from the readme page. I've been monitoring the LED firing with my phone's camera.

mdhiggins, 2018-08-22:
You are pulsing 3 times and repeating 10 times so 30 signals total will be sent

RenderPrism, 2018-08-22:
Right, makes sense - still seems like it's not making it to the pin output, is there any verbose logging I can bring up to try to debug?