mdhiggins/ESP8266-HTTP-IR-Blaster #207: Multiple IR Blasters

chrisp79, 2018-08-25:
Hi. So I've got my IR Blaster put together and up and running fine. I've got it set up to accept HTTP requests from an HTTP add-in in Homebridge. I don't think it's too relevant, but here's excerpt from my Homebridge JSON file...

	{
            "accessory": "Http",
            "name": "Air Conditioner",
            "switchHandling": "realtime",
            "http_method": "GET",
            "on_url":      "http://82.xx.243.xxx:80/json?plain=[{'data':[8710,4376.....",
            "off_url":     "http://82.xx.243.xxx:80/json?plain=[{'data':[8718,4376,.....",
            "service": "Switch",
            "brightnessHandling": "no",
	    "actions": ""
	}

So, having got that all set up and working nicely in my bedroom, I thought I'd build another IR Blaster to live in the lounge for other things in there.

I presumed I could give this a different (local) IP address and that it would be assigned a different external IP address, however, I always get the same external IP address no matter what local one I give it. Also, no matter what port number I give it, only port 80 seems to work. I've replicated the Port Forwarding setup I did in  on my router that got port 80 working, for other ports, but it doesn't seem to make any difference.

I've noticed in the ircontroller.ino code there's a hard-coded line "int port = 80;" and I've tried changing that too but no difference.

I know that both devices are working fine in their own right, as if I give the new device the same IP address and port as the original one it works fine (as long as the original one is powered down, else they conflict with each other).

Is what I'm trying to do even possible?

Many thanks,
Chris

mdhiggins, 2018-08-25:
I think the whole point of using home bridge is that home bridge handles the request from the external network and that forwards the commands to devices on the internal network no?

chrisp79, 2018-08-25:
Yes, that's how I understand it too. However, the only way Alexa can talk to these IR Blasters is via HTTP GET commands. So my set up is that you talk to Alexa, she sends an (external) command to my (internal) Homebridge which sends an (external) HTTP GET command to the IR Blaster.

chrisp79, 2018-08-25:
Oh, right. Sorry, I think I get what you're saying. Why am I bothering getting Homebridge to use the external HTTP address and not just using the internal one? That's a very good point! I've never tried it but will do.

It doesn't solve my problem though I don't think. I still can't seem to be get two IR Blasters to sit on my local network simultaneously.

mdhiggins, 2018-08-25:
You can definitely have more than one. If you use the local IP address they will have different IPs now (instead of the same shared external).

The solution if you’re not using homebridge is to just use different ports for each device and set up your external IP port forwarding to forward those ports to the specific local IP address

I personally have 3 of these running on my network without issue so I can attest that it works without problems. Not using homebridge though. But there’s no point to using external IP forwarding if you’re going to route through homebridge. Either use homebridge and have that talk to the local IP or don’t use it and forward the external ports directly to the device and use the built in security features via the IR Controller Alexa skill. No need to do both and you’ll only create problems trying to do both.

From: chrisp79
Sent: Saturday, August 25, 2018 12:51 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Multiple IR Blasters (#207)

Oh, right. Sorry, I think I get what you're saying. Why am I bothering getting Homebridge to use the external HTTP address and not just using the internal one? That's a very good point! I've never tried it but will do.
It doesn't solve my problem though I don't think. I still can't seem to be get two IR Blasters to sit on my local network simultaneously.

closed 2018-10-01 by mdhiggins