mdhiggins/ESP8266-HTTP-IR-Blaster #208: failed to connect to router

HJ29, 2018-08-25:
Hi,

I think my device unable to connect to router.
Hardware: nodemcu v3
IWIP varient: v2 low memory

I got the error message:
*WM: WiFi save
*WM: Parameter
*WM: hostname
*WM: irblaster
*WM: Parameter
*WM: passcode
*WM: qwer1234
*WM: Parameter
*WM: port_str
*WM: 667
*WM: Parameter
*WM: user_id
*WM: ***
*WM: static ip
*WM: 192.168.1.33
*WM: static gateway
*WM: 192.168.1.1
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.33
*WM: Already connected. Bailing out.
Should save config
WiFi connected! User chose hostname 'irblaster' passcode 'qwer1234' and port '667'
 config...
{"hostname":"irblaster","passcode":"qwer1234","port_str":"667","user_id":"***","ip":"192.168.1.33","gw":"192.168.1.1","sn":"255.255.255.0"}
Writing config file
Config written successfully
*WM: freeing allocated params!
WiFi configuration complete
Local IP: 192.168.1.33
URL to send commands: http://irblaster.local:667
ArduinoOTA started
MDNS http service added. Hostname is set to irblaster.local:667
HTTP Server started on port 667
Error retrieving external IP
HTTP Code: -1
connection refused

I can't access device via hostname or ip address.
I tried with http://irblaster.local:667,  http://192.168.1.33:667, http://192.168.1.33. None of them worked.
It looks like the device failed to connect to the router. I log into router and check the connected devices, there is no IR blaster device.
My laptop local ip address is 192.168.1.14, subnet mask 255.255.255.0, default gateway: 192.168.1.1
I don't think it is caused by ip config.

Any clue why is this happening?

mdhiggins, 2018-08-25:
#122 
#187 

Basically you need to use ESP8266 Arduino 2.4.0 not anything newer and used lwIP v1.4 otherwise you'll get networking problems

Also, what's your router's local IP address?

HJ29, 2018-08-25:
ip to access router? 192.168.1.1 if that is what you mean.

HJ29, 2018-08-25:
The device is able to connect to router after i changed the IWIP varient to v1.5 higher bandwidth.
Thank you very much.

closed 2018-08-25 by HJ29

Gorhams, 2018-08-28:
I have a similar problem.  I unable access the device (http://hostname.local:port/).  I've tried everything and always get "This site can’t be reached 192.168.0.35.local’s server IP address could not be found".  Everything else is working great; I copied a raw code from serial monitor and been using it for a few days now.  Any ideas?

mdhiggins, 2018-08-28:
And what version of ESP8266 and lwIP are you using? I assume you saw my answer before and verified that?

mdhiggins, 2018-08-28:
> Yes I verified it. lwIP 1.4 prebuilt on a NodeMCU.

Can you post your serial output? And verify the network settings correspond to your own network subnet from the computer you’re trying to access from

Gorhams, 2018-08-28:
ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"yunoblastme","passcode":"pass","port_str":"667","user_id":"amzn1.account.AGVTJYXNXQX5P3EDPYY5BRPFEXKA","ip":"192.168.0.35","gw":"192.168.0.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.0.35
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.0.35
WiFi connected! Usr chose hostname 'yunoblastme' passcode 'pass' and port '667'
*WM: freeing allocated params!
WiFi configuration complete
Local IP: 192.168.0.35
URL to send commands: http://yunoblastme.local:667
ArduinoOTA started
MDNS http service added. Hostname is set to yunoblastme.local:667
TTP Server started on port 667Turning off the LED to save power.
External IP: xx.xxx.xx.xxx
External IP address request took 3117 ms
No errors detected with security configuration
EPOCH time obtained for security checks
Ready to send and receive IR signals

Yes they match.  My computer IP is different 192.168.0.24.  I gave the irblaster 192.168.0.35 and set it up in the router with port forward

Gorhams, 2018-08-28:
ESP8266 verison is 2.4.0 in board manager and set  IWIP varient to v1.4 prebuilt