mdhiggins/ESP8266-HTTP-IR-Blaster #211: LED not lighting - is there a problem with my RAW code?

jxwolstenholme, 2018-09-04:
Hi there,

I've spent a couple of days trying to get the LED to light up with a seemingly fairly odd protocol by Lutron lights. As I don't have an IR remote for them, I have had to use the pronto hex code that they provide in their documentation:

0000 0069 0007 0000 032a 010e 005a 010e 005a 005a 0168 005a 005a 010e 005a 00b4 005a 01c2

Converting this with IrSzrutinizer I get the raw code:

[20518,6839, 2280,6839, 2280,2280, 9119,2280, 2280,6839, 2280,4560, 2280,11399]

This doesn't seem to light up the LED and I have tried the test Sony code which does, so I can only assume that the circuit is wired correctly.

Is there any way of debugging why this might not be working for me? Is it perhaps something to do with how seemingly short the code is?

http://192.168.1.152:667/json?pass=mypass&out=2&plain=[{"type”:”raw”,”data”:[20518,6839, 2280,6839, 2280,2280, 9119,2280, 2280,6839, 2280,4560, 2280,11399],”khz”:40,"repeat":3,"rdelay":800}]

Many thanks for any help in advance.

mdhiggins, 2018-09-04:
Haven't had time to test this yet mostly because I don't know any pronto codes for my devices but here is a beta blueprint that supports sending pronto codes

```
#include <FS.h>                                               // This needs to be first, or it all crashes and burns

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>                                      // https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <ESP8266mDNS.h>                                      // Useful to access to ESP by hostname.local

#include <ArduinoJson.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoOTA.h>
#include "sha256.h"

#include <Ticker.h>                                           // For LED status
#include <NTPClient.h>

// User settings are below here

const bool getExternalIP = true;                              // Set to false to disable querying external IP

const bool getTime = true;                                    // Set to false to disable querying for the time
const int timeOffset = -14400;                                // Timezone offset in seconds

const bool enableMDNSServices = true;                         // Use mDNS services, must be enabled for ArduinoOTA

const unsigned int captureBufSize = 150;                      // Size of the IR capture buffer.

// WEMOS users may need to adjust pins for compatability
const int pinr1 = 14;                                         // Receiving pin
const int pins1 = 4;                                          // Transmitting preset 1
const int pins2 = 5;                                          // Transmitting preset 2
const int pins3 = 12;                                         // Transmitting preset 3
const int pins4 = 13;                                         // Transmitting preset 4
const int configpin = 10;                                     // Reset Pin

// User settings are above here
const int ledpin = BUILTIN_LED;                               // Built in LED defined for WEMOS people
const char *wifi_config_name = "IR Controller Configuration";
const char serverName[] = "checkip.dyndns.org";
int port = 80;
char passcode[20] = "";
char host_name[20] = "";
char port_str[6] = "80";
char user_id[60] = "";
const char* fingerprint = "8D 83 C3 5F 0A 09 84 AE B0 64 39 23 8F 05 9E 4D 5E 08 60 06";

char static_ip[16] = "10.0.1.10";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";

DynamicJsonBuffer jsonBuffer;
JsonObject& deviceState = jsonBuffer.createObject();

ESP8266WebServer *server = NULL;
Ticker ticker;

bool shouldSaveConfig = false;                                // Flag for saving data
bool holdReceive = false;                                     // Flag to prevent IR receiving while transmitting

IRrecv irrecv(pinr1, captureBufSize);
IRsend irsend1(pins1);
IRsend irsend2(pins2);
IRsend irsend3(pins3);
IRsend irsend4(pins4);
[... 1465 lines cut ...]
  server->handleClient();
  decode_results  results;                                        // Somewhere to store the results

  if (irrecv.decode(&results) && !holdReceive) {                  // Grab an IR code
    Serial.println("Signal received:");
    fullCode(&results);                                           // Print the singleline value
    dumpCode(&results);                                           // Output the results as source code
    copyCode(last_recv_4, last_recv_5);                           // Pass
    copyCode(last_recv_3, last_recv_4);                           // Pass
    copyCode(last_recv_2, last_recv_3);                           // Pass
    copyCode(last_recv, last_recv_2);                             // Pass
    cvrtCode(last_recv, &results);                                // Store the results
    last_recv.timestamp = timeClient.getEpochTime();               // Set the new update time
    last_recv.valid = true;
    Serial.println("");                                           // Blank line between entries
    irrecv.resume();                                              // Prepare for the next value
    digitalWrite(ledpin, LOW);                                    // Turn on the LED for 0.5 seconds
    ticker.attach(0.5, disableLed);
  }
  delay(200);
}
```

Sample URL using your pronto code
`http://10.0.1.46:667/json?pass=pass&plain=[{type:'pronto',data:[0000,0069,0007,0000,032a,010e,005a,010e,005a,005a,0168,005a,005a,010e,005a,00b4,005a,01c2]}]`

jxwolstenholme, 2018-09-07:
@mdhiggins Thanks for sharing! unfortunately it still doesn't work for me. The led is now lighting but not working with the pronto codes that I have access to.

Is there any chance that the following (taken from the protocol specification) makes sense to you as to why it might not work? I feel like it might be some sort of non-standard format.

IR Carrier Frequency: 40.0 kHz 
Duty Cycle: 40% 
Single Bit Time: 2.288 milliseconds 
Baud Rate: 437 bps 
Command Length: 36 bits 
Command Duration: 82.368 milliseconds 
Logic One: Presence of IR modulated at 40.0 kHz 
Logic Zero: Absence of IR 
Transmit Order: Transmit the most significant bit first 
General Function: IR code is transmitted while a button is held down 
Timeout Function: Timeouts may not occur until at least seven-seconds of continuous IR transmission has taken place.

[image]

mdhiggins, 2018-09-07:
Actually I think I there's an error in the above blueprint that's not translating the array of URL parameters for the pronto codes into the appropriate uint_16 format needed. I'll have a new test blueprint for you shortly

mdhiggins, 2018-09-07:
Alright give this a try

And for the pronto codes prefix each entry with a `0x` to indicate they are in HEX format

Example URL

`http://10.0.1.46:667/json?pass=pass&plain=[{type:'pronto',data:[0x0000, 0x0069, 0x0007, 0x0000, 0x032a, 0x010e, 0x005a, 0x010e, 0x005a, 0x005a, 0x0168, 0x005a, 0x005a, 0x010e, 0x005a, 0x00b4, 0x005a, 0x01c2]}]`

```
#include <FS.h>                                               // This needs to be first, or it all crashes and burns

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>                                      // https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <ESP8266mDNS.h>                                      // Useful to access to ESP by hostname.local

#include <ArduinoJson.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoOTA.h>
#include "sha256.h"

#include <Ticker.h>                                           // For LED status
#include <NTPClient.h>

// User settings are below here

const bool getExternalIP = true;                              // Set to false to disable querying external IP

const bool getTime = true;                                    // Set to false to disable querying for the time
const int timeOffset = -14400;                                // Timezone offset in seconds

const bool enableMDNSServices = true;                         // Use mDNS services, must be enabled for ArduinoOTA

const unsigned int captureBufSize = 150;                      // Size of the IR capture buffer.

// WEMOS users may need to adjust pins for compatability
const int pinr1 = 14;                                         // Receiving pin
const int pins1 = 4;                                          // Transmitting preset 1
const int pins2 = 5;                                          // Transmitting preset 2
const int pins3 = 12;                                         // Transmitting preset 3
const int pins4 = 13;                                         // Transmitting preset 4
const int configpin = 10;                                     // Reset Pin

// User settings are above here
const int ledpin = BUILTIN_LED;                               // Built in LED defined for WEMOS people
const char *wifi_config_name = "IR Controller Configuration";
const char serverName[] = "checkip.dyndns.org";
int port = 80;
char passcode[20] = "";
char host_name[20] = "";
char port_str[6] = "80";
char user_id[60] = "";
const char* fingerprint = "8D 83 C3 5F 0A 09 84 AE B0 64 39 23 8F 05 9E 4D 5E 08 60 06";

char static_ip[16] = "10.0.1.10";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";

DynamicJsonBuffer jsonBuffer;
JsonObject& deviceState = jsonBuffer.createObject();

ESP8266WebServer *server = NULL;
Ticker ticker;

bool shouldSaveConfig = false;                                // Flag for saving data
bool holdReceive = false;                                     // Flag to prevent IR receiving while transmitting
[... 1473 lines cut ...]
void loop() {
  timeClient.update();
  ArduinoOTA.handle();
  server->handleClient();
  decode_results  results;                                        // Somewhere to store the results

  if (irrecv.decode(&results) && !holdReceive) {                  // Grab an IR code
    Serial.println("Signal received:");
    fullCode(&results);                                           // Print the singleline value
    dumpCode(&results);                                           // Output the results as source code
    copyCode(last_recv_4, last_recv_5);                           // Pass
    copyCode(last_recv_3, last_recv_4);                           // Pass
    copyCode(last_recv_2, last_recv_3);                           // Pass
    copyCode(last_recv, last_recv_2);                             // Pass
    cvrtCode(last_recv, &results);                                // Store the results
    last_recv.timestamp = timeClient.getEpochTime();               // Set the new update time
    last_recv.valid = true;
    Serial.println("");                                           // Blank line between entries
    irrecv.resume();                                              // Prepare for the next value
    digitalWrite(ledpin, LOW);                                    // Turn on the LED for 0.5 seconds
    ticker.attach(0.5, disableLed);
  }
  delay(200);
}
```

jxwolstenholme, 2018-09-25:
Never ended up testing the pronto support unfortunately but did get help from the guys at IRremoteESP8266 in decoding the Lutron protocol.

They pushed out support in  v2.5 which I have then submitted a request to add to your repository

closed 2018-09-25 by jxwolstenholme