mdhiggins/ESP8266-HTTP-IR-Blaster #213: Sent code displayed is wrong

regeg, 2018-09-18:
I'm successfully getting the command from Alexa the blaster, but the code blaster reports sending is:

Sent | Command | Type | Length | Address
18:59:48 | FFFFFFFFFFFFFFFF | NEC | 0 | 0X0

Rather than the code sent.  The "NEC" is correct, but that's the only thing that is.

Any recommendations on where to start digging into things?
Thanks!

mdhiggins, 2018-09-18:
Post your URL without your IP address so I can see what you’re sending

regeg, 2018-09-19:
So....  When I sent a code today - it worked.  (Volume commands.  I haven't done a JSON packet yet to turn it on w/ the repeated commands.)  But, I'm seeing random sets of FF's on the web page.

[image]

mdhiggins, 2018-09-19:
That’s probably just some IR interference or noise coming from either another device or event ambient light / sunlight
Nothing to be concerned about

From: regeg
Sent: Wednesday, September 19, 2018 6:12 PM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Sent code displayed is wrong(#213)

So.... When I sent a code today - it worked. (Volume commands. I haven't done a JSON packet yet to turn it on w/ the repeated commands.) But, I'm seeing random sets of FF's on the web page.

regeg, 2018-09-19:
Well, it's not behaving deterministicly.  I have the following commands defined:
[image]

It looks like the JSON isn't being processed properly.  (I did the 'ON' first then the 'OFF')

[image]

And here's the serial output:
[image]

These are successive commands without changing anything.

closed 2018-09-19 by regeg