mdhiggins/ESP8266-HTTP-IR-Blaster #214: Cannot connect to webserver after configuration

RienduPre, 2018-09-20:
Hi,

I can configure the wifi setup without a problem, but when it connects to my home wifi network I cannot open the IR management web interface, nor it can be pinged on the given ip adres.

However logging seems normal (see below)

ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.10
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
Entered config mode
192.168.4.1
IR Controller Configuration
*WM: 
*WM: Configuring access point... 
*WM: IR Controller Configuration
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Scan done
*WM: DUP AP: Home2
*WM: Home2
*WM: -70
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Parameter
*WM: hostname
*WM: IRBLASTER
*WM: Parameter
*WM: passcode
*WM: 12345678
*WM: Parameter
*WM: port_str
*WM: 80
*WM: Parameter
*WM: user_id
*WM: 
*WM: static ip
*WM: 192.168.0.144
*WM: static gateway
*WM: 192.168.0.254
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.0.144
*WM: Already connected. Bailing out.
Should save config
WiFi connected! User chose hostname 'IRBLASTER' passcode '12345678' and port '80'
 config...
{"hostname":"IRBLASTER","passcode":"12345678","port_str":"80","user_id":"","ip":"192.168.0.144","gw":"192.168.0.254","sn":"255.255.255.0"}
Writing config file
Config written successfully
*WM: freeing allocated params!
WiFi configuration complete
Local IP: 192.168.0.144
URL to send commands: http://IRBLASTER.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to IRBLASTER.local:80
HTTP Server started on port 80
Ready to send and receive IR signals
Turning off the LED to save power.

mdhiggins, 2018-10-01:
And you've got everything configured on the same subnet? Post your computer and router IP configurations as well as the URL you're trying

cap60552, 2018-10-04:
I am experiencing the same thing.      
I can ping the ESP for a few seconds after a reboot/connect cycle and if i hit the webserver instantly after it boots and connects i can view the page, but it is only operational for a couple of seconds.   the whole device seems to crash at that point.    

I did make sure to use the lwIP variant 1.4, but it didn't seem to make a difference.  I also tried 2 different devices, a standard nodeMCU 1.0 device, and a wemos v1 device.  Both experienced the same problem.

Sawoff, 2018-10-04:
I've the same problem here

cap60552, 2018-10-04:
I missed that bug. I'll try downgrading the board library tonight when I get home.

jonsel087, 2018-10-09:
Hi all, really hoping someone can help me debug. I am unable to connect to the webserver, even after downgrading to ESP8266 2.4.0 and using "IwIP variant Prebuilt". The serial plotter reads:
```
ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.10
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 10.0.1.10
WiFi connected! User chose hostname 'IRBLASTER' passcode '2345' and port '80'
*WM: freeing allocated params!
WiFi configuration complete
Local IP: 10.0.1.10
URL to send commands: http://IRBLASTER.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to IRBLASTER.local:80
HTTP Server started on port 80
Turning off the LED to save power.
Error retrieving external IP
HTTP Code: -1
connection refused
External IP address request took 5002 ms
Ready to send and receive IR signals
Signal received:
One line: FFFFFFFFFFFFFFFF:NEC:0 (Repeat)
uint16_t  rawData[3] = {8864,2360, 520};  // NEC FFFFFFFFFFFFFFFF
uint64_t  data = 0xFFFFFFFFFFFFFFFF;
Turning off the LED to save power.
```

From what i see the issue is that it's not setting the default IP to my subnet. Any ideas? Thanks in advance!

cap60552, 2018-10-09:
One thing you could try is to turn off the static up and use DHCP.  Just try commenting out :
wifiManager.setSTAStaticIPConfig(sip, sgw, ssn);

And wifiManager will default to DHCP.    Then you can make sure everything is still running and hasn't crashed.

jonsel087, 2018-10-10:
> One thing you could try is to turn off the static up and use DHCP. Just try commenting out :
> wifiManager.setSTAStaticIPConfig(sip, sgw, ssn);
> 
> And wifiManager will default to DHCP. Then you can make sure everything is still running and hasn't crashed.

Cap, thank you so much! That indeed got it working. I was able to get the local server running and configure the device. It's showing up on my Alexa app, unfortunately when I try to issue a command to Alexa it says the device is not responding. If I open up the device on the app, it just shows device info, not the Power On/Off functionality i added. Any idea if that's normal?

mdhiggins, 2018-10-10:
That’s normal. Did you forward your ports and are able to view the device from your external IP? And if DHCP is working then I would guess you didn’t configure your static network settings correctly

skorokithakis, 2018-10-10:
Why is it defaulting to static, though? DHCP is the better default for the vast majority of users.

skorokithakis, 2018-10-10:
I'm afraid this is happening to me with DHCP as well. It seems like the ESP is disassociating from the WiFi access point, as I can no longer see it in the list of devices on the router after a few seconds. I don't know why this might be, though...

skorokithakis, 2018-10-10:
Even with LwIP 1.4, I am experiencing crashes. Does anyone have a working build they can upload somewhere?

jonsel087, 2018-10-11:
I'm not able to connect to the esp8266 with port forwarding turned on (am trying to connect to external IP from my cell phone, not sure if that's right. will need to find some documentation). When connecting locally, getting unresponsive pages/poor performance. Additionally, seeing these logs in my router: 
```
Wed Oct 10 22:18:01 2018 | Critical (3) | SYNC Timing Synchronization failure - Failed to acquire QAM/...
Wed Oct 10 22:18:03 2018 | Warning (5) | MDD message timeout;CM-MAC=00:40:36:4b:f7:40;CMTS-MAC=00:01:...
```

mdhiggins, 2018-10-11:
DHCP isn't the default because its a static device with port forwarding, using DHCP doesn't make sense as any time DHCP reassigns IP address your port forwarding settings will fail

Post your full Arduino build settings (should be displayed at the status bar at the bottom of your IDE), and post all your router IP address settings (IP, subnet) and post the static settings you were trying to assign. 

Only thing you don't need to post is your external IP address, and I wouldn't advise that.

mdhiggins, 2018-10-11:
Also, are there any special networking circumstances? Like are you behind a double NAT for instance?

skorokithakis, 2018-10-11:
@mdhiggins I assign static IP addresses on my router, so the DHCP server always assigns the same address, and it's easier than typing all the information in. Myself, I didn't build using Arduino IDE, I used PlatformIO with 2.4.0 and LwIP 1.4, as in #222, but there's a lot of instability, the device disassociates, hangs, etc.

skorokithakis, 2018-10-11:
I think if you run the firmware in #222 you'll see the same thing, since the build is pretty repeatable.

mdhiggins, 2018-10-11:
Do you have all the same issues when using Arduino? I’ve never used platform IO

skorokithakis, 2018-10-11:
I'm not entirely sure, I've never used Arduino and IDE I don't even have it installed :/

mdhiggins, 2018-10-11:
I mean its tough for me to support a platform I've never used and didn't design the project for. Are others using PIO and having the same experience?

And I acknowledge that routers can keep IP addresses by using static settings and matched the MAC addresses but not all routers have this feature so static makes sense; plus its very basic information that it needs, just an IP and the subnet/router. It would be nice if Wifimanager more directly supported both modes but it does not.

I'm looking at the platform IO settings that are from the git pull and I would try changing the  board

Use `nodemcuv2` instead of whatever is in there and also change to stable 2.4.0 not 2.4.0-rc2 as I'm not sure when the networking memory leak was introduced but it might be present in the rc2 build, just use the stable build.

mdhiggins, 2018-10-11:
Also looking at the build flags and I see PIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH
Not 100% sure if that is equivalent to lwIP v1.4 prebuilt that's included with the nodemcuv2 board option that Arduino IDE uses (the high bandwidth and low bandwidth variants are not options when you switch to the correct board) but looking at the PIO documentation I don't see any additional build flags

mdhiggins, 2018-10-11:
d581294747eee5a7dbef4ed8ff2227a420dd9a79

May need to mess with the lwIP variant still but give that a try

skorokithakis, 2018-10-11:
My problem seems to have been with my router, I realized this when my ESPs started disconnecting one by one. In my case I set my router's encryption to TKIP/AES and everything was fixed. Ignore me, PlatformIO builds work fine.

jonsel087, 2018-10-12:
> DHCP isn't the default because its a static device with port forwarding, using DHCP doesn't make sense as any time DHCP reassigns IP address your port forwarding settings will fail
>
> Post your full [...]

Hi Mike,
Thanks for the support, really hoping to get this working! Noted about DHCP, I will switch to static once i get it running (for some reason can't retrieve external IP when using static). I've double checked the IPs for now to make sure they haven't changed.
* Here's my build settings: `NodeMCU 1.0 (ESP-12E Module), 80MHz, 4M (1M SPIFFS), v1.4 Prebuilt, Disabled, Non, 115200 on Com4`
* Chip Model: `ESP8266MOD`
* Gateway: 192.168.0.1
* ESP ip: 192.168.0.19:80
* i have set up port forwarding and specified the external/internal IP addresses and the correct port. There is also a DMZ function i turned on for the local IP which bypasses firewall

I ran through the process again, this time specifying my amazon  user id in the `user_id` field. I updated the passcode and edited the request url but getting the same result. (i left hostname at default, i'm unable to connect using mdns anyways so using local ip).

This time I'm getting a new message from the serial plotter:
```
Epoch time from timeServer is unexpectedly old, probably failed connection to the time server. Check your network settings
37
Invalid EPOCH time, security checks may fail if unable to sync with NTP server
```

Dumb question, but could you please confirm which of the urls from captured signal to specify? (i.e, external, hostname, local) I'm assuming external is correct, but what are the others for?

mdhiggins, 2018-10-12:
I wouldn't use both port forwarding and DMZ, just do port forwarding as you are opening up way more ports than you need to use DMZ

I assume your router's local IP address is also 192.168.0.1 and you can access the ESP8266's web portal

Your build settings all look correct, but its still failing to get the EPOCH time. EPOCH time is unix time, which should be a large number (can see the current unix time here https://www.epochconverter.com/)

Sometimes I've seen it fail due to strange DNS setups where the router didn't support DNS hosting, a limitation to WiFiManager's static IP address settings. Was the EPOCH time giving you an error and small number when you did the settings via DHCP?

One thing you can try to take DNS out of the equation is to modify the blueprint and change line 71 from
`const char* poolServerName = "time.google.com";` to `const char* poolServerName = "216.239.35.0";`
This is the direct IP to time.google.com so you don't depend on DNS and see if that fetches the appropriate EPOCH time

mdhiggins, 2018-10-12:
And yes, Alexa needs to be pointing to the external IP address since the commands from the skill come from Amazon's servers outside of your local network, and you should be able to access your device's status page from your external IP address

cap60552, 2018-10-12:
I use DHCP to assign a static ip. Recently I added a secured iot vlan and
was able to shift all of my IR blasters to the new subnet and had
everything up and running on the new subnet in a matter of minutes without
having to recompile or try to get OTA to work.  This is especially nice in
the case of the blasters I have embedded in waterproof boxes mounted
outside used for controlling cheap floodlights.

closed 2018-10-20 by mdhiggins