mdhiggins/ESP8266-HTTP-IR-Blaster #217: Channel Setup Type missing

maskiel, 2018-10-01:
Hello !
I am trying to user the chennel-Setup. This is my base-URL:
`http://xxx.domain.xy:1234/json?pass=*******`
and for the channel 1 for example I have defined `data:'201',type:'RC5',length:12`
Alexa says "Okay" after I Say the command to switch to channel 1 (In german "Alexa, schalte Fernseher auf Programm 1"), the LED at the MCU blinks once, but the TV doesn´t switch. 
Volume Control and channel up/down works perfect. 
When I am looking at the Arduino IDE, I can see, that sending e.g. VOL UP it Shows `210:rc5:12`
But when I call Programm the MCU receives `201::12`
Ooops ! The type ist missing :-( 

Channel Setup is the only part of the commandsetup, that uses json. The other ones were defined as msg. 
Im I doing wrong ? And if so, what ? 

Greetings from Northern Germany
Martin

mdhiggins, 2018-10-01:
Could you send your full JSON object to me and I'll take a look?

Don't post here as it may contain your public IP address, email it to me please

mdhiggins, 2018-10-01:
Also if you could post the serial output for your attempt that's failing I can take a look at that as well