mdhiggins/ESP8266-HTTP-IR-Blaster #218: Change channel on Roku TV

tmurphe, 2018-10-05:
I am trying to figure out a way to use your IR Controller Alexa skill to change channels on a Roku TV. I believe that it is not possible? I have everything working except for the Alexa skill. I can send a request from a web browser to the ESP which then correctly constructs the request according to the Roku API and sends it to the Roku TV to change the channel. It works both internally and externally so port forwarding is good. I thought perhaps I could edit the JSON object created by the device setup webpage but I'm afraid it's above  my current abilities. Roku TV's do not allow direct channel access via IR but only through the API. If you could give me an idea on a next step or perhaps a work-a-round I would really appreciate it.

You have done a tremendous job with this project and I'm awed at the time and effort you have put into it. Your continued support of it is also above and beyond. Great work!

mdhiggins, 2018-10-10:
I have a Roku TV but I don't use cable so can't say for sure.

Looks like via the API the command sound be something like http://192.168.1.134:8060/launch/tvinput.dtv?ch=1.1

The channel controller configuration option as it is now wouldn't be compatible with this though. However I could probably modify things to get that working. Does that API command work to change the channel though?

tmurphe, 2018-10-10:
Yes the API command works perfect. A modification would be greatly appreciated. Another item on my wish list would be to be able to load Roku "channels" directly (launch/appID) however this command is sent using an HTTP POST with no body. Thanks for responding!

tmurphe, 2018-10-11:
Unless I did something wrong it appears the update to tehpsyc.pythonanywhere.com isn't working. 
I added the local IP for my Roku TV to the Channel Controller.
Defined my channels in Channel Setup.
Updated Device.
In the Alexa app I clicked "forget" on the device (Living Room TV).
I ran Discover and the device was found: Living Room TV 
                                                                  Living room TV via IR Controller	
Me: "Alexa change channel to CBS on living room TV."
Alexa: "Sorry, living room TV appears to be malfunctioning."
Serial Monitor: Connection received - JSON
                         JSON parsing failed
Me: "Alexa change channel to eleven dot one on living room TV."
Alexa: "Sorry, living room TV appears to be malfunctioning."
Serial Monitor: Connection received - JSON
                         JSON parsing failed
I'm sorry, I didn't realize launching Roku apps was already supported. I didn't see anything in the readme or anyway to set it up to work with Alexa. Can you point me to what I missed? Thanks again!

mdhiggins, 2018-10-11:
Did you fill out the baseURL part above it with the json URL?
Can you post a screenshot of those 2 settings (the baseURL for channel controller and the local IP setting) you can blur out your external IP address but the local is safe to include

tmurphe, 2018-10-11:
When I went back into the device setup web page the Roku IP address was no longer there. I'm guessing what happened is that I entered the Roku IP and then went to the channel setup page without updating the device and the IP was lost. I have re-entered the IP, updated device, and did the discover steps again. Now the command is getting generated but the TV is not responding.

Serial Monitor: Connection received - JSON
http://192.168.1.102:8060/launch/tvinput.dtv=11.1
Sending roku command
Turning off the LED to save power.

The command should be: 
http://192.168.1.102:8060/launch/tvinput.dtv?ch=11.1

tmurphe, 2018-10-11:
Yes! Works great on most channels. Alexa is understanding both channel name and channel number. Some not responding. For example, Alexa isn't understanding "CW" but will accept the channel "forty six dot one". I'll look into it further. I'm sure it's on the Amazon end and not the IR Controller. Mr. Higgins thank you so much for your efforts. I lost the ability to change channels via Alexa when I bought the Roku TV and realized how lazy I'd become! I'm back to a couch slug. Really appreciate your magic. I agree with other posters that it would be nice to buy you a beer or something.

mdhiggins, 2018-10-11:
Found the failure in the logs for CW and it might actually be a stupid case sensitivity issue. Usually the channel listings should be forced to lower case but it seems on your channel listing CW is capitalized. I'm improved the matching algorithm to be case insensitive so try CW again

mdhiggins, 2018-10-11:
And to answer your question about app launching, you can just use the generic roku command to launch whatever app if you have the appID

http://xxx.xxx.xxx.xxx:port/msg?pass=yourpass&type=roku&data=launch/appID&ip=10.0.1.3

You can either build that in to the power on command which is what I do for plex, or create a 'scene controller' and use that to activate different apps on the TV

tmurphe, 2018-10-11:
Yup CW works now. The only one not working is "laff". I changed it to "laugh" thinking it might work but Alexa is still confused. I can access the channel via the number so it's not a big deal. Take a look if you're curious. I'll play around with your suggestions. Thanks for the info.

tmurphe, 2018-10-12:
"EDIT: Did you by chance change it in the RAW channel JSON data? Either way, the backend will handle case insensitivity now"

No, no changes to the RAW JSON data.

Sorry it took awhile to wrap this old brain around what is happening behind the scenes. Thanks to your response on how you set up Plex I realized I could just make Roku apps different devices. Now, "Alexa turn on Hulu" and viola! Magic.

mdhiggins, 2018-10-12:
Glad you got it all working and this is a nice feature to have added to the Roku functionality

Gonna close this issue for now feel free to reopen with any problems

closed 2018-10-12 by mdhiggins