mdhiggins/ESP8266-HTTP-IR-Blaster #219: German Skill commands

maskiel, 2018-10-09:
Hello !

Is there a german "vocabulary-book"  for the Skill ? 

I found some thread here about the Channel-Controller and some works for me. 
I wonder if there is a commandlist for the playback controller. 

Best regards from Northern Germany
Martin

maskiel, 2018-10-10:
Can you please write a link to that Website ? I haven´t found ist yet. 
I tried several Commands and they seem to be very inconsistant, which is uncomfortable to rememember. 

e.g. 

...schalte Fernseher EIN/AUS  (switch TV on/off)
...schalte Fernseher Programm 5  (switch TV program 5)
...Fernseher Lautstärke stummschalten (TV mute)
...schalte Fernseher lauter (switch TV louder) 
...Ein Kanal vor/zurück auf Fernseher (one Channel up/down on TV)
...wechsel auf Fernseher den Kanal zu ZDF (change on TV the channel to ZDF)

………………..
sometimes TV on beginning, sometimes at the end of the command. It seems to be a bit inconsistant. I am searching for a common Grammar in the possible commands. 

I would like to have vocabulary to use the playback-Controller, which I want to use to start and playback timeshift-recording.

mdhiggins, 2018-10-20:
https://developer.amazon.com/de/docs/device-apis/alexa-channelcontroller.html

That's the most documentation Alexa has published, its not a lot but that's all there is an they do sometimes make changes without updating. Unfortunately its out of my hands and I don't actually have any control as a skill developer of what the commands actually are

closed 2018-10-20 by mdhiggins