mdhiggins/ESP8266-HTTP-IR-Blaster #221: Doesn't Compile

bspiegel33, 2018-10-10:
Great project.  I realize it's been some time since ver 2 was posted.  I am trying to compile the code and getting errors.  Can you identify which versions of the code you used so I can get this to compile?  Much appreciated.  BTW, I am using the ver2 IRController.ino file.

mdhiggins, 2018-10-10:
I'm not sure specifically, I would try and upgrade to the latest version since the older ones are not supported

closed 2018-10-10 by mdhiggins

skorokithakis, 2018-10-10:
I made [a fork to add PlatformIO](https://github.com/skorokithakis/ESP8266-HTTP-IR-Blaster) to the library, but it, indeed, [fails with errors](https://travis-ci.com/skorokithakis/ESP8266-HTTP-IR-Blaster/builds/87495047). If you can help me debug this, I can issue a PR for the PlatformIO integration, which will make building and flashing the program basically just `git clone <repo>; platformio run -t upload`.

skorokithakis, 2018-10-10:
@bspiegel33 You can use #222, it compiles fine.

bspiegel33, 2018-10-10:
Thanks!  Got it to compile.  I'm new to github so didn't see these additional pull requests.

skorokithakis, 2018-10-10:
Oh I just made them a few minutes ago.

bspiegel33, 2018-10-10:
Well, my luck on timing.  Thanks again for sharing it.