mdhiggins/ESP8266-HTTP-IR-Blaster #224: Check errors -

vantax, 2018-10-20:
Thank you for your  fast response to last error.

Do you have a donation page?

Now seing check errors following.

IRController:946: error: 'LUTRON' was not declared in this scope

     case LUTRON:       output = "LUTRON";             break;

          ^

C:\Temp\IRController\IRController.ino: In function 'void cvrtCode(Code&, decode_results*)':

IRController:1200: error: 'uint64ToString' was not declared in this scope

   strncpy(codeData.data, uint64ToString(results->value, 16).c_str(), 40);

                                                           ^

IRController:1205: error: 'kRawTick' was not declared in this scope

       r += results->rawbuf[i] * kRawTick;

                                 ^

C:\Temp\IRController\IRController.ino: In function 'void dumpRaw(decode_results*)':

IRController:1254: error: 'kRawTick' was not declared in this scope

     uint32_t x = results->rawbuf[i] * kRawTick;

                                       ^

C:\Temp\IRController\IRController.ino: In function 'void dumpCode(decode_results*)':

IRController:1287: error: 'kRawTick' was not declared in this scope

     Serial.print(results->rawbuf[i] * kRawTick, DEC);

                                       ^

C:\Temp\IRController\IRController.ino: In function 'void irblast(String, String, unsigned int, int, int, int, int, long int, IRsend)':

IRController:1397: error: 'class IRsend' has no member named 'sendLutron'

         irsend.sendLutron(data, len);

                ^

Using library IRremoteESP8266 at version 2.2.0 in folder: C:\Users\Ian\Documents\Arduino\libraries\IRremoteESP8266 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Ian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi 
Using library WiFiManager at version 0.12 in folder: C:\Users\Ian\Documents\Arduino\libraries\WiFiManager 
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\Ian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer 
Using library DNSServer at version 1.1.0 in folder: C:\Users\Ian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\DNSServer 
Using library ESP8266mDNS in folder: C:\Users\Ian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS (legacy)
Using library ArduinoJson at version 5.11.1 in folder: C:\Users\Ian\Documents\Arduino\libraries\ArduinoJson 
Using library ESP8266HTTPClient at version 1.1 in folder: C:\Users\Ian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266HTTPClient 
Using library ArduinoOTA at version 1.0 in folder: C:\Users\Ian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ArduinoOTA 
Using library Sha in folder: C:\Program Files (x86)\Arduino\libraries\Sha (legacy)
Using library Ticker at version 1.0 in folder: C:\Users\Ian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\Ticker 
Using library NTPClient at version 3.1.0 in folder: C:\Users\Ian\Documents\Arduino\libraries\NTPClient 
exit status 1
'LUTRON' was not declared in this scope

mdhiggins, 2018-10-20:
Looks like you’re using an old version of the IRRemote library

vantax, 2018-10-20:
Thank you so much for your support, library update resolved.

closed 2018-11-10 by mdhiggins