mdhiggins/ESP8266-HTTP-IR-Blaster #225: esp8266 serial output says its connected to my wifi, but it's not

nadnerb33, 2018-10-27:
I'm having issues getting this set up.
I can connect to the wifi config page by grounding GPIO10, I set this up and save it.
Serial port shows the below indicating it has connected to my WiFi network, but  access via IP or .local addresses does not work and no sign of any connection being made in my router logs.

```
11:33:50.892 -> *WM: 192.168.0.65
11:33:50.892 -> *WM: static gateway
11:33:50.892 -> *WM: 192.168.0.1
11:33:50.892 -> *WM: static netmask
11:33:50.892 -> *WM: 255.255.255.0
11:33:50.926 -> *WM: Sent wifi save page
*WM: Connecting to new AP
11:33:52.894 -> *WM: Connecting as wifi client...
11:33:52.928 -> *WM: Custom STA IP/GW/Subnet
11:33:52.928 -> *WM: 192.168.0.65
11:33:52.928 -> *WM: Already connected. Bailing out.
Should save config
11:33:52.965 -> WiFi connected! User chose hostname 'irblaster' passcode 'xxx' and port '8080'
11:33:52.965 ->  config...
11:33:52.965 -> {"hostname":"irblaster","passcode":"pass","port_str":"8080","user_id":"xxx","ip":"192.168.0.65","gw":"192.168.0.1","sn":"255.255.255.0"}
11:33:53.002 -> Writing config file
11:33:53.002 -> Config written successfully
11:33:53.002 -> *WM: freeing allocated params!
11:33:53.002 -> WiFi configuration complete
11:33:53.002 -> Local IP: 192.168.0.65
11:33:53.002 -> URL to send commands: http://irblaster.local:8080
11:33:53.002 -> ArduinoOTA started
11:33:53.002 -> MDNS http service added. Hostname is set to irblaster.local:8080
11:33:53.002 -> HTTP Server started on port 8080
11:33:53.002 -> Starting UDP
11:33:53.036 -> Local port: 8888
11:33:53.036 -> Waiting for sync
11:33:53.036 -> Transmit NTP Request
11:33:53.036 -> time.google.com: 0.0.0.0
No NTP Response :-(
11:33:54.505 -> No errors detected with security configuration
11:33:54.505 -> Epoch time from timeServer is unexpectedly old, probably failed connection to the time server. Check your network settings
11:33:54.540 -> 25
11:33:54.540 -> Invalid EPOCH time, security checks may fail if unable to sync with NTP server
11:33:54.540 -> Ready to send and receive IR signals
Turning off the LED to save power.
```
If I reset the esp8266 without GPIO10 set to ground it just boots back into the WiFi config mode.
```
11:40:46.922 -> ESP8266 IR Controller
11:40:46.922 -> Config pin GPIO10 set to: 1
11:40:46.922 -> mounted file system
11:40:46.922 -> reading config file
11:40:46.922 -> opened config file
11:40:46.922 -> {"hostname":"irblaster","passcode":"pass","port_str":"8080","user_id":"xxx","ip":"192.168.0.65","gw":"192.168.0.1","sn":"255.255.255.0"}
11:40:46.922 -> parsed json
11:40:46.922 -> *WM: Adding parameter
11:40:46.922 -> *WM: hostname
11:40:46.922 -> *WM: Adding parameter
11:40:46.922 -> *WM: passcode
11:40:46.922 -> *WM: Adding parameter
11:40:46.922 -> *WM: port_str
11:40:46.922 -> *WM: Adding parameter
11:40:46.922 -> *WM: user_id
11:40:46.922 -> Using Static IP
11:40:46.922 -> *WM: 
11:40:46.922 -> *WM: AutoConnect
11:40:46.922 -> *WM: Connecting as wifi client...
11:40:46.922 -> *WM: Custom STA IP/GW/Subnet
11:40:46.922 -> *WM: 192.168.0.65
11:40:46.922 -> *WM: Using last saved values, should be faster
11:40:46.922 -> *WM: Connection result: 
11:40:46.922 -> *WM: 0
11:40:46.922 -> Entered config mode
11:40:46.922 -> 192.168.4.1
11:40:46.922 -> IR Controller Configuration
11:40:46.922 -> *WM: 
11:40:46.922 -> *WM: Configuring access point... 
11:40:46.922 -> *WM: IR Controller Configuration
11:40:46.922 -> *WM: AP IP address: 
11:40:46.922 -> *WM: 192.168.4.1
11:40:46.922 -> *WM: HTTP server started
```
I've re-flashed and gone through the config multiple times ensuring everything is correct but have the same issues. I've even put in the wrong wifi password to see if it does anything different, but nope it says it is connected.

mdhiggins, 2018-10-27:
Can you post your build settings?
What version of the ESP8266 core are you using?
What version of lwIP are you building against?
What board do you have selected?

nadnerb33, 2018-10-28:
My settings are:
[image]
Originally, I left things on default which was IwIP set to `v2 Lower Memory,` but since changing to `v1.4 Higher Bandwidth` I have managed to get a bit further and access the webserver via my WiFi. The serial output reports my external IP correctly meaning it did manage to connect to my WiFi and the outside internet. I managed to load the blue header of the IR Controller webpage but the rest of the page didn't load.

After resetting the board a few times I noticed I can ping it for a short while but it consistently stops after 10/11 pings and is then no longer reachable. This happens even if I just reset the board and don't even attempt to load the webpage.
[image]

My first thought was a power issue since I was still powering it via a USB cable from my Macbook (which I assume can easily power it). But I've since tried running via USB from a 2A phone charger and also via a 5v source on a breadboard - all display the same issue.
I have 2 boards and both behave in the same way.

I'm new to ESP8266/Arduino so may be missing something really obvious!
Thanks

mdhiggins, 2018-10-28:
You need to downgrade to 2.4.0 for the ESP8266 version, you’re probably using 2.4.2 or 2.4.1 which has some kind of networking memory leak issue with WiFiManager

nadnerb33, 2018-10-29:
Thanks, I was indeed on 2.4.2 - I just installed the latest from [https://github.com/esp8266/Arduino#](https://github.com/esp8266/Arduino#) so have downgraded and it now works. 
Thanks again

closed 2018-10-29 by nadnerb33