mdhiggins/ESP8266-HTTP-IR-Blaster #227: strange hitachi tv remote behaviour

Jmilesy, 2018-11-02:
When i "read" the code from the Hitachi Tv remote i seem to get different codes from the power button:

12:40:09 | C60BDB3D | UNKNOWN | 4 | 0x0
12:40:09 | 84071B38 | UNKNOWN | 6 | 0x0
12:40:04 | D3C1CEA0 | UNKNOWN | 11 | 0x0
12:40:03 | FBCBF994 | UNKNOWN | 11 | 0x0
12:39:57 | 6D58FCC7 | UNKNOWN | 9 | 0x0
12:41:52 | 84C | RC5 | 12 | 0x1
12:41:51 | 4C | RC5 | 12 | 0x1
12:41:50 | 4C | RC5 | 12 | 0x1
12:41:50 | 84C | RC5 | 12 | 0x1
12:42:19 | 96C7569F | UNKNOWN | 11 | 0x0
12:42:18 | 81C65657 | UNKNOWN | 12 | 0x0
12:42:18 | E99DB8FE | UNKNOWN | 11 | 0x0
12:42:18 | 7361AF8F | UNKNOWN | 5 | 0x0
12:42:18 | 942574A7 | UNKNOWN | 11 | 0x0

these are from just pressing the power button, so unsure which code I should use.  Has anyone successfully set these up?

mdhiggins, 2018-11-10:
Not sure, looks like either a strange encoding or a very poorly received signal. Do other remotes work? Can you try scanning the remote at close range in a dark environment?

Jmilesy, 2018-11-10:
All other remotes have worked fine, its just this one, which annoyingly is my main TV.

its Hitachi, so am surprised it does not use a standard protocol.

mdhiggins, 2018-11-10:
Hitachi actually uses fairly complex proprietary protocols, I would suggest heading over to https://github.com/markszabo/IRremoteESP8266 since that's where the IRRemote library is developed and see if they can help. I would also suggest trying to update your ESP8266 library as it looks like he's been working on improving hitachi support with updates as recently as 6 days ago

mdhiggins, 2018-11-10:
Another option would be to check out some of the discrete codes for Hitachi TVs and see if you can send them as pronto codes

http://www.remotecentral.com/cgi-bin/codes/hitachi/tv/

Jmilesy, 2018-11-10:
Brilliant thank you, i did look at the discrete codes, but was not sure how to use them with the blaster

mdhiggins, 2018-11-10:
Try this for power

```
http://ip:port/json?plain=[{type:'pronto',data:[0x0000, 0x006B, 0x0000, 0x002A, 0x0000, 0x00AC, 0x0000, 0x0015, 0x0000, 0x0015, 0x0000, 0x0015, 0x0000, 0x0015, 0x0000, 0x0040, 0x0000, 0x0015, 0x0000, 0x0040, 0x0000, 0x0015, 0x0000, 0x0040, 0x0000, 0x0040, 0x0000, 0x0040, 0x0000, 0x0040, 0x0000, 0x0015, 0x0000, 0x0040, 0x0000, 0x0015, 0x0000, 0x0040, 0x0000, 0x0040, 0x0000, 0x0040, 0x0000, 0x0040, 0x0000, 0x0015, 0x0000, 0x0040, 0x0000, 0x0015, 0x0000, 0x0015, 0x0000, 0x0015, 0x0000, 0x0015, 0x0000, 0x0015, 0x0000, 0x0015, 0x0000, 0x0040, 0x0000, 0x0015, 0x0000, 0x0040, 0x0000, 0x0040, 0x0000, 0x0040, 0x0000, 0x05F1, 0x0000, 0x0056, 0x0000, 0x0E4F, 0x0000, 0x0056, 0x0000, 0x0E4F, 0x0000, 0x0056, 0x0000, 0x0E4F, 0x0000, 0x0056, 0x0000, 0x12F7]}]
```

closed 2019-01-03 by mdhiggins