mdhiggins/ESP8266-HTTP-IR-Blaster #228: Alexa skill in Italy

teohell, 2018-11-04:
Hi Michael, 
Alexa has finally arrived in Italy!
Would you please render the skill available in italy as well, I don't care if it is in english language, as I am using it already in that way, but now that I have converted the account the former skill does not work anymore
thanks

teohell, 2018-11-06:
Here we go:

> IR Controller Smart Home Skill that allows registration of your TV / other infrared smart home controller devices with Alexa. Your http URLs are translated into natural language smart home operations.
> 
> Register your IR Controller smart devices at https://tehpsyc.pythonanywhere.com/
> 
> Skill is designed to work with the https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster schematic and software. It may be possible to use similar HTTP enabled devices as well. The IR Blaster can be easily assembled with parts under $10 and provides full home theater control.

Skill per Casa Intelligente che permette l'utilizzo del tuo Controller Infrarosso Multicanale basato su ESP8266 con Alexa per comandare tutti i dispositivi IR della casa. I comandi impostati come URL http verranno tradotti in operazioni intelligenti basati su comandi vocali.

Registra il tuo IR Controller all'indirizzo https://tehpsyc.pythonanywhere.com/

La Skill è progettata per funzionare con gli schemi, il software e le istruzioni fornite all'indirizzo https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster
E' possibile utilizzare in alternativa altri dispositivi basati su comanid HTPP.
Il dispositivo supportato e descritto nel link github  può essere facilmente assemblato e programmato con componenti del valore complessivo di circa 10€, e permette di controllare qualsiasi dispositivo IR presente nell'impianto Home Theater di casa o con qualsiasi altro dispositvo che accetti comandi IR.

teohell, 2018-12-15:
Hey Michael, did you have the chanche to implement the skill to Italy? It's still not working for me 
Matteo

lamerjack, 2019-02-08:
Hi, Any progress on the italian skill?
Is it possible to install the english one?

AlexMun070282, 2019-05-20:
Hi, I' m also interested in the italian skill. I' m offering my support for the translation too. Please let us know. Thanks

raptusis, 2019-11-04:
Still not possible to have the skill in Italian? it is a pity!
This is a very useful project, and I'm sure many Italians would take advantage of it

xsiriusx, 2020-02-12:
It is sad that there is no answer to this request !!!

TheMCMan, 2020-02-12:
Just setup on IFTTT Alexa or Google Assistant webhook in Italian.

AlexMun070282, 2020-02-13:
Could you please explain more in detail the procedure, how to connect your http-IR-Blater using FTTT and then Alexa ?
The direct integraiton with Alexa cponatin all the commands, with webhoook we need to create command by command …
Is there any other way ?

As you can see there are few people that are interested in the italian version.
Maybe we can help with the translation ... It can be an idea ..

Just let us know

TheMCMan, 2020-02-13:
The ESP8266-HTTP-IR-Blaster project

Yes, I have step-by-step instructions for making IFTTT webhooks that work with both Alexa and Google Assistant.  Currently, I use more than 50 IFTTT webhooks that link my Alexa and the IRBlaster together.  Each webhook is a different voice phrase action.  This means my TV now responds to 50 different voice commands; such as Turn On, Pause, Play etc...

The basic idea is that Alexa sends a webpage post link to your IRBlaster.  Then the IRBlaster sends out the IR pulses to the TV.  If you can post a link to your IRBlaster, you can control virtually anything.

Using the mdhiggins setup :), you can view and send links from the browser interface.  This way you can tell if the IRBlaster is actually working before working on the voice part.  If you can get to this stage of the project, then you can build up your IFTTT webhooks.

1.  Goto IFTTT Development website called IFTTT Platform for Business
(it's free)
https://platform.ifttt.com/

2.  Goto Applet
Create a New Applet

3.  The Trigger
Trigger = Amazon Alexa or Google Assistant
Select Say a simple phrase from dropdown

4.  Field label What phrase? 
Visible = set by you
Value = tv on (Or replace with Italian equivalent)

******************************
Important -- Step 5 -- Language, is only available when using Google Assistant
******************************

5.  Field label Language 
Visible = set by you
Value = English

6.  The Action
Click Add Action Button
In Search for Service field enter Webhooks
From the dropdown field, select Make a Web Request

7.  Field label URL
Visible = set by you
Value = http://192.168.1.250:667/msg?code=E0E040BF:SAMSUNG:32

8.  Field label Method
Visible = set by you
Value = Select Post from dropdown list

9.  Field label Content Type
Visible = set by you
Value = text/plain

10.  Field label Body
Visible = set by you
Value = leave blank

After you save the Applet scroll down a bit and turn it on.  Then press the connect button.  After you press the connect button you will be brought over to an Amazon login page so the webhook with be created under you Amazon profile. 

On a personal note:  I prefer the Google Assistant webhooks, because it uses more of a natural language.  For example, in Google Assistant I say, OK Google, Turn on My TV.  While for Alexa, I have to say Alexa, Trigger My TV On.  Alexa uses the word Trigger to tell it to use a webhook.  Google just knows it's a webhook.

lamerjack, 2020-04-03:
Thank you very much!
Interesting post. i'm doing some tests about it.
i have some question:
point 7 and 8.
The request looks more a GET than a POST. Is it correct?
point 7.
http://192.168.1.250:667/msg?code=E0E040BF:SAMSUNG:32
This is a local network address...
So will be Alexa or google that will send the command? Correct?

mdhiggins, 2020-04-03:
Any external source requires your external IP address, so Alexa and Google cannot use that as its a local IP address

giuliomoro, 2020-08-18:
@mdhiggins any update on having this available for installation on an Italian Alexa device? Is there anything further we could help with?

AlexMun070282, 2020-08-18:
I'm also available, please involve us to complete it.
We can help.
thanks

mdhiggins, 2020-08-18:
[translate.pdf](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/5090677/translate.pdf)

I need translation of all these elements and I can try to submit to Amazon for approval

mdhiggins, 2020-08-18:
And if anyone can do Spanish let me know, amazon has finally made submitting in other languages not a huge problem

AlexMun070282, 2020-08-18:
ok this is fine for me i can do the italian version. But from the pdf I
cannot do anything ...
can you give me something to work on ?
thanks

mdhiggins, 2020-08-18:
> IR Controller Smart Home Skill that allows registration of your TV / other infrared smart home controller devices with Alexa. Your http URLs are translated into natural language smart home [...]

AlexMun070282, 2020-08-18:
IR Controller Smart Home Skill permette l'utilizzo del tuo Controller
Infrarosso Multicanale basato su ESP8266 con Alexa per comandare tutti i
dispositivi IR della casa. I comandi impostati come URL http verranno
tradotti in operazioni intelligenti basati su comandi vocali.

Registra il tuo IR Controller all'indirizzo
https://tehpsyc.pythonanywhere.com/

La Skill è progettata per funzionare con gli schemi, il software e le
istruzioni fornite all'indirizzo
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster
E' possibile utilizzare in alternativa altri dispositivi basati su comandi
HTPP.
Il dispositivo supportato e descritto nel link github può essere facilmente
assemblato e programmato con componenti del valore complessivo di circa
10€, e permette di controllare qualsiasi dispositivo IR presente
nell'impianto Home Theater di casa o con qualsiasi altro dispositivo che
accetti comandi IR.

mdhiggins, 2020-08-18:
I'll also need all the sample commands in whatever Amazon uses for their API in Italian, I assume it would be a direct translation but that's required. Also the short description

giuliomoro, 2020-08-18:
> Alexa, turn on bedroom TV

-> Alexa accendi la televisione della camera

> Alexa, mute bedroom TV

-> Alexa, muta la televisione della camera

> Alexa, turn up the volume by 5 on bedroom TV

-> Alexa, alza il volume della televisione della camera di 5.
[though I am not 100% sure this is right, @AlexMun070282 ? ]

Short description:
> IR Controller Smart Home Skill

-> Controllo infrarossi Smart Home

AlexMun070282, 2020-08-18:
Actually you can leave as it is, I think most of the people that are
working with those stuff can understand.
If you want an italian translation for the short description, you can take
this: "Telecomando IR per casa intelligente"

Example phrases are:

1. Alexa, accendi la TV della camera
2. Alexa, metti in muto la TV della camera
3. Alexa, alza il volume di 5 sulla TV della camera

Let me know if you need other support,

Feel free to ping me.
regards Alessandro

mdhiggins, 2020-08-18:
Submitted for review we'll see what amazon says

giuliomoro, 2020-08-18:
amazing, thanks @mdhiggins

AlexMun070282, 2020-08-19:
thanks, @mdhiggins <https://github.com/mdhiggins>

mdhiggins, 2020-08-19:
Failed the first time, resubmitted

The process is very frustrating, for every language I need a local amazon account (so I've had to make about 7 amazon accounts in various languages). And for each local amazon account I need to set up forwarding to a test device that has to issue successful responses to the voice commands. Had a port issue with one of the accounts that caused it to fail.

Biggest problem now is that the person who helped me out by giving me an Amazon India account (you need an Indian phone number to make one) has changed their password so I can't update the test data. Tried a workaround but might fail the review again because of India. If even one region fails they fail the whole update.

giuliomoro, 2020-08-19:
Ah damn! I really appreciate the time you are taking to do this, let me know if I can help with anything (note: I am away from my Echo Dot until Tuesday).

AlexMun070282, 2020-08-19:
If you need to test it in Italian , I’m also available. Regards 

Inviato da iPhone

mdhiggins, 2020-08-19:
I don't test it, Amazon has to test it to approve it so they handle that

mdhiggins, 2020-08-25:
Looks like it got approved and should be live

giuliomoro, 2020-08-25:
YES ! great! thanks a lot. Will get to it soon.

giuliomoro, 2020-08-28:
tested it in Italian and it seems to work, except that several of the capabilities exposed by https://tehpsyc.pythonanywhere.com/ are not supported by Alexa in Italian (see [here](https://developer.amazon.com/en-US/docs/alexa/device-apis/list-of-interfaces.html)), which is very sad :( I guess I will have to abuse the scenes API to circumvent this limitation ...

mdhiggins, 2020-08-28:
Yeah nothing I can do about that, amazon has been expanding support though so it should automatically work when they add it

closed 2020-08-28 by mdhiggins