mdhiggins/ESP8266-HTTP-IR-Blaster #230: No send command

armacali, 2018-11-10:
My serial info of the wemos D1 Mini

{"hostname":"amcl_ir","passcode":"654321","port_str":"567","user_id":"amzn1.account.AHGXCFGYFDXEGMJUXXXXXXXX","ip":"192.168.10.175","gw":"192.168.10.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.10.175
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.10.175
WiFi connected! User chose hostname 'amcl_ir' passcode '654321' and port '567'
*WM: freeing allocated params!
WiFi configuration complete
Local IP: 192.168.10.175
URL to send commands: http://amcl_ir.local:567
ArduinoOTA started
MDNS http service added. Hostname is set to amcl_ir.local:567
HTTP Server started on port 567
Starting UDP
Local port: 8888
Waiting for sync
Transmit NTP Request
time.google.com: 216.239.35.12
Receive NTP Response
Turning off the LED to save power.
Error retrieving external IP
HTTP Code: -1
connection refused
External IP address request took 5073 ms
No errors detected with security configuration
EPOCH time obtained for security checks
Ready to send and receive IR signals
Connection received
External IP: 181.XXX.XXX.XX
External IP address request took 511 ms
Signal received:
One line: 15D943B8:UNKNOWN:34
uint16_t  rawData[67] = {4478,4524, 578,1678, 576,1680, 572,542, 540,578, 570,540, 572,544, 574,1680, 574,554, 564,1676, 580,1678, 578,536, 576,538, 574,1680, 548,586, 584,1650, 548,568, 574,1682, 544,1710, 572,544, 570,544, 544,1710, 548,1710, 574,538, 546,568, 522,594, 574,540, 604,1650, 576,1682, 572,544, 572,538, 546,1706, 578,1680, 604};  // UNKNOWN 15D943B8

Turning off the LED to save power.
Connection received
Connection received
Connection received - JSON
Failed security check, signature is too old
Server: 0
Local: 1541870154
MID: 
 
______________________________
does not send codes always the same message "Unauthorized, HMAC security authentication failed"

Help please

mdhiggins, 2018-11-10:
You turned on the advanced security authentication and then aren't providing any of the authentication data (alexa automatically adds this via the skill).

You either need to send the commands from Alexa or clear out your amazon user_id from the settings to turn the security features off (you can leave passcode)

closed 2018-11-10 by mdhiggins

armacali, 2018-11-13:
You turned on the advanced security authentication and then aren't providing any of the authentication data (alexa automatically adds this via the skill). (how do I turn it off)

You either need to send the commands from Alexa or clear out your amazon user_id from the settings to turn the security features off (you can leave passcode) (Which passcode)

mdhiggins, 2018-11-13:
Reset your wifi manager settings and leave the userid blank when you fill out that info
Reset pin can be set to ground to trigger a reset ( see readme )

armacali, 2018-11-13:
solved using Reset your wifi manager, thanks