mdhiggins/ESP8266-HTTP-IR-Blaster #231: IR receiver noise

AjJordy, 2018-12-13:
There is a lot of noise on output of IR receiver. Without me sending any signal. How can I fix it ? 

```
Turning off the LED to save power.
# Signal received:
One line: 4CB0FADD:UNKNOWN:3
uint16_t  rawData[5] = {296,408, 138,5934, 100};  // UNKNOWN
4CB0FADD

# Signal received:
One line: 24AE7D4C:UNKNOWN:3
uint16_t  rawData[5] = {200,5530, 170,1166, 416};  // UNKNOWN
24AE7D4C

Turning off the LED to save power.
# Signal received:
One line: 4995F1ED:UNKNOWN:3
uint16_t  rawData[6] = {178,6004, 138,14, 380,214 };  // UNKNOWN
4995F1ED

Turning off the LED to save power.
# Signal received:
One line: 4DB5DAAA:UNKNOWN:4
uint16_t  rawData[7] = {104,14784, 324,88, 314,6340, 104};  // UNKNOWN
4DB5DAAA

# Signal received:
One line: 4CB0FADF:UNKNOWN:3
uint16_t  rawData[5] = {168,72, 100,646, 232};  // UNKNOWN
4CB0FADF

Turning off the LED to save power.
# Signal received:
One line: 4CB0FADC:UNKNOWN:3
uint16_t  rawData[5] = {558,46, 206,76, 178};  // UNKNOWN
4CB0FADC

Turning off the LED to save power.
# Signal received:
One line: 9B9D692F:UNKNOWN:4
uint16_t  rawData[8] = {342,268, 142,410, 240,306, 330,114 };  // UNKNOWN
9B9D692F

Turning off the LED to save power.
# Signal received:
One line: 24AE7D4E:UNKNOWN:3
uint16_t  rawData[5] = {230,370, 258,198, 206};  // UNKNOWN
24AE7D4E

Turning off the LED to save power.
# Signal received:
One line: E2E95598:UNKNOWN:4
uint16_t  rawData[8] = {236,256, 140,14602, 160,28, 6252,108 };  // UNKNOWN
E2E95598

Turning off the LED to save power.
# Signal received:
One line: 8E1C1C16:UNKNOWN:4
uint16_t  rawData[7] = {106,72, 124,262, 196,14534, 254};  // UNKNOWN
8E1C1C16

Turning off the LED to save power.
# Signal received:
One line: 6FB84EC6:UNKNOWN:4
uint16_t  rawData[7] = {134,14772, 318,42, 240,780, 198};  // UNKNOWN
6FB84EC6
```

PicoMitchell, 2018-12-18:
From my experience with IR stuff so far, some noise is normal. With IRController in particular, my "solution" was to comment out the lines that make the board blink when a signal is received.

The IR noise is normal and nothing to worry about, I'm assuming you just don't want to see the blinking all the time like me.

mdhiggins, 2018-12-18:
This is exactly correct

closed 2018-12-18 by mdhiggins