mdhiggins/ESP8266-HTTP-IR-Blaster #232: error: 'DynamicJsonBuffer' does not name a type

jenswes, 2018-12-14:
Hi,
thanks for your work, sounds very promising :-) 

but when building the sketch i've this error:

Arduino: 1.8.8 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

IRController:54:1: error: 'DynamicJsonBuffer' does not name a type
IRController:55:27: error: 'jsonBuffer' was not declared in this scope
 JsonObject& deviceState = jsonBuffer.createObject();

also it would be nice if you could upload a bin file for flashing, for a test this could be enough

thanks,

jens

jenswes, 2018-12-14:
well, looks like more people have this:

"other workaround: rollback to a non-beta version of ArduinoJSON library, this is a known bug"

'ill try this

jenswes, 2018-12-14:
ok, that helps a lot :-) it builds fine now

mdhiggins, 2018-12-14:
Don’t use the beta of ArduinoJson and that will get better

closed 2018-12-16 by mdhiggins

MrityunjaiKumar, 2019-08-01:
Added specific ArduinoJson version in platform.ini. Created a pull request to merge.  

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/pull/260