mdhiggins/ESP8266-HTTP-IR-Blaster #233: Stalling after boot

PicoMitchell, 2018-12-18:
IRController used to work fine for me, but after updating to recent versions (including the current latest), I've run into a consistent issue.

When installing and booting IRController, it installs fine and initially functions. I can setup and load the page in the browser and send codes. But, after a few minutes or so, it seems to totally stop working. I'm assuming it's hung up some how. The page won't load and it doesn't receive commands from Alexa or via the browser.

If I reboot the ESP-12E (v2), it works fine for a few minutes and then the same thing happens.

Could you help with some troubleshooting to figure out what could be happening? It there some logging I could turn on to see where its getting hung if I'm monitoring it from my computer?

This is happening on two separate boards, both of which used to work fine with older versions of IRController. But, regrettably I didn't keep track of which new versions broke it. I would try to keep them up-to-date and didn't pay too close of attention since it all installed fine and would initially work. I would only notice later on that they stopped working and figured the next update may fix things. But, that seems to not be the case after some months. So, maybe I'm the only one with this issue? Seems odd that the same thing is happening on two separate boards though.

It's worth noting that I can't be positive if the trouble is because of the latest IRController or the latest ESP8266 Arduino libraries or some other library. I just know that some older installation used to work fine. If I get the time, I could download older and older versions to try to see which one works reliably to at least have some clue as to the changes that broke things for me.

Thanks for any help you can provide with this, I would love to have these two IR blasters working reliably again.

mdhiggins, 2018-12-18:
What build settings are you using?

Specifically what board, what Arduino ESP8266 Core version, and what lwIP settings

You need to be on:
Esp8266 2.4.0
LwIP v1.4 prebuilt

PicoMitchell, 2018-12-18:
Thanks for getting back so quickly!

These are the boards I'm using: https://www.aliexpress.com/item/2015-New-product-Wireless-module-NodeMcu-Lua-Wifi-Nodemcu-WIFI-Network-Development-Board-Based-ESP8266-High/32521100830.html?spm=a2g0s.9042311.0.0.27424c4dYjL5QN

I see now the latest IRController version has a custom PlatformIO configuration. I dropped a slightly modified source (hard coding configuration for each IR Blaster as well as commenting out light flashes) file into my previous PlatformIO project which was using default settings. This may be my problem!

I'll re-upload using the custom configuration shortly and let you know if that works properly.

PicoMitchell, 2018-12-18:
I had the board correctly set to "nodemcuv2" in my previous PlatformIO settings, but my platform was incorrectly the default latest version of "espressif8266" and I did not have the lwip2 build flag, so that was also incorrectly using the default setting.

I've just re-uploaded using the custom configuration included in the latest project and everything seems so far so good. But, because of the kind of issue it's been, I'll see if things stay working after an hour or so to confirm that there isn't another issue going on.

Sorry if this was just my own oversight. Hopefully if that's all it was this may help someone in the future :-)

Thanks again for getting back to me so quickly!

mdhiggins, 2018-12-18:
Its all good. It’s a known bug caused by a memory leak with WiFiManager and newer versions of lwIP

From: Pico Mitchell
Sent: Tuesday, December 18, 2018 12:35 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Comment
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Stalling after boot (#233)

I had the board correctly set to "nodemcuv2" in my previous PlatformIO settings, but my platform was incorrectly the default latest version of "espressif8266" and I did not have the lwip2 build flag, so that was also incorrectly using the default setting.
I've just re-uploaded using the custom configuration included in the latest project and everything seems so far so good. But, because of the kind of issue it's been, I'll see if things stay working after an hour or so to confirm that there isn't another issue going on.
Sorry if this was just my own oversight. Hopefully if that's all it was this may help someone in the future :-)
Thanks again for getting back to me so quickly!

PicoMitchell, 2018-12-18:
My two IRControllers continue to function perfectly this morning. Closing this issue 👍

closed 2018-12-18 by PicoMitchell

mdhiggins, 2018-12-18:
Glad to hear it!