mdhiggins/ESP8266-HTTP-IR-Blaster #236: Geeklink IR Blaster - Cheap

arduino43, 2019-01-02:
I just reprogrammed one of these IR Blasters from AliExpress with ESP8266-HTTP-IR-Blaster. For anyone looking for a complete solution that will work with this code these are a great.

https://www.aliexpress.com/item/New-Geeklink-Smart-Home-WIFI-IR-4G-Universal-Intelligent-Remote-Controller-For-iOS-Android-Compatible-with/32958748327.html

Inside the unit is an ESP-12F with headers you can use to connect with either an Arduino or a USB-to-Serial adapter.

mdhiggins, 2019-01-03:
That's pretty cool, thanks for sharing

kaypohl, 2019-01-08:
Could you please share some pictures from the inside of the unit? And is there anything to change or only to flash the HTTP-IR-Blaster Firmware?

Kay

arduino43, 2019-01-08:
https://ibb.co/341KV5Y

No need to modify anything. 

How I did it:
USB to Serial (5v)

5v to board 5v pin
RX to board TX
TX to board RX
GND to board GND

Also note that you will need to place a jumper from the KEY to GND to put the ESP into flash mode when uploading.

Pin 5 is the IR Receiver
Pin 14 is the IR Transmitter(s)
Pin 12 is the red status LED on the front of the unit
Pin 13 is the orange L:ED on the front of the unit

mdhiggins, 2019-01-08:
[image]

Like  this? Not seeing where the 5v pin should connect on this image though

arduino43, 2019-01-08:
Like this....

https://ibb.co/smpnfZF

mdhiggins, 2019-01-08:
Ah gotcha, the existing micro USB port on there isn't usable?
Gonna order one today and mess around

arduino43, 2019-01-08:
Not for programming the board, only for powering the device.

benlye, 2019-01-08:
Ordered one as well :-)

benlye, 2019-01-08:
Adding the USB<->Serial wiring here for in-line reference:
[image]

kaypohl, 2019-01-08:
Ordered 2 :-) Thanks for sharing this information.

benlye, 2019-01-18:
Received mine today.  Wired it up and was eventually able to flash it.  

Tip that might save others some head-scratching - the mini USB cable that comes with it is NOT a data cable, so don't bother trying to use it with your CH340G!

arduino43, 2019-01-18:
Yeah I did mention that.

arduino43, 2019-01-18:
> No, not the USB port - I got that that wasn't for flashing, I meant the supplied cable doesn't carry data.

Ah ok yeah. I actually also ran into issues where other USB cables wouldn’t power the device. 

Wondering if the cable supplied has nonstandard wiring.

benlye, 2019-01-18:
> Wondering if the cable supplied has nonstandard wiring.

Certainly seems that way - mine only wants to power on with the supplied cable too.

arduino43, 2019-01-18:
Also, I found that the device would be come unresponsive at times.

I started using the MQTT branch and it worked much better.

benlye, 2019-01-18:
> Also, I found that the device would be come unresponsive at times. I started using the MQTT branch and it worked much better.

Good to know - I'll keep an eye on it.

mdhiggins, 2019-01-18:
MQTT branch? Link?

And the unresponsive stuff is probably a wifimanager issue make sure you’re using lwIP v1.4 and ESP8266 Core 2.4.0

arduino43, 2019-01-18:
I compiled it with 1.4 prebuilt but it would become unresponsive after 1-2 hours for some reason.

kaypohl, 2019-01-20:
> Also, I found that the device would be come unresponsive at times. I started using the MQTT branch and it worked much better.
> […](#)
> Sent from my iPhone

Can you please tell me more about the MQTT Branch?

Kay

arduino43, 2019-01-24:
If you have a multimeter you can easily find out - just follow the trace to the pin chip.

You could also upload a simple push button sketch using different pins and check when the pin is pulled high when pressed.

kaypohl, 2019-01-24:
I have a multimeter. But I can't follow the trace. I will try the option with the button sketch.

kaypohl, 2019-01-24:
@arduino43 do you still have problems that the device becomes unresponsive?

mdhiggins, 2019-01-24:
Just got my device and USB adapter so I'll be trying this out tomorrow. Keep me posted on any discoveries

kaypohl, 2019-01-24:
@arduino43 @mdhiggins  I've got the same problem. The device becomes unresponsive after a while. Mostly after sending commands with normal remote to my TV. When I check the webserver there are some weird readings from the IR receiver. Also the LED flashes without sending or receiving. Really strange.

arduino43, 2019-01-24:
Use pull request that added MQTT. I use MQTT but it also doesn’t crash using HTTP either.

kaypohl, 2019-01-25:
@arduino43 But you wrote that your device becomes also unresponsive after a time. What did you do that your device stay responsive either with Http?

arduino43, 2019-01-25:
It does not using MQTT pull

kaypohl, 2019-01-25:
Yes ok. But do you use still http without problems?

arduino43, 2019-01-25:
Yes

kaypohl, 2019-01-26:
> Yes

And what did you changed? You wrote that you had problems, and now your are still using http without problems?

Kay

kaypohl, 2019-01-27:
@mdhiggins I think the "freez" of the esp12 has to do with the ir receiver. After I've done this:

//irrecv.enableIRIn();

there was no freeze anymore. Of course I am not able to receive new codes.

Kay

benlye, 2019-01-28:
Mine is doing the same - dying after a random amount of time, less time if there is a lot of use of the actual remotes.

Don't know what others devices are doing, but mine always dies with the red LED on, which makes me wonder if it's something in the `loop()` that is going wrong.

kaypohl, 2019-01-29:
> Mine is doing the same - dying after a random amount of time, less time if there is a lot of use of the actual remotes.
> 
> Don't know what others devices are doing, but mine always dies with the red LED on, which makes me wonder if it's something in the `loop()` that is going wrong.

Mine is also dying with red led on. As a workaround I disabled the receiving of ir signals. Since then the device stays available.

benlye, 2019-01-30:
I seemed to get better stability by commenting out most of the lines which handle receiving IR transmissions in the `loop()` function - basically making it so that all that happens when another remote is used is the blaster's LED flashes.

I'm going to work through re-enabling each line to see if the stability problem comes back.

CodeFinder2, 2019-01-30:
Any tips on opening the device? Don't want to damage it... 🙈

kaypohl, 2019-01-31:
> I use the MQTT version and it never goes offline unexpectedly

Where can I find this MQTT version?

arduino43, 2019-01-31:
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/pull/197

benlye, 2019-01-31:
FYI, S1 pulls GPIO16 (the flash pin) low, so can be used to put the board into upload mode.

mdhiggins, 2019-02-01:
Please share

From: Ben Lye
Sent: Friday, February 1, 2019 6:02 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Michael Higgins; Mention
Subject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Geeklink IR Blaster - Cheap(#236)

I've created a tweak which puts the device into WiFi setup mode with a long press of the button, if anyone is interested.
I have it so that the orange LED comes on when button is pressed. If the button is held for 3s the LED blinks rapidly, then if the button continues to be held for 2s more the device launches WiFi setup mode.

benlye, 2019-02-01:
Here you go:
https://github.com/benlye/ESP8266-HTTP-IR-Blaster/tree/geeklink-config-button

I've found it's sometimes necessary to do an additional power cycle if the button isn't released quickly enough.

Regarding the stability, my device seems better with these two lines commented: https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/blob/master/src/IRController.ino#L1626-L1627

Need a bit more runtime to be conclusive, then I'll try to work out why.  I've looked at the MQTT PR and I don't see anything jumping out as to why that might behave differently.

CodeFinder2, 2019-03-03:
Has anyone verified that pin 14 actually uses **all** 7 IR LEDs on the board for sending?

A simple `void loop() { irsend.sendSony(0xa90, 12); }` for testing (whereby `irsend` is initialized to pin 14) **just causes one LED to blink** (the centered one) to blink (tested using my phone's camera) so I am concerned that we may need to use multiple pins in parallel? Any thoughts/comments on this?

arduino43, 2019-03-03:
Easy to test… I will take a look.

Im doubtful that only a single LED is being used.

CodeFinder2, 2019-03-06:
Any insights, @arduino43? 🙈

arduino43, 2019-03-06:
It fires on all LEDS.

vicwomg, 2019-03-10:
I spent the better part of yesterday getting this to work. 

This was not immediately clear to me since it was buried in the comments: you must change the receiver/transmitter/led pins, the main branch code doesn't work without modification. Here is the correct pin config:

```
// WEMOS users may need to adjust pins for compatability
const int pinr1 = 5;                                          // Receiving pin
const int pins1 = 14;                                         // Transmitting preset 1
const int pins2 = 14;                                         // Transmitting preset 2
const int pins3 = 14;                                         // Transmitting preset 3
const int pins4 = 14;                                         // Transmitting preset 4
const int configpin = 0;                                      // Reset Pin
const int ledpin = 12;                                        // Red LED
const int ledpin2 = 13;                                       // Orange LED
```

In the end, I used benlye's codeline here, which already contains above:
https://github.com/benlye/ESP8266-HTTP-IR-Blaster/blob/geeklink-config-button/src/IRController.ino

I also commented out lines 1692-1693 as recommended for stability. This did not affect the device's ability to receive codes.

```
 //   fullCode(&results);                                           // Print the singleline value
 //   dumpCode(&results);                                           // Output the results as source code
```
So far, so good. It has been running for a whole day without issue. The range and coverage is great. Much better than the 2-LED setup I had before, which was very directional and low range despite all the resistor values I tried. Also looks much less DIY.

CodeFinder2, 2019-03-16:
@arduino43 How did you make sure that all LEDs are used? I am asking because when I am looking at the LEDs (using my phone's camera) when the device sends an IR command, just the single LED in the middle is blinking. Additionally, it appears to me that reception rate of my devices (TV, AVR) is quite bad/low.

Nevertheless, are you controlling your LEDs by a *single* IRSend object initialized to pin 14 only? (Thus, I removed the non needed pickIRsend(), pin{2,3,4}, etc objects.)

That's what I am doing and maybe I am doing something wrong in the code. 🤔

CodeFinder2, 2019-03-16:
@vicwomg I think the PR with the MQTT stuff was created against the master branch using an old commit/version which was still using EasyNTPClient. So merging it against the current master branch creates errors. I had similar issues. ;-)

benlye, 2019-03-19:
> @arduino43 How did you make sure that all LEDs are used? I am asking because when I am looking at the LEDs (using my phone's camera) when the device sends an IR command, just the single LED in the [...]

I checked my device this morning using my mobile phone.  It is definitely blasting from multiple LEDs, using only pin 14 for sending.  If yours doesn't then maybe there is a hardware problem?

FWIW, since I commented out the [this line](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/blob/master/src/IRController.ino#L1627) the device has been perfectly stable:
`
dumpCode(&results);                                           // Output the results as source code
`

CodeFinder2, 2019-04-02:
@benlye Thank you very much for your reply and, in particular, for checking the IR LEDs of your device while sending.

Does yours look like this? This is what my device looks like when I add a `digitalWrite(14, HIGH);` to the `main()`function
[image]

Appears to me that just the IR LED in the middle is used. Also, when I add the enclosure back to the PCB, all my devices are basically unable to receive anything unless I place the IR Blaster directly (< 20cm) in front of them and rotate it so that to top of it face my TV's IR receiver. Really annoying and, thus, useless for me...

Would love to hear your feedback/opinion on this! :-)

By the way, I've no stability issues whatsoever even with the `fullCode()` and `dumpCode()` lines included. But this may also related to (or a positive side effect of) the major refactoring of the entire code base. However, I also haven't used the device on a daily base yet due to the above mentioned IR sending (range) issue.

stritti, 2019-04-15:
I just received a new - unchanged device: I checked, there all IR LEDs are used. Just to let you know.

CodeFinder2, 2019-04-15:
> I just received a new - unchanged device: I checked, there all IR LEDs are used. Just to let you know.

Thank you very much! :-) Would you mind extracting (and uploading) the original firmware from the device using (see, e. g., [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool#optional-backup-firmware)):
```esptool.py --port COM5 read_flash 0x00000 0x100000 image1M.bin```

whereby "COM5" is the appropriate serial port. I would like to test it on my device to see if it's software related. Thank you very much again!

stritti, 2019-04-15:
> Thank you very much! :-) Would you mind extracting (and uploading) the original firmware from the device using (see, e. g., [...]

I'll try. First of all I have to add connections wires :)

CodeFinder2, 2019-04-15:
Thank you so much! If you need help, just let me know. ;-)

stritti, 2019-04-15:
shit, I think I killed the 5V-Pin on the device. Is here a way to connect somewhere else?  The SONOFF S20 is quite easier to modify :/

CodeFinder2, 2019-04-15:
:-( 

Can you show me a picture of how your board looks like right now? I am not having the board at hand (I can take a look in the evening today when I am back home) but maybe there's another 5V pad/pin on the bottom of board (?).

stritti, 2019-04-15:
That was also my first try, but there is nothing on the downside.
It looks like following, but the 5V pad isn't there any more:
https://user-images.githubusercontent.com/23322579/50835547-de54ff80-134e-11e9-97f9-2bcf8051dbc5.jpg

CodeFinder2, 2019-04-15:
Puh, not good. I just had a look at the bottom and yes, there's nothing like that. Hm. If you have a soldering iron, you can try to add solder to the location where the pad was previously but I am also not sure if it will adhere there.

I have another idea but honestly, I don't know if that works or if it even destroys the board/ESP completely: you can use an old micro USB cable, cut the cable, strip the isolation and identify the 5V and GND wires (advisable to use a multi meter, just to be sure). Then you take your USB-to-serial adapter and connect 5V and GND to the 5V and GND wires of the cut USB cable respectively. Plug the cut micro USB cable into the board. Finally, you just need to solder a pin header to the TX, RX and GND holes on the board in order to connect them to your USB-to-serial adapter as well. Make sure that the GND pin (next to the RX/TX pins), the GND pin of your USB-to-serial adapter as well as the GND wire of your cut USB cable are all connected (same ground). After that, connect the USB-to-serial adapter to your PC.

Note that you can enter flash mode of the ESP either by shorting KEY and GND (e. g., with a jumper if you added an appropriate pin header like I did in my photo) or by keeping the button "S1" pressed during boot/power up. ;-)

But again, I don't know if that will actually work. If you try it, please don't forget to extract the original firmware.

Hope that is clear. :-)

2016for, 2019-05-06:
This is amazing! Do you know of any other devices that can be flashed?

Like Broadlink RM mini 3, or those Tuya thingies?

arduino43, 2019-05-20:
Judging by the pictures you could flash this device.

Just follow the wiring diagram for the GeekLink IR Blaster, its basically the same process (put the board into programming mode and flash the updated firmware)

CodeFinder2, 2019-05-27:
> @benlye Thank you very much for your reply and, in particular, for checking the IR LEDs of your device while sending.
> 
> [...]
> 
> Appears to me that just the IR LED in the middle is used. Also, when I add the enclosure back to the PCB, all my devices are basically unable to receive anything unless I place the IR Blaster directly (< 20cm) in front of them and rotate it so that the top of it faces my devices' IR receiver. Really annoying and, thus, useless for me...
> 
> Would love to hear your feedback/opinion on this! :-) [...]

Just for those who are interested: I can now confirm that my IR LEDs were broken (3 out of 7) and since they are controlled in series (2 in a row, except for the middle LED), all LEDs on the "outer ring" were off. I soldered new IR LEDs and now all work fine. Chinese quality ...

[3 more replies]

closed 2020-04-24 by mdhiggins