mdhiggins/ESP8266-HTTP-IR-Blaster #237: It just... doesn't work.

hukoro, 2019-01-04:
I wish I could give more information, but there are no error codes given. Code compiles fine. Config mode works fine. Set up all the WiFi information. It connects to my network, but after a couple of seconds, it seems like everything just locks up. Router says that the ESP8266 is still connected to the network, but the web server cannot be reached by IP address. I have the serial monitor open and I test the IR receiver. It recognizes that it is receiving codes for a couple of seconds, but once it seems to lock up, it no longer recognizes that it is receiving codes. The LED on the IR receiver I am using still lights up to indicate it is receiving a signal. I have had this same issue with two separate ESP8266 modules. I wish there was more information to give, but that is all I have. Here is a link to the [Pastebin](https://pastebin.com/41RSJwVm) of the serial log. It just ends when the program stops running and the LED on the ESP8266 is stuck on.

mdhiggins, 2019-01-04:
Know bug if you're not using the versions of things you're supposed to be. Later version of ESP8266 Arduino and lwIP introduced a memory leak that causes this

Make sure you're using ESP8266 2.4.0 and lwIP v1.4 prebuilt and this will fix itself

hukoro, 2019-01-04:
Sorry. I'm an idiot. Had the wrong core version installed. Closing. Thank you for the quick response.

closed 2019-01-04 by hukoro