mdhiggins/ESP8266-HTTP-IR-Blaster #242: DHCP support?

jdeltoft, 2019-01-16:
The youtube video is great, but it shows what looks like an old config screen that doesn't have the new entry fields for static ip, static gateway, static subnet.  Are those optional?  I deleted the default text hoping it would use dhcp but it didn't seem to work.  It always wanted to go to 10.0.1.10 which seems to NOT work for my LAN.

mdhiggins, 2019-01-16:
You shouldn't use DHCP since this device will require port forwarding, you're better off just manually setting an IP address in your supported range, doesn't need to be 10.0.1.10 though

If you want to turn DHCP on you should be able to remove ` wifiManager.setSTAStaticIPConfig(sip, sgw, ssn);` to have it default back. Again this isn't recommended though.

closed 2019-01-16 by mdhiggins

jdeltoft, 2019-01-16:
Ok, thanks, yea, I was going to update here that I removed those wifi manager lines and got dhcp working as I liked.  For me, using mdns is much more convenient to find the ip of this thing, and then I usually like to preserve the IP based on mac address.  Then I don't have to think about which ip's are free on the system (DHCP did that for me) and during initial setup I can just ping hostname.local to find it and test things out.  Just my personal preference.  Would be nice maybe if you make those fields blank in the IR configuration web page, that it reverts to DHCP.  That would have been more intuitive (and is what I originally tried).

Thanks for the project!  AWESOME work!  I love nodemcu's.  I can't thank you enough for sharing!