mdhiggins/ESP8266-HTTP-IR-Blaster #245: exec: "make": executable file not found in %PATH%

jsg56, 2019-01-31:
Got this error compiling the latest code for the IRBlaster.

jsg56, 2019-02-02:
The cause is unknown, but here are the files and versions from the Library that compiled and uploaded into the Nodemcu:

File                         | Version

IRremoteESP8266  | 2.5.4
ESP8266WiFi          | 1.0.0
WiFiManager          | 0.12.0
ESP8266mDNS        | 1.2.0
ArduinoJson            | 5.11.1
ESP8266WebServer | 1.0.0
ESP8266HTTPClient | 1.2.0
ArduinoOTA            | 1.0.0
CryptoSuite             | Not in IDE
Time                        | 1.4.0
Ticker                       | 1.0.0
NTP Client                | 3.1.0
Easy NTP Client        | 1.0.0

Now to see if it runs.....

mdhiggins, 2020-04-24:
Appears to be related to your ESP8266WiFi library, I would verify/reinstall/update that to see if this improves things

closed 2020-04-24 by mdhiggins