mdhiggins/ESP8266-HTTP-IR-Blaster #247: "Device did not support this" when using ChannelController

HerrPausL, 2019-02-20:
Hello,
thank you for this great work. :-)

I am using IR-Blaster with German-Alexa.

Power-Controller is working. :-)

But when i try to use the Channel Controller with command "Alexa, wechsel auf SatBox zu Kanal eins" (in english: "Alexa, change channel to 1 on SatBox" Alexa give me the awnser "SatBox unterstützt das nicht" (in english: "SatBox did not support this").

My Base URL is: http://1234.myfritz.net:8080/json?pass=geheim
Button 1 is: data:'C80521001',type:'RC6',length:36

I already removed the device and perform a new "device discovery" but still the same. :-(

Any ideas how to fix this?

BR Bernd

mdhiggins, 2019-02-20:
I can take a look at your JSON data if you want to email it to me, don't post it as it may contain access information but I'll check and see if anything looks malformed

HerrPausL, 2019-02-20:
Hi,
thank you very much - but i am confused.
What else shoud i send you:
> My Base URL is: http://1234.myfritz.net:8080/json?pass=geheim
Button 1 is: data:'C80521001',type:'RC6',length:36

BR Bernd

mdhiggins, 2019-02-20:
On the configuration page on tehpsyc.pythonanywhere.com, scroll to the bottom and open up the full JSON packet, and send that

HerrPausL, 2019-02-25:
Sorry for the late feedback :-(

With the help from Michael i was abel to get the IR Blaster to run.
- my Device name "SatBox" did not work with the german ChannelController - after change it to "Fernseher" is was ok.

Only problem left (but this is a issue of Alexa): German Channels like "Sat1", "Pro7", "Kabel1" did not work. I try different settings (like_ "Sat. 1", "Sat Eins", "Sat 1", ...) non did work properly.

Thanks Michael for his great Software and Support.

BR Bernd

AK-Homberger, 2019-10-08:
Hi,

is there an solution to channel 1 problem (German Alexa)?

I stll have this problem. All other channels are working. Only channel 1 results in "Devices does not support this.". I checked the IR codes. But these are similar for all channels.

Regards,
Andreas

mdhiggins, 2019-10-08:
Sadly this is an Alexa problem. If the command isn’t recognized it never makes it to my skill. You have to submit feedback to Amazon

AK-Homberger, 2019-10-08:
Hi Michael.
Thanks for your fast respone.
But the command is recognised by Alexa. But Alexa's response is: "This command does work for the  device". All other channels are working. There must be something special/different for the channel 1.

Regards,
Andreas

mdhiggins, 2019-10-08:
Yeah but it’s not actually recognized appropriately. It’s not recognized as a channel change command and never invokes my API. The smart home API is different than the regular skill API. I don’t get to define what people say that’s predefined and handled by amazon. I can only respond to a premade set of commands and if there’s a failure to recognize those commands my code is never invoked.

mdhiggins, 2019-10-08:
If you have the capacity to do so, using the Alexa skill simulator in German you could try and see what command Alexa is inappropriately trying to execute for channel 1. My German is non existent 

https://smart-home-system.org/en/test-alexa-skills/#Testing_with_the_utterances_profiler

AK-Homberger, 2019-10-08:
Thanks for the testing info.
But if you are right then channel one is not working for any TV skill for German Alexa.
I could not find so far a similar problem description for other TV skills. Many other problems, but not for channel one.

mdhiggins, 2019-10-08:
Honestly I would not be surprised if that was the case. There's a very small percentage of smart home skills that actually implement the channel controller API and I identified multiple bugs that I had to contact amazon to correct on their end before anything was working. I was one of the first skills to take advantage of it.

Its a problem with the way its translated into a command and it doesn't get recognized correctly and Alexa tries to send an unrelated/unexpected command. There's tons of bugs still, lots of inconsistency. Overall the smart home API isn't very well developed

closed 2020-04-24 by mdhiggins