mdhiggins/ESP8266-HTTP-IR-Blaster #248: Compatible Device "Orvibo CT10W"

HerrPausL, 2019-02-20:
Hello,
just a short Info about a compatible device.

This device is using the ESP8266 and looks realy nice: https://www.orvibo.com/en/product/xiaofang.html

The PIN-Out is:
//  0 = fw update
//  2 = ?
//  4 = button (input) 
//  5 = IR Rx (input)
// 12 = blue LED (output)
// 13 = ?
// 14 = IR Tx (output)
// 15 = red LED (output)

Picture of PCB:
[http://bplagens.bplaced.net/ESP8266/Top.PNG](url)
[http://bplagens.bplaced.net/ESP8266/Bottom.PNG](url)

My ArduinoIDE settings:
[http://bplagens.bplaced.net/ESP8266/ArduinoIDE.PNG](url)

At Aliexpress the price is aprox 16€.
At Amazon i payed ~30€

BR Bernd

CodeFinder2, 2019-04-25:
How did you open the device?

HerrPausL, 2019-04-26:
Hi,

after work i will made a picture for you how i did the job.
But basicly just a knife and soft-force.

BR Bernd

CodeFinder2, 2019-04-26:
Thank you so much! :-)

I already removed the base plate successfully. I am just unable to get the PCB out of the plastic case. Also, the switch is still covered by some "(elastic) plastic cover" and I don't know how to get it out of it without destroying the PCB and/or plastic cover/case...

HerrPausL, 2019-04-26:
Hi Adrian,

the "trick" is: do not remove the base-plate (i made the same mistake) you
have to lift the black-cube from the silver-base
At http://bplagens.bplaced.net/ESP8266/ i have add a few pictures how i
managed it to open the enclosure.

Be carefull - on the inner-side of the black-cube are some clips - two
where broken :-(
But this is not a big point - it still fits!

[image: image.png]
When the black-cube is removed - you can remove very easy the pcb - it just
clipped into the base of the silver lower part.

BR Bernd

CodeFinder2, 2019-04-27:
Thank you very much! I successfully managed to open it. However, my ESP does not seem to enter flash mode even though I connected "GP0" to "GND". I am not even sure, if my PCB has an ESP8266 since it differs from yours (nevertheless, its label says "orvibo 0w8266-02z"):
[image]
[image]

Even worse and for whatever reason, I am not able to reset the device anymore (by pressing the "S1" button at the bottom until the red LED blinks quickly). When I power up the device, after some time, the blue LED (sometimes?) lights up shortly but then remains off (forever?). Nothing seems damaged and according to my multimeter, the device seems to have power (measured 5 and 3.3V at the appropriate pads respectively). I've also tried to connect via serial monitor (115200 bauds) but nothing is printed. Also, no Wifi hotspot is created (anymore) and it does not connect to my router anymore. (I used the HomeMate app to test the device once.)

Strange: the LEDs do not light up when I power it (either using the micro USB port or via serial 3.3V interface). :-/

Any ideas what I can try? Initially, I intended to backup the original firmware using *esptool* but this seems impossible now and obviously, I am even unable to flash this repo's firmware... :-(

CodeFinder2, 2019-05-27:
I managed to repair my board. And I can confirm that it actually has an ESP8266. ;-)

For others: please note that the pin header holes have a contact spacing of 2mm (not 2.54mm) so would need something like [this](https://www.aliexpress.com/item/20-Pcs-Per-Lot-2mm-Pitch-2x3-Pin-6-Pin-Female-Double-Row-Straight-Pin-Header/32499983748.html?spm=2114.search0104.3.43.4a5a5b44nCDbl0&ws_ab_test=searchweb0_0,searchweb201602_10_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10843_10059_10884_10887_321_322_10103,searchweb201603_53,ppcSwitch_0&algo_expid=1bd82391-c717-451a-8a82-2050e03fa741-6&algo_pvid=1bd82391-c717-451a-8a82-2050e03fa741&transAbTest=ae803_3) (or its appropriate male counterpart).

closed 2020-04-24 by mdhiggins

cydia2020, 2020-06-20:
Thanks a lot for this, I did not know it was ESP8266 based.

mutantxxxx, 2020-11-25:
May I ask , I have flashed the standard tas build to Orvibo , but now want it to learn IR codes ,, a little unsure how to go about that .I use domoticz as the base home automation 
Probably better to put tas ir build on unit .just learning codes

hkrob, 2020-11-27:
@mutantxxxx  I am about to follow the same path and install Tasmota.
I presume 'learning' is a process of sniffing IR commands, I found "Receiving IR Commands" on https://tasmota.github.io/docs/Tasmota-IR/ which looks promising.

Would you mind sharing how you flashed Tasmota on your magic cube?

> May I ask , I have flashed the standard tas build to Orvibo , but now want it to learn IR codes ,, a little unsure how to go about that .I use domoticz as the base home automation
> Probably better to put tas ir build on unit .just learning codes

cydia2020, 2020-11-27:
See IRReceived and IRSend. I think "Learning" is basically sending what you've received using the IRSend command

Malakai-DB, 2022-03-08:
@HerrPausL @CodeFinder2 
I don't really manage to open the case (I already damaged it).
Should I try to lift the black-cube from the silver-base as shown in
1)
[image]
or
2)
[image]

If you have pictures of yours opened it will be most helpful (the link from this [post](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/248#issuecomment-487078940) gives a 404 error).

cydia2020, 2022-03-09:
You should separate the black cube from the silver base, as shown in picture 1

HerrPausL, 2022-03-09:
Sorry,
i did not managed to find the pictures right now.
This pictures where made quickly - i hope it will help you:
[image]

At all 4 sides you will have 2 lugs which will snap into the cover (and a little bit of glue in the middle). The best way is to use a sharp knive.
[image]

Malakai-DB, 2022-03-09:
Thank you both for your input. I managed to open it and flash Tasmota and now the fun begins integrating it in Home Assistant.

HerrPausL, 2022-10-11:
Hello,
sorry for the broken link!
After work I will try to find the old pictures.
If I'm honest, I can't remember exactly where it was to open - so I'll have
to look at the pictures as well.

BR Bernd

thomsunrd, 2024-04-04:
> I managed to repair my board. And I can confirm that it actually has an ESP8266. ;-)
>
> For others: please note that the pin header holes have a contact spacing of 2mm (not 2.54mm) so would need [...]

Hi @CodeFinder2 , how did you managed to repair the board, mine lits up with faint blue light. Any tips.

CodeFinder2, 2024-04-04:
> Hi @CodeFinder2 , how did you managed to repair the board, mine lits up with faint blue light. Any tips.

Sorry, this is too long ago, I don't really remember and I am not using the device anymore. 😐

sledim, 2025-05-03:
> Hello, just a short Info about a compatible device.
> 
> This device is using the ESP8266 and looks realy nice: https://www.orvibo.com/en/product/xiaofang.html
> 
> The PIN-Out is: // 0 = fw update // 2 = ? // 4 = button (input) // 5 = IR Rx (input) // 12 = blue LED (output) // 13 = ? // 14 = IR Tx (output) // 15 = red LED (output)
> 
> Picture of PCB: http://bplagens.bplaced.net/ESP8266/Top.PNG http://bplagens.bplaced.net/ESP8266/Bottom.PNG
> 
> My ArduinoIDE settings: http://bplagens.bplaced.net/ESP8266/ArduinoIDE.PNG
> 
> At Aliexpress the price is aprox 16€. At Amazon i payed ~30€
> 
> BR Bernd

Hi. Pls help flash this device for ir send and copy

sledim, 2025-05-03:
Usb, arduino?