mdhiggins/ESP8266-HTTP-IR-Blaster #252: No access point advertised

stredwizard, 2019-04-18:
Hi, having real problems getting this up and running on an ESP8266MOD, the sketch compiles and is loaded to the ESP fine, watching in the serial monitor it all appears normal but there is no access point advertised.

I did wonder if it was due to the limits of ESP module, but I flashed openmqtt gateway on it without a problem and works as expected.

This is using the 2.4.2 esp8266 Arduino core, and 2.0 IP variant, the 1.4 (prebuilt) is not an available as an option, if I use 2.4.0 core, I can select the 1.4 prebuilt variant, but the device has no output in the serial monitor and no access point either.

any ideas please

Tim3100, 2019-05-09:
>> Failed to mount FS?
You might be able to fix this issue by selecting 'Flash Size 4M (1M SPIFFS)'

bluetrepidation, 2019-05-30:
Same issue here but the file system is mounting....

`ESP8266 IR Controller
Config pin GPIO14 set to: 1
mounted file system
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.10
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 10.0.1.10
WiFi connected! User chose hostname '' passcode '' and port '80'
*WM: freeing allocated params!
WiFi configuration complete
Local IP: 10.0.1.10
URL to send commands: http://ESP_C0519D.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to ESP_C0519D.local:80
HTTP Server started on port 80
Starting UDP
Local port: 8888
Waiting for sync
Transmit NTP Request
time.google.com: (IP unset)
No NTP Response :-(
Error retrieving external IP
HTTP Code: -1
connection refused
External IP address request took 0 ms
Ready to send and receive IR signals
Turning off the LED to save power.
`

bgalport, 2019-06-14:
I had the same issue, I ended up clearing the default network values in the code on lines 50-52. It then went into config mode and brought up the access point. I also find that grounding GIO10 does not force the unit into config mode.

50 char static_ip[16] = ""; //"10.0.1.10";
51 char static_gw[16] = ""; //"10.0.1.1";
52 char static_sn[16] = ""; //"255.255.255.0";

bluetrepidation, 2019-06-15:
> I had the same issue, I ended up clearing the default network values in the code on lines 50-52. It then went into config mode and brought up the access point. I also find that grounding GPIO10 does not force the unit into config mode.
> 
> 50 char static_ip[16] = ""; //"10.0.1.10";
> 51 char static_gw[16] = ""; //"10.0.1.1";
> 52 char static_sn[16] = ""; //"255.255.255.0";

This got the wifi manager ap to pop up and configure for me. I'm also seeing GPIO10 not forcing the wifi manager into configuration mode. Thanks for the tip.

closed 2019-10-21 by stredwizard