mdhiggins/ESP8266-HTTP-IR-Blaster #257: Web server not responding

bgalport, 2019-06-13:
I'm having problems getting this fully working on my NodeMCU. I configured the unit via the web config page and from the serial log (below) it seems to be running (I can see IR messages on the console) but I cannot connect to the Web server using the the mDNS address or it's assigned IP of 192.168.10.38:667. I note from the log that the NTP request fails and that it does not respond to ping but I'm not sure if it should. The same device works fine on the local network with other code using the same (reserved) IP address. I saw the another issue reporting the same problem apparently related to static IP address usage but I'm using DHCP. Any suggestions?

Web related Library versions:
- ESP8266WebServer v1.0.0
- ESP8266Wifi v1.0.0
 
**Serial Console Log**
_ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"Irblaster","passcode":"Blast","port_str":"667","user_id":"","ip":"192.168.10.38","gw":"192.168.10.1","sn":"255.255.255.0"}
parsed json
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.10.38
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.10.38
WiFi connected! User chose hostname 'Irblaster' passcode 'Blast' and port '667'
*WM: freeing allocated params!
WiFi configuration complete
Local IP: 192.168.10.38
URL to send commands: http://Irblaster.local:667
ArduinoOTA started
MDNS http service added. Hostname is set to Irblaster.local:667
HTTP Server started on port 667
Starting UDP
Local port: 8888
Waiting for sync
Transmit NTP Request
time.google.com: (IP unset)
No NTP Response :-(
Ready to send and receive IR signals
Turning off the LED to save power._

bluetrepidation, 2019-06-15:
I hate the same issue. I can see the ESP connected to my access point but no response when trying to load the web page.

mosbror, 2019-06-20:
I had a similar problem. Try using the development version of WiFiManager. It's available on their GitHub page.
This may be bacause the WiFi.config function now has DNS as the second argument instead of Gateway, as it was in earlier versions. But I could be totally wrong. Never the less, the development version of WiFiManager works for me.

carranzafp, 2019-06-23:
Can you elaborate more how to install the wifimanager development?  do I have to uninstall the already installed library on Arduino library manager?  also I downloaded wifimanager master branch, is that the good one?   So far I have the board responding only on the first 5 seconds or so (I see serial log and entering on web interface) but after that time acts dead and I have to reset it

carranzafp, 2019-06-23:
It seems fixed, I had to force a full rebuild, I disabled mdns, querytime and query of the external ip as I will not use those, dont know if that helped, now its working fine (no hangs)

hgghyxo, 2019-06-25:
My webserver lasts 10-15 secs before it cannot be reached anymore. 
Cannot find out the reason for it, and none of the above helped

mosbror, 2019-06-25:
> My webserver lasts 10-15 secs before it cannot be reached anymore.
> Cannot find out the reason for it, and none of the above helped

I had this exact symptom. First I thought that there were a problem with the webserver, I later noticed the entire NodeMCU was unresponsive, not just the webserver.

Make sure to set "Erase flash" to "All Flash Contents" if you're uploading it through the Arduino IDE. Also you could try the development version on WiFiMangager, it's available on their github page.
[image]

hgghyxo, 2019-06-25:
As I found 0.14 is the latest WifiManager and I already using it. Earsing the whole memory didn't helped either...  
Also I set everything like yours, even the debug level 😅 
No luck...

hgghyxo, 2019-06-25:
Not sure if this, but when I reinstalled ESP8266 core platform version 2.4.0 instead of the prev installed 2.5.2.... now it does not froze... however, still cannot receive signals. But that problem is for another day 😅

crazy888, 2019-06-27:
I have the same problem.  Downgrading my ESP core to 1.4.2 does not change anything.
I can not connect to the webserver......

Anyone figure out whats the problem?

carranzafp, 2019-06-28:
Uninstall the wifimanager from the Arduino library manager and download manually the DEVELOPMENT version of WiFi manager (on the sketch folder it’s fine) then just rebuild

hgghyxo, 2019-06-28:
What is helped me is found under the Board manager, not in the 'manage libraries' section 
[image]

morganchristiansson, 2019-07-07:
Guys, you are amazing.

I was also having problems with my Wemos D1 mini freezing and hanging after a few seconds.

Reducing the delay() in the loop() made it a little better no idea why...

Downgrading esp8266 platform to 2.4.0 has resolved the freezes! woohoo it's alive!

morganchristiansson, 2019-07-07:
I have tried esp8266 platform 2.5.2, 2.4.2 and 2.4.1. 2.4.0 is the only one that doesn't hang.

mdhiggins, 2019-11-07:
Try the new dev branch
Requires WiFiManager developer build from Github, not the one in the Arduino library

closed 2019-11-07 by mdhiggins