mdhiggins/ESP8266-HTTP-IR-Blaster #258: Web Server Not Responding: Static IP issue

MrBlueWolves, 2019-06-22:
When I set up the wifi network, the page at 192.168.4.1 asks for a Static IP, subnet mask and gateway. This page has more fields than the one in the youtube tutorial. I have tried leaving those fields blank and assigning a static IP but when I monitor the serial port the static IP always ends up being set to 10.0.1.10 and I can't access the web server. My router uses 192.168.1.XXX. Is there a way to disable the static IP setting or to have it actually use the static IP that I assign it. Preferably I'd just like to disable it. I have tried this with lwIP 1.4 (Higher Bandwith) and 2.0 (Higher Bandwith) and 2.0 (Lower memory). When I use lwIP 2.0 I can still see it receive IR messages over the serial monitor. I am using Arduino IDE 1.8.9 and the sparkfun wifi board. https://www.sparkfun.com/products/retired/15000

[image]

morganchristiansson, 2019-07-07:
This was mentioned in another issue.

Set these variables to empty variables.

```
char static_ip[16] = "";
char static_gw[16] = "";
char static_sn[16] = "";
```

MrBlueWolves, 2019-07-07:
That didn't fix it. The set up page still has static IP etc. and will not get on my personal internet.

morganchristiansson, 2019-07-08:
Check out #257 I found lots of help there when trying to solve freezes of my WeMos D1 R2

mdhiggins, 2020-01-27:
This should be resolved on newer builds, especially if using the development branch of WiFi Manager

closed 2020-01-27 by mdhiggins