mdhiggins/ESP8266-HTTP-IR-Blaster #266: Reboot upon receiving IR code

constantinultius, 2019-10-04:
Board just reboots upon receiving any IR signal from my AC remote (Goodman).

Here is how it looks like in the console.   Is it still possible to receive and replay the signal?  

Signal received:
uint16_t  rawData[149] = {4422,4362, 570,1580, 568,506, 572,1578, 570,506, 570,502, 572,504, 568,506, 568,1582, 568,1582, 570,504, 570,504, 570,502, 570,1580, 572,504, 570,1578, 568,1582, 568,506, 568,1582, 568,506, 570,504, 570,504, 572,1578, 568,1580, 570,1580, 570,1582, 568,1580, 570,1582, 588,1558, 592,1558, 590,1560, 590,1558, 572,1582, 566,1582, 568,1582, 570,1578, 572,1580, 570,1580, 590,1560, 592,1556, 572,1580, 570,504, 570,1580, 568,508, 564,1582, 594,480, 570,506, 570,1578, 570,1580, 570,5176, 4426,4358, 590,484, 570,1580, 568,504, 572,1578, 570,1580, 572,1578, 570,1580, 570,502, 570,504, 570,1578, 570,1580, 570,1580, 570,504, 570,1580, 574,500, 570,504, 568,1580, 570,504, 568,1580, 570,1580, 568,1580, 570,506, 568,504, 570};  // UNKNOWN 
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12

All other signals works fine, here is example of my TV remote:
Signal received:
uint16_t  rawData[25] = {2450,550, 672,528, 674,524, 672,526, 674,526, 672,526, 676,522, 672,524, 1250,524, 674,526, 670,528, 670,526, 672};  // SONY 10
uint32_t  address = 0x1;
uint32_t  command = 0x0;
uint64_t  data = 0x10;

mdhiggins, 2019-10-04:
A lot of the AC units don’t work and overload the available memory

From: ConstantinUltius
Sent: Friday, October 4, 2019 10:20 AM
To: mdhiggins/ESP8266-HTTP-IR-Blaster
Cc: Subscribed
Subject: [mdhiggins/ESP8266-HTTP-IR-Blaster] Reboot upon receiving IR code(#266)

Board just reboots upon receiving any IR signal from my AC remote (Goodman).
Here is how it looks like in the console. Is it still possible to receive and replay the signal?
Signal received:
uint16_t rawData[149] = {4422,4362, 570,1580, 568,506, 572,1578, 570,506, 570,502, 572,504, 568,506, 568,1582, 568,1582, 570,504, 570,504, 570,502, 570,1580, 572,504, 570,1578, 568,1582, 568,506, 568,1582, 568,506, 570,504, 570,504, 572,1578, 568,1580, 570,1580, 570,1582, 568,1580, 570,1582, 588,1558, 592,1558, 590,1560, 590,1558, 572,1582, 566,1582, 568,1582, 570,1578, 572,1580, 570,1580, 590,1560, 592,1556, 572,1580, 570,504, 570,1580, 568,508, 564,1582, 594,480, 570,506, 570,1578, 570,1580, 570,5176, 4426,4358, 590,484, 570,1580, 568,504, 572,1578, 570,1580, 572,1578, 570,1580, 570,502, 570,504, 570,1578, 570,1580, 570,1580, 570,504, 570,1580, 574,500, 570,504, 568,1580, 570,504, 568,1580, 570,1580, 568,1580, 570,506, 568,504, 570}; // UNKNOWN
ets Jan 8 2013,rst cause:4, boot mode:(3,6)
wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
All other signals works fine, here is example of my TV remote:
Signal received:
uint16_t rawData[25] = {2450,550, 672,528, 674,524, 672,526, 674,526, 672,526, 676,522, 672,524, 1250,524, 674,526, 670,528, 670,526, 672}; // SONY 10
uint32_t address = 0x1;
uint32_t command = 0x0;
uint64_t data = 0x10;

ameoba32, 2019-10-04:
I am doubts about memory overflow, IR can't send so much data to overflow nodemcu memory.

I tried IRrecvDumpV2 example from IRremoteESP8266 library and it captures the message correctly.  Even decodes the content of message, wow.

Guess I have to dig deeper or to write own code to handle AC.

Library   : v2.6.6

Encoding  : MIDEA
Code      : A19348FFFF45 (48 bits)
Mesg Desc.: Power: On, Mode: 3 (Heat), Celsius: On, Temp: 25C/77F, Fan: 2 (Medium), Sleep: Off, Swing(V) Toggle: Off
Raw Timing[199]:
   +  4452, -  4332,    +   598, -  1554,    +   598, -   476,    +   598, -  1550, 
   +   596, -   478,    +   596, -   478,    +   596, -   478,    +   596, -   478, 
   +   598, -  1550,    +   596, -  1556,    +   596, -   478,    +   598, -   476, 
   +   598, -  1552,    +   600, -   474,    +   600, -   474,    +   598, -  1552, 
   +   598, -  1552,    +   598, -   476,    +   598, -  1550,    +   598, -   476, 
   +   598, -   478,    +   596, -  1554,    +   598, -   476,    +   596, -   478, 
   +   598, -   476,    +   596, -  1554,    +   598, -  1552,    +   598, -  1552, 
   +   596, -  1554,    +   596, -  1554,    +   598, -  1552,    +   596, -  1552, 
   +   600, -  1550,    +   596, -  1556,    +   596, -  1554,    +   596, -  1554, 
   +   596, -  1556,    +   596, -  1554,    +   598, -  1552,    +   598, -  1552, 
   +   596, -  1554,    +   596, -   480,    +   594, -  1556,    +   596, -   478, 
   +   594, -   480,    +   594, -   480,    +   590, -  1558,    +   594, -   480, 
   +   594, -  1556,    +   594, -  5152,    +  4424, -  4360,    +   596, -   480, 
   +   596, -  1554,    +   594, -   482,    +   590, -  1560,    +   594, -  1556, 
   +   592, -  1556,    +   594, -  1556,    +   594, -   480,    +   590, -   484, 
   +   590, -  1558,    +   592, -  1558,    +   594, -   480,    +   568, -  1582, 
   +   590, -  1560,    +   592, -   482,    +   590, -   484,    +   570, -  1580, 
   +   590, -   482,    +   572, -  1578,    +   572, -  1578,    +   570, -   504, 
   +   570, -  1580,    +   568, -  1580,    +   592, -  1558,    +   570, -   506, 
   +   568, -   506,    +   570, -   504,    +   570, -   504,    +   568, -   506, 
   +   570, -   504,    +   568, -   504,    +   572, -   504,    +   570, -   504, 
   +   568, -   506,    +   570, -   504,    +   570, -   504,    +   570, -   504, 
   +   568, -   506,    +   570, -   504,    +   568, -   506,    +   566, -  1582, 
   +   570, -   504,    +   570, -  1580,    +   570, -  1578,    +   570, -  1580, 
   +   568, -   506,    +   570, -  1580,    +   570, -   506,    +   568

uint16_t rawData[199] = {4452, 4332,  598, 1554,  598, 476,  598, 1550,  596, 478,  596, 478,  596, 478,  596, 478,  598, 1550,  596, 1556,  596, 478,  598, 476,  598, 1552,  600, 474,  600, 474,  598, 1552,  598, 1552,  598, 476,  598, 1550,  598, 476,  598, 478,  596, 1554,  598, 476,  596, 478,  598, 476,  596, 1554,  598, 1552,  598, 1552,  596, 1554,  596, 1554,  598, 1552,  596, 1552,  600, 1550,  596, 1556,  596, 1554,  596, 1554,  596, 1556,  596, 1554,  598, 1552,  598, 1552,  596, 1554,  596, 480,  594, 1556,  596, 478,  594, 480,  594, 480,  590, 1558,  594, 480,  594, 1556,  594, 5152,  4424, 4360,  596, 480,  596, 1554,  594, 482,  590, 1560,  594, 1556,  592, 1556,  594, 1556,  594, 480,  590, 484,  590, 1558,  592, 1558,  594, 480,  568, 1582,  590, 1560,  592, 482,  590, 484,  570, 1580,  590, 482,  572, 1578,  572, 1578,  570, 504,  570, 1580,  568, 1580,  592, 1558,  570, 506,  568, 506,  570, 504,  570, 504,  568, 506,  570, 504,  568, 504,  572, 504,  570, 504,  568, 506,  570, 504,  570, 504,  570, 504,  568, 506,  570, 504,  568, 506,  566, 1582,  570, 504,  570, 1580,  570, 1578,  570, 1580,  568, 506,  570, 1580,  570, 506,  568};  // MIDEA A19348FFFF45
uint64_t data = 0xA19348FFFF45;

ameoba32, 2019-10-04:
As solution I found "heatpumpir" library that can control AC and it is implemented as plugin to ESPEasy, that has WebUI to control it via http.

closed 2019-10-04 by constantinultius