mdhiggins/ESP8266-HTTP-IR-Blaster #267: Could you post the original code (in the video)?

notSoTechnical, 2019-11-06:
The current program doesn't compile...

mdhiggins, 2019-11-06:
Current code compiles fine, make sure you're using the right version of the libraries listed
And this is github, you can roll back to any version at any time

notSoTechnical, 2019-11-06:
Thanks for replying.  I'm very new to this and Github, so I'm not sure how to roll back to the older version... I made it compiled by deleted this #include "sha256.h" because I can't find this library.  It's doesn't work because the my computer can't locate the wifi hotpot.  I will run through the whole process again.  

#include <FS.h>                                               // This needs to be first, or it all crashes and burns

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <ESP8266WiFi.h>
#include <WiFiManager.h>                                      // https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <ESP8266mDNS.h>                                      // Useful to access to ESP by hostname.local

#include <ArduinoJson.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
#include <ArduinoOTA.h>
//#include "sha256.h"

#include <Ticker.h>                                           // For LED status
#include <TimeLib.h>

// User settings are below here

const bool getExternalIP = true;                              // Set to false to disable querying external IP

const bool getTime = true;                                    // Set to false to disable querying for the time
const int timeZone = -5;                                      // Timezone (-5 is EST)

const bool enableMDNSServices = true;                         // Use mDNS services, must be enabled for ArduinoOTA

const unsigned int captureBufSize = 150;                      // Size of the IR capture buffer.

// WEMOS users may need to adjust pins for compatability
const int pinr1 = 14;                                         // Receiving pin
const int pins1 = 4;                                          // Transmitting preset 1
const int pins2 = 5;                                          // Transmitting preset 2
const int pins3 = 12;                                         // Transmitting preset 3
const int pins4 = 13;                                         // Transmitting preset 4
const int configpin = 10;                                     // Reset Pin

// User settings are above here
const int ledpin = BUILTIN_LED;                               // Built in LED defined for WEMOS people
const char *wifi_config_name = "IR Controller Configuration";
const char serverName[] = "checkip.dyndns.org";
int port = 80;
char passcode[20] = "";
char host_name[20] = "";
char port_str[6] = "80";
char user_id[60] = "";
const char* fingerprint = "8D 83 C3 5F 0A 09 84 AE B0 64 39 23 8F 05 9E 4D 5E 08 60 06";

char static_ip[16] = "10.0.1.10";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";

DynamicJsonBuffer jsonBuffer;
JsonObject& deviceState = jsonBuffer.createObject();

ESP8266WebServer *server = NULL;
Ticker ticker;

bool shouldSaveConfig = false;                                // Flag for saving data
bool holdReceive = false;                                     // Flag to prevent IR receiving while transmitting

IRrecv irrecv(pinr1, captureBufSize);
IRsend irsend1(pins1);
IRsend irsend2(pins2);
IRsend irsend3(pins3);
IRsend irsend4(pins4);

[... 1526 lines cut ...]
}

void loop() {
  ArduinoOTA.handle();
  server->handleClient();
  decode_results  results;                                        // Somewhere to store the results

  if (irrecv.decode(&results) && !holdReceive) {                  // Grab an IR code
    Serial.println("Signal received:");
    fullCode(&results);                                           // Print the singleline value
    dumpCode(&results);                                           // Output the results as source code
    copyCode(last_recv_4, last_recv_5);                           // Pass
    copyCode(last_recv_3, last_recv_4);                           // Pass
    copyCode(last_recv_2, last_recv_3);                           // Pass
    copyCode(last_recv, last_recv_2);                             // Pass
    cvrtCode(last_recv, &results);                                // Store the results
    last_recv.timestamp = now();                                  // Set the new update time
    last_recv.valid = true;
    Serial.println("");                                           // Blank line between entries
    irrecv.resume();                                              // Prepare for the next value
    digitalWrite(ledpin, LOW);                                    // Turn on the LED for 0.5 seconds
    ticker.attach(0.5, disableLed);
  }
  delay(200);
}

mdhiggins, 2019-11-06:
Don't delete imports you'll break everything that depends on them and create a ton of problems. The script isn't broken your own environment is just missing things. Modifying the script is not the correct solution you need to fix your missing dependencies. They are there for a reason.

The sha256 library is on the video, rolling back your library isn't going to help and is part of the Cryposutie library which is linked directly in the readme

The library is here, just download and install it (its not on the arduino libraries interface, has to be manually downloaded):
https://github.com/jjssoftware/Cryptosuite

No need to post the entire source code. Post the error messages you're getting when you compile would probably be way more useful for debugging but only after you get all your dependencies in a row.

As a piece of general advice that is in no way meant to be rude, this is a complicated project not meant for someone who doesn't know what they're doing. You might be better off reviewing some basics of C++ and Arduino programming before you delve into this. Additionally, if you don't know what you're doing the last thing you should do is touch code you don't understand. The errors are coming from your software environment and dependencies being incorrect or incomplete and changing the code while it may eventually result in something that compiles will almost certainly result in something that doesn't function like its supposed to and will be impossible to update or support.

notSoTechnical, 2019-11-06:
Hi Thank you for trying to help.  I added the library and this is what I got:  

> Arduino: 1.8.9 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"
> 
> IRController:54:1: error: 'DynamicJsonDocument' does not name a type
>  DynamicJsonDocument deviceState(1024);
>  ^
> /Users/demorgan/Desktop/IRController/IRController.ino: In function 'String getUserID(String)':
> IRController:246:5: error: 'DynamicJsonDocument' was not declared in this scope
>      DynamicJsonDocument json(1024);
>      ^
> IRController:246:25: error: expected ';' before 'json'
>      DynamicJsonDocument json(1024);
>                          ^
> IRController:247:21: error: 'json' was not declared in this scope
>      deserializeJson(json, payload);
>                      ^
> IRController:247:34: error: 'deserializeJson' was not declared in this scope
>      deserializeJson(json, payload);
>                                   ^
> IRController:248:36: error: expected primary-expression before '>' token
>      uid = json["user_id"].as<String>();
>                                     ^
> IRController:248:38: error: expected primary-expression before ')' token
>      uid = json["user_id"].as<String>();
>                                       ^
> /Users/demorgan/Desktop/IRController/IRController.ino: In function 'bool setupWifi(bool)':
> IRController:391:9: error: 'DynamicJsonDocument' was not declared in this scope
>          DynamicJsonDocument json(1024);
>          ^
> IRController:391:29: error: expected ';' before 'json'
>          DynamicJsonDocument json(1024);
>                              ^
> IRController:392:9: error: 'DeserializationError' was not declared in this scope
>          DeserializationError error = deserializeJson(json, buf.get());
>          ^
> IRController:392:30: error: expected ';' before 'error'
>          DeserializationError error = deserializeJson(json, buf.get());
>                               ^
> IRController:393:23: error: 'json' was not declared in this scope
>          serializeJson(json, Serial);
>                        ^
> IRController:393:35: error: 'serializeJson' was not declared in this scope
>          serializeJson(json, Serial);
>                                    ^
> IRController:394:14: error: 'error' was not declared in this scope
>          if (!error) {
>               ^
> IRController:462:5: error: 'DynamicJsonDocument' was not declared in this scope
>      DynamicJsonDocument json(1024);
>      ^
> IRController:462:25: error: expected ';' before 'json'
>      DynamicJsonDocument json(1024);
>                          ^
> IRController:463:5: error: 'json' was not declared in this scope
>      json["hostname"] = host_name;
>      ^
> IRController:476:31: error: 'serializeJson' was not declared in this scope
>      serializeJson(json, Serial);
>                                ^
> /Users/demorgan/Desktop/IRController/IRController.ino: In lambda function:
> IRController:567:5: error: 'DynamicJsonDocument' was not declared in this scope
>      DynamicJsonDocument root(1024);
>      ^
> IRController:567:25: error: expected ';' before 'root'
>      DynamicJsonDocument root(1024);
>                          ^
> IRController:568:5: error: 'DeserializationError' was not declared in this scope
>      DeserializationError error = deserializeJson(root, server->arg("plain"));
>      ^
[... 33 lines cut ...]
> IRController:1275:33: error: 'kRawTick' was not declared in this scope
>        r += results->rawbuf[i] * kRawTick;
>                                  ^
> /Users/demorgan/Desktop/IRController/IRController.ino: In function 'void dumpRaw(decode_results*)':
> IRController:1324:39: error: 'kRawTick' was not declared in this scope
>      uint32_t x = results->rawbuf[i] * kRawTick;
>                                        ^
> /Users/demorgan/Desktop/IRController/IRController.ino: In function 'void dumpCode(decode_results*)':
> IRController:1357:39: error: 'kRawTick' was not declared in this scope
>      Serial.print(results->rawbuf[i] * kRawTick, DEC);
>                                        ^
> /Users/demorgan/Desktop/IRController/IRController.ino: In function 'void irblast(String, String, unsigned int, int, int, int, int, long int, IRsend)':
> IRController:1467:16: error: 'class IRsend' has no member named 'sendLutron'
>          irsend.sendLutron(data, len);
>                 ^
> Multiple libraries were found for "ArduinoOTA.h"
>  Used: /Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ArduinoOTA
>  Not used: /Users/demorgan/Documents/Arduino/libraries/ArduinoOTA
> exit status 1
> 'DynamicJsonDocument' does not name a type
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
>

mdhiggins, 2019-11-06:
All those errors are because the version of your script isn't compatible with ArduinoJson version 6
I actually just updated the github page to be version 6 compatible so if you upgrade your blueprint those errors should go away

mdhiggins, 2019-11-06:
Alternatively you could use version 5, though I'd recommend just migrated to the latest version which I finally had time to work on today

TheMCMan, 2019-11-06:
It all compiled nicely for me using version 5.  But I was using v2 of NodeMCU while the board in question appears to be v1.  I see there is even a v3 of NodeMCU out now for only $2.50 on AliExpress, including shipping.

mdhiggins, 2019-11-06:
Shouldn't make a huge difference other than maybe needing some subtle wiring tweaks or pin adjustments but the code should all be the same and compatible

v3 isn't an official one and is larger and lots of people don't like it

notSoTechnical, 2019-11-06:
Hi!  You are I changed it to version 6 and the json errors are gone.  However, I got these errors now: 
Any idea what did I do wrong? 

```
Arduino: 1.8.9 (Mac OS X), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
/Users/demorgan/Downloads/ESP8266-HTTP-IR-Blaster-master 2/src/IRController/IRController.ino: In function 'String encoding(decode_results*)':
IRController:1016:10: error: 'LUTRON' was not declared in this scope
     case LUTRON:       output = "LUTRON";             break;
          ^
/Users/demorgan/Downloads/ESP8266-HTTP-IR-Blaster-master 2/src/IRController/IRController.ino: In function 'void cvrtCode(Code&, decode_results*)':
IRController:1275:33: error: 'kRawTick' was not declared in this scope
       r += results->rawbuf[i] * kRawTick;
                                 ^
/Users/demorgan/Downloads/ESP8266-HTTP-IR-Blaster-master 2/src/IRController/IRController.ino: In function 'void dumpRaw(decode_results*)':
IRController:1324:39: error: 'kRawTick' was not declared in this scope
     uint32_t x = results->rawbuf[i] * kRawTick;
                                       ^
/Users/demorgan/Downloads/ESP8266-HTTP-IR-Blaster-master 2/src/IRController/IRController.ino: In function 'void dumpCode(decode_results*)':
IRController:1357:39: error: 'kRawTick' was not declared in this scope
     Serial.print(results->rawbuf[i] * kRawTick, DEC);
                                       ^
/Users/demorgan/Downloads/ESP8266-HTTP-IR-Blaster-master 2/src/IRController/IRController.ino: In function 'void irblast(String, String, unsigned int, int, int, int, int, long int, IRsend)':
IRController:1467:16: error: 'class IRsend' has no member named 'sendLutron'
         irsend.sendLutron(data, len);
                ^
Multiple libraries were found for "ArduinoOTA.h"
 Used: /Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ArduinoOTA
 Not used: /Users/demorgan/Documents/Arduino/libraries/ArduinoOTA
exit status 1
'LUTRON' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

TheMCMan, 2019-11-06:
On Oct 10, 2019 I successfully compiled and uploaded, to NodeMCU, the current sketch of the day.

The following steps were made to make this happen.

Arduino ESP8266MOD was the board.  AKA "ESP-12E CP2102 NodeMcu Lua V2.”  The listed CP2102 version represents the serial communications driver needed to access the project board

1. Install Arduino v1.8.9
2. Install Board Manager library ESP8266 board v2.5.2
3. IRremoteESP8266 v2.6.5 by David Conran, Sebastien Warin, Mark Szabo, Ken Shirriff
4. WiFiManager v0.15.0-beta by Tzapu
5. ArduinoJson v5.11.2 by Benoit Blanchon
6. Sha copied into libraries directory.  (Zip version Jan 31, 2018)
7. Time v1.5.0 by Michael Margolis

To enable the board to mount the file system, I had to change the board settings to "4M 
(1M SPIFFS)" before uploading.

For me to get the code working with an actual IR Remote, I had to comment out a line near the end of the ino file:  //  delay(200);

I also changed and commented out near the top of the ino file (as my network is 192.168.1.x):

char static_ip[16] = ""; //"10.0.1.10";
char static_gw[16] = ""; //"10.0.1.1";
char static_sn[16] = ""; //"255.255.255.0";

With the current update yesterday, Nov 5, 2091, I'm guessing the same versions will continue to work, with the exception of item 5. ArduinoJson v5.11.2.  This would need to be changed to v6.x as indicated by mhiggins. I have not tested this.

Hope this helps someone.

If someone could write out the steps needed to get this working with Google Nest mini, that would be fantastic lol

mdhiggins, 2019-11-06:
@notSoTechnical 
What version of the IRRemote library are you using? I think all the Lutron errors and the missing kRawTick are all things that were added after 2.5 so you must be predating that in your library

@TheMCMan 
You shouldn't need to comment out the IP address elements, they are only default values that are overwritten by the saved configuration file
I'm also surprised on ESP8266 core you're not getting a memory leak with that config
You do need at least 1M SPIFFS, higher should work fine too (that's where the config is stored)
There's a dev branch I just uploaded that uses the latest version of WiFi manager (the one that's still in development, not yet uploaded to the Library Manager yet, but is also compatible with the latest version of everything (Arduino 1.8.10, latest ESP8266 core, latest IRRemote, and latest ArduinoJSON) and has more flexible dynamic/static IP address options (you can go full dynamic or static with the option to define DNS)
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/tree/dev
You're welcome to test that out
With regards to Google Home I don't think their API has any TV specific functionality yet and I am not involved with much google platform stuff so it'll be up to someone else to develop that

notSoTechnical, 2019-11-06:
Hi I just updated the IRremote library to 2.7... but I'm getting new eorrs:
Arduino: 1.8.9 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
/Users/demorgan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/IRController.ino.cpp.o: in function `dumpCode(decode_results*)':
/Users/demorgan/Downloads/ESP8266-HTTP-IR-Blaster-master 2/src/IRController/IRController.ino:1069: undefined reference to `Sha256'
Multiple libraries were found for "ArduinoOTA.h"
 Used: /Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/ArduinoOTA
 Not used: /Users/demorgan/Documents/Arduino/libraries/ArduinoOTA
/Users/demorgan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /Users/demorgan/Downloads/ESP8266-HTTP-IR-Blaster-master 2/src/IRController/IRController.ino:1070: undefined reference to `Sha256Class::initHmac(unsigned char const*, int)'
/Users/demorgan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/IRController.ino.cpp.o:/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266/WString.h:295: undefined reference to `Sha256Class::resultHmac()'
/Users/demorgan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: sketch/IRController.ino.cpp.o: in function `validateHMAC(String, String, String, String)':
/Users/demorgan/Downloads/ESP8266-HTTP-IR-Blaster-master 2/src/IRController/IRController.ino:187: undefined reference to `Sha256Class::initHmac(unsigned char const*, int)'
/Users/demorgan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /Users/demorgan/Downloads/ESP8266-HTTP-IR-Blaster-master 2/src/IRController/IRController.ino:190: undefined reference to `Sha256Class::resultHmac()'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mdhiggins, 2019-11-06:
It looks like you did not install the Sha256/Cryptosuite library correctly
I would either explain what you did or try and verify that

notSoTechnical, 2019-11-06:
I added the cloned zip file (by Sketch -> Include library -> Add .zip library) from the website you told me https://github.com/jjssoftware/Cryptosuite

mdhiggins, 2019-11-06:
Make sure you directory structure is correct and you have all the files. Find your libraries folder and verify. Here is a screenshot of mine
[image]

notSoTechnical, 2019-11-06:
Hey, Mike, I go it to compiled now! Thank you so much!  I changed the folder name to Sha256 from Crytosuite-master

mdhiggins, 2019-11-06:
Great news!

notSoTechnical, 2019-11-06:
Hey Mike my wifi doesn't show ESP's hotspot.  

This is the output after uploading the code to the board: 
```
;l d⸮⸮| ⸮$⸮|⸮d⸮c|ǃ⸮⸮{⸮c{l l⸮⸮| ⸮d⸮|⸮d⸮c<ǃ⸮⸮{⸮c⸮b⸮⸮og⸮l'o⸮⸮⸮#p⸮⸮${l{$x⸮o⸮⸮$⸮⸮b'⸮|d⸮⸮b⸮⸮og⸮ l⸮⸮l`⸮g'l`or⸮⸮⸮nc⸮d{⸮⸮oc⸮l⸮⸮d$⸮⸮d`⸮⸮o⸮
ESP8266 IR Controller
Config pin GPIO10 set to: 1
failed to mount FS
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.10
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 10.0.1.10
WiFi connected! User chose hostname '' passcode '' and port '80'
WiFi configuration complete
Local IP: 10.0.1.10
URL to send commands: http://ESP_5DDA1F.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to ESP_5DDA1F.local:80
HTTP Server started on port 80
Starting UDP
Local port: 8888
Waiting for sync
Transmit NTP Request
time.google.com: (IP unset)
No NTP Response :-(
Error retrieving external IP
HTTP Code: -1
connection refused
External IP address request took 0 ms
Ready to send and receive IR signals
Turning off the LED to save power.

```

mdhiggins, 2019-11-06:
Looks like its not booting into configuration mode
I think this is a problem with the latest version of WiFiManager
If you connect GPIO10 to a ground connection while the device is turning on it should boot to hotspot mode
Notice at the top of the serial output where it says
`Config pin GPIO10 set to: 1`
When you're booting into config mode that will show a 0 instead of a 1

If you can get it set to 0 instead of 1 you might need to switch to the dev branch to get things fixed on the newer version of the libraries you're using

notSoTechnical, 2019-11-06:
I just grounded GPIO10 and this is what I got: 

```
Arduino: 1.8.9 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Sketch uses 494836 bytes (47%) of program storage space. Maximum is 1044464 bytes.
Global variables use 48084 bytes (58%) of dynamic memory, leaving 33836 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
2.6
esptool.py v2.6
Serial port /dev/cu.usbserial-1410
Connecting....
Traceback (most recent call last):
  File "/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/upload.py", line 25, in <module>
Chip is ESP8266EX
Features: WiFi
MAC: 2c:f4:32:5d:da:1f
Uploading stub...
    esptool.main(fakeargs)
  File "/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 2694, in main
Running stub...
Stub running...
Configuring flash size...
    detect_flash_size(esp, args)
  File "/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 2073, in detect_flash_size
    flash_id = esp.flash_id()
  File "/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 567, in flash_id
    return self.run_spiflash_command(SPIFLASH_RDID, b"", 24)
  File "/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 812, in run_spiflash_command
    old_spi_usr = self.read_reg(SPI_USR_REG)
  File "/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 475, in read_reg
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr))
  File "/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 332, in command
    p = self.read()
  File "/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 277, in read
    return next(self._slip_reader)
  File "/Users/demorgan/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool/esptool.py", line 1877, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
esptool.FatalError: Timed out waiting for packet header

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

TheMCMan, 2019-11-06:
It's the same failed to mount FS (File System) error I received.  Needs to change board setting to 1M SPIFFS

mdhiggins, 2019-11-06:
Yeah set the filesystem to at least 1M SPIFFS, though 2M or 3M should be fine too. Also, you only need to have it set to ground while rebooting, not reflashing

notSoTechnical, 2019-11-06:
Still not working after changing filesystem to 1M, 2M, 3M SPIFFS

notSoTechnical, 2019-11-06:
I'm still getting 
`Config pin GPIO10 set to: 1 ` after resetting + grounding GPIO10 before uploading

mdhiggins, 2019-11-06:
Then it’s not grounded during boot
Make sure you have the correct pin grounded

notSoTechnical, 2019-11-06:
Hi Mike.  Am I suppose to do the following for reboot?   

To get the ESP in program mode, GPIO0 must be low while booting:
1. press and hold the reset button.
2. press and hold the program button.
3. release the reset button, the ESP will boot in program mode.
4. release the program button.
5. upload the sketch.

mdhiggins, 2019-11-06:
Just upload the sketch how you normally would
Unplug or turn off power
Ground GPIO10
Turn device back on

Or if your board has a reset button like mine does, turn it on, ground GPIO10, press reset button

notSoTechnical, 2019-11-06:
Still not discovering the wifi hotspot... did you change the wifi name? 
[image]

mdhiggins, 2019-11-06:
Nope, name is the same. Does the serial output indicate its in AP mode? The light should be blinking on the device too

Post the serial output you've got with it grounded during boot

Also, as stated above, there's some issues with entering pairing mode on the beta version of WiFi manager that I've resolved in the dev branch of this project (that would require you to update to the latest version of WiFiManager on their GitHub) so it may not work until you do this, but you should at least verify the grounding aspect of things worked

notSoTechnical, 2019-11-06:
Output: 

```
rl d⸮⸮| ⸮l⸮|⸮$⸮c|⸮⸮⸮⸮;⸮c⸮c⸮⸮g'⸮lng⸮⸮⸮cx⸮⸮l;l{lp⸮'⸮⸮l⸮⸮co⸮<l⸮⸮c⸮⸮go⸮ d⸮⸮$`⸮ng$`'{⸮⸮⸮gc⸮ls⸮⸮'c⸮d⸮⸮ll⸮⸮l ⸮⸮n⸮
ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
*WM: Adding parameter
*WM: hostname
*WM: Adding parameter
*WM: passcode
*WM: Adding parameter
*WM: port_str
*WM: Adding parameter
*WM: user_id
Using Static IP
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.1.10
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 10.0.1.10
WiFi connected! User chose hostname '' passcode '' and port '80'
*WM: freeing allocated params!
WiFi configuration complete
Local IP: 10.0.1.10
URL to send commands: http://ESP_5DDA1F.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to ESP_5DDA1F.local:80
HTTP Server started on port 80
Starting UDP
Local port: 8888
Waiting for sync
Transmit NTP Request
time.google.com: (IP unset)
No NTP Response :-(
Error retrieving external IP
HTTP Code: -1
connection refused
External IP address request took 0 ms
Ready to send and receive IR signals
Turning off the LED to save power.

```

mdhiggins, 2019-11-06:
Still not grounded
What board are you using?

notSoTechnical, 2019-11-06:
The one you are using I think. 
[image]

mdhiggins, 2019-11-06:
Ah that's the problem, that's a LoLin V3. The GPIO10 pin can't be set to ground at boot unlike the official boards; other people have posted similar issues.

There are two potential fixes: one would be to change which pin is your reset pin in the blueprint which is defined around line 31
```
// WEMOS users may need to adjust pins for compatability
const int pinr1 = 14;                                         // Receiving pin
const int pins1 = 4;                                          // Transmitting preset 1
const int pins2 = 5;                                          // Transmitting preset 2
const int pins3 = 12;                                         // Transmitting preset 3
const int pins4 = 13;                                         // Transmitting preset 4
const int configpin = 10;                                     // Reset Pin
```

You can try to swap pin 3 or 4 with the configpin, like so
```
// WEMOS users may need to adjust pins for compatability
const int pinr1 = 14;                                         // Receiving pin
const int pins1 = 4;                                          // Transmitting preset 1
const int pins2 = 5;                                          // Transmitting preset 2
const int pins3 = 10;                                         // Transmitting preset 3
const int pins4 = 13;                                         // Transmitting preset 4
const int configpin = 12;                                     // Reset Pin
```

Though you may need to play around with that since I don't have that board. The other problem might be whatever the default state of that pin is, for example if the default state is 0 it will always try to reset. The reset basically just needs to be the opposite.

That is defined around line 519 or so, you can swap HIGH and LOW to be opposite
```
 if (!setupWifi(digitalRead(configpin) == LOW))
```

Alternatively you can just swap HIGH and LOW without changing pins, though then you'll need to refresh every time where as fixing the pin will allow you to reset the device without needing a computer to reflash

notSoTechnical, 2019-11-06:
Hey Mike!  I haven't tested with an IR sender yet, but I think I got the wifi hotspot available after I cloned from https://github.com/tzapu/WiFiManager/tree/development 

[image]

notSoTechnical, 2019-11-06:
Thank you so much!!!

notSoTechnical, 2019-11-06:
Nvm, It says the url website can't be reached... Does port 80 or port 667 matter?

mdhiggins, 2019-11-07:
The configuration portal will always be 192.168.4.1 and port 80. Once you configure it to connect to your in home WiFi it will be at whichever IP address you choose (and if you don’t know about setting of IP addresses they have to be similar to your router) and whichever port you specified. And obviously you have to connect back to your main router after configuration. 
My router for example is 10.0.1.1, I have my device set to 10.0.1.200, the gateway is 10.0.1.1, dns is 10.0.1.1, and subnet is 255.255.255.0
I set the port to 667
Then I access via 10.0.1.200:667

notSoTechnical, 2019-11-08:
Hi Mike... I still can't get it to work after swapping the pins or setting LOW to HIGH.  I'm out of ideas.  Any more suggestions?   Do I need to do something to the router?  like port forward 667 (I don't know what I'm talking about :(

TheMCMan, 2019-11-08:
Hi.  If you can't get it, I suggest getting a version 2 board with a CP2102 chip set.  V2 worked for me as posted earlier.

notSoTechnical, 2019-11-08:
I have v2 (I think).  See below:
[image]

But Arduino can't recognize the port for v2... Maybe it's broken?  I have a mac ... Did you need to install the CH340 driver?

TheMCMan, 2019-11-08:
Yes, that is a version 2 board.  I installed the cp2102 driver.  If you look at the chip next to the USB mini jack, this number is written on the chip.  It's your serial communications chip.  You might need a magnifying glass to see it lol.  That's why you need the cp2102 driver.  Sorry, I don't have a Mac.

notSoTechnical, 2019-11-08:
Hey McMan.  Thank you!  I'm able to receive the IR code now but I'm not getting the whole link http://ip:80//msh?code=123445: NEC32 

Any idea? 

[image]

TheMCMan, 2019-11-08:
Not sure.. kinda sleepy lol.

In the image, maybe you need to click on the time links under Received..

For me I get http://ip:667/msg?code=E0E040BF:SAMSUNG:32&pass=mypass

If that doesn't work, I can check in the morning and get back with you : )

mdhiggins, 2019-11-08:
@notSoTechnical Not sure what you mean by not getting the whole link

No spaces in the links first off

I notice it looks like you went with all the default IP address settings. If you don't use that same subnet (IE 10.0.1.xxx) you should probably reconfigure these options to be inline with your home network. The fact that you are able to access the page though means you might be using that subnet and are fine.

It does look like the device is failing to access external network resources, hence the external IP failing and your EPOCH time not working. That's only necessarily a problem if you intend to use the Alexa skill with it, though it may be a symptom of an underlying network configuration problem.

I would say please tell use what settings you've chosen for your network configuration and also if you could post what network settings your laptop has. Its totally safe to share your local IP address information, just don't share your external IP address

I'll post a screenshot of what I'm looking for

mdhiggins, 2019-11-08:
[image]

TheMCMan, 2019-11-08:
*****************
*****************

yeah, totally looks like a network issue.

*****************
*****************

[image]

*****************
*****************

The network error message is identified in the image above.
While the image below is how a properly working setup looks like.

*****************
*****************

[image]

*****************
*****************

**i have noticed that sometimes the arduino sketch does not upload properly.  sometimes it even looks like it uploaded correctly but the device is not actually available on the network.

You can check to see if the device (the ESP8266 device), is working correctly by using network scanner tools.  Whenever I upload the arduino sketch and set my IPs, I then scan my home network to see how the device is looking, or is behaving.

In the image below, I have uploaded the sketch to the ESP8266 and setup the website network settings.  But a network scan shows the device is somehow not working correctly.**

*****************
*****************

[image]

*****************
*****************

After re-uploading the sketch, a network scan shows the device is now working correctly.

*****************
*****************

[image]

*****************
*****************

The network scanner used was "Advanced IP Scanner

*****************
*****************

notSoTechnical, 2019-11-08:
Hey Mike.  Screenshot: 

[image]

mdhiggins, 2019-11-08:
I would configure your ESP8266 to be on the same 10.0.0.xxx instead of 10.0.1.xxx

Additionally your DNS and Gateway should both be 10.0.0.1 and not 10.0.1.1

TheMCMan, 2019-11-09:
Guess it was one step too many...

notSoTechnical, 2019-11-10:
No, I haven't had time to try it yet :( I have too many programs due this week and I suck at programming

notSoTechnical, 2019-11-12:
Hey @mdhiggins @TheMCMan 

I got it to work after reconfiguring the ESP, DNS, and Gateway!  Thank you!

TheMCMan, 2019-11-13:
Congratulations!  Now what do you do when your home IP address changes?

notSoTechnical, 2019-11-13:
Thanks!  Do I need to reconfigure then? Is there a way to make it static?

TheMCMan, 2019-11-13:
Well, that's where the project ends..  For me, I used what's known as a Dynamic DNS service on a Raspberry Pi because I don't leave my computer on 24/7.  Then I added security software to the Pi to protect the ESP8266.  But these are all optional of course.

mdhiggins, 2019-11-13:
Glad it’s working! Gonna close this out for now

closed 2019-11-13 by mdhiggins