mdhiggins/ESP8266-HTTP-IR-Blaster #268: pio compile issue, irremoteesp8266.h linker issue

tablatronix, 2019-11-13:
This is odd, I am having linker issues compiling. No other problems, its installed, its compiles on its own.. all other dependancies are found..
any ideas?

```
/Users/shawn/projects/microcontrollers/dev/libraries/ESP8266-HTTP-IR-Blaster/src/IRController.ino:3:29: fatal error: IRremoteESP8266.h: No such file or directory
```

```
$ pio lib show irremoteesp8266 
Looking for irremoteesp8266 library in registry
Found: https://platformio.org/lib/show/1089/IRremoteESP8266
IRremoteESP8266
===============
#ID: 1089
Send and receive infrared signals with multiple protocols (ESP8266/ESP32)

Version: 2.7.0, released Wed Oct 30 06:22:05 2019
Manifest: https://raw.githubusercontent.com/crankyoldgit/IRremoteESP8266/master/library.json
Repository: https://github.com/crankyoldgit/IRremoteESP8266.git

```

tablatronix, 2019-11-13:
```PHP
[Tue Nov 12 19:03:17 2019]
$ pio lib install 
Library Storage: /Users/shawn/projects/microcontrollers/dev/libraries/ESP8266-HTTP-IR-Blaster/.pio/libdeps/esp12e
LibraryManager: Installing id=1089
IRremoteESP8266 @ 2.7.0 is already installed
LibraryManager: Installing id=567
WifiManager @ 0.15.0-beta is already installed
LibraryManager: Installing id=64
Downloading...
Unpacking...
ArduinoJson @ 6.13.0 has been successfully installed!
LibraryManager: Installing id=44
Downloading...
Unpacking...
Time @ 1.5 has been successfully installed!
LibraryManager: Installing Cryptosuite
Cryptosuite @ 8d69121 is already installed

Library Storage: /Users/shawn/projects/microcontrollers/dev/libraries/ESP8266-HTTP-IR-Blaster/.pio/libdeps/sanguino_atmega1284p

[Tue Nov 12 20:02:48 2019]
```

TheMCMan, 2019-11-13:
I got something like this once.  I didn't have to run/compile settings correct.  They were set like I was using a Linux machine rather than a windows machine.  I'm not at my PC right now..

tablatronix, 2019-11-13:
it seems to be missing from my dependency graph..

```PHP
Scanning dependencies...
Dependency Graph
|-- <ESP8266-HTTP-IR-Blaster>
|   |-- <ESP8266WiFi> 1.0
|   |-- <Time> 1.5
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <WifiManager> 1.0.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <DNSServer> 1.1.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WebServer> 1.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 6.11.3
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.1
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 6.11.3
|-- <Time> 1.5
|-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <WifiManager> 1.0.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266HTTPClient> 1.1
|   |-- <ESP8266WiFi> 1.0
Building in release mode
```

tablatronix, 2019-11-13:
I got it compiled, I had to move my environments around, not sure why, but I removed platformio.ini and used my own and my own build path, and had to set `src_dir = ESP8266-HTTP-IR-Blaster/src` in it.

I also had to change include ticker.header to ticker.h

closed 2020-01-27 by mdhiggins