mdhiggins/ESP8266-HTTP-IR-Blaster #271: Blink LED based from URL command

modemmike, 2019-12-27:
I was wondering if I could code in a simple way to identify a until by sending a command to the board via URL. Maybe blink an LED? Anyone got a suggestion?

Example
http://192.168.8.144:8282/blink-10-sec

modemmike, 2019-12-27:
Was easier than I thought:

const int blinkpin = BUILTIN_LED;

  server->on("/blink-2min", []() {
    Serial.println("Blink request received");
    server->send(200, "text/plain", "blinking");
    uint32_t period = 2 * 60000L;       // 2 minute
    for( uint32_t tStart = millis();  (millis()-tStart) < period;  ){
      digitalWrite(blinkpin, HIGH);
        delay(50);
      digitalWrite(blinkpin, LOW);
        delay(50);
    }
    digitalWrite(blinkpin, HIGH);
  });

RienduPre, 2020-01-05:
Hi,
The url example in your first post I do understand, but how do I use the method you post in your last post?

I 'think' my unit does not sent any IR signals. It receives IR signals and the web interface shows that the IR signals are successfully sent, but it doesn't work

closed 2020-03-20 by mdhiggins