mdhiggins/ESP8266-HTTP-IR-Blaster #272: Issues with fresh installed ArduinoIDE, boards, libraries

rexxer777, 2020-01-10:
I reinstalled my Windows and had to reinstall everything.
After installing Arduino IDE, boards I couldn't compile IR-Blaster.
I collected and installed all the libraries and compiled it successfully.
But it doesn't work:
1. When I try to send json link it says JSON parsing failed (but the same link works on my previous device)
2. IR-diod doesn't send any signals. No signal on PIN? I tried GPIO4, GPIO5, different ir-diodes, transistors - nope.

mdhiggins, 2020-01-10:
The new version of Arduino JSON is more strict so you’ll need to make sure your json is perfect and has quotes everywhere there’s a string or it won’t work

As for number 2 I have no idea what IR-diod is

rexxer777, 2020-01-10:
I tried this one from IR-blaster page and it looks OK (tried to use double quotes and HTTP codes %27, %22):
http://192.168.30.71:80/json?plain=[{'data':[3396,1822,444,1248,446,298,462,284,468,286,468,1240,448,292,462,290,464,344,448,1246,446,294,460,292,462,290,464,1268,422,294,462,286,468,344,452,1242,448,292,462,288,466,286,466,1266,424,290,464,286,466,344,452,1240,450,290,464,288,466,286,466,1242,450,288,466,286,466,344,452,1240,452,284,466,288,464,288,466,1240,456,280,470,286,466,344,452,288,466,286,466,1240,454,282,468,286,466,1242,452,284,468,354,452,1240,454,1238,454,284,466,288,466,288,464,288,464,290,464,354,452,1240,454,1240,452,1238,454,1238,454,286,466,288,464,288,464,346,450,290,464,288,464,1244,450,286,466,288,464,288,464,288,464,346,450,290,464,288,464,288,464,288,464,288,464,288,464,290,464,346,452,290,464,288,464,288,464,288,466,288,464,288,464,290,464,346,452,288,464,288,464,288,464,288,464,288,464,288,464,290,464,346,450,290,464,288,464,288,464,288,464,290,464,290,462,290,464,344,452,1244,450,1240,452,1242,450,1240,452,284,466,288,464,290,464,1252,450],'type':'raw','khz':38}]

mdhiggins, 2020-01-10:
And does that give you the JSON error or just doesn't send a signal?

closed 2020-01-27 by mdhiggins