mdhiggins/ESP8266-HTTP-IR-Blaster #274: Device not accessible after wifi setup, and reset through setting GPIO to GND not working

KonTiki1957, 2020-03-06:
Hi, I was experimenting with your project in order to use it with home assistant after some tinkering (no Alexa) but as mentioned I was not able to do anything because the web interface port configured did not let me access it, I could only access the generic page on port 80. Moreover the reset procedure through GPIO 10 did not work and when I wanted to start over I had first to erase the flash and recompile

mdhiggins, 2020-03-06:
Might need to check or change your pin configuration then

Post some serial output and I can try and help

Include some basic information about your local WiFi IP configuration as well subnet, router IP, DNS). It’s a core feature that has been stable for a while so it’s most likely a setup or hardware specific issue. 

Also post your entire build parameters in Arduino and include lwIP version

KonTiki1957, 2020-03-07:
thank you for your answer.

you will find as attachment the terminal window from the beginning to the moment after wifi was set (the router in dhcp assigned then the ip 192.168.63.174 to the device and I chose port 5763), the port scan on the device stating that the port 5763 is not open and the image of the default apache2 page showing that the web server is working properly.

The device is a NodeMCU Lua Lolin module equipped with an ESP8266 12E but on the Arduino IDE I chose ESP generic

Any Hint would be appreciated, thanks

Federico

PS.: as previously stated the "reset" through GPIO 10 to ground connection doesn't work
[serial_monitor.pdf](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/4302380/serial_monitor.pdf)
[image]
[image]

mdhiggins, 2020-03-08:
Kind of looks like you haven't configured the IP address information correctly and that its using the default 10.0.1.10 which isn't on the same subnet as the rest of your network. Are you setting those things and to what values?

KonTiki1957, 2020-03-09:
Ok, thank you that was the problem. In fact I saw those settings whose meaning I did not understand but as I couldn't find out any information about them in the "user guide" and I don't know anything about Alexa I thought they might be correlated to it. So far so good :-) but I now have another problem you may be able to help me find a solution for: all the exercise was meant to be able to control a MITSUBISHI air conditioner which has a very long string of IR data ( I had to increase the "buffer size") but moreover the IR signal gets seen as "UNKNOWN"  (although it seems to arrive complete) and as such doesn't get sent afterward. Can you give me a hint about what path to follow to solve the problem. Btw as I saw that your sketch has the capability to discover "MITSUBISHI" I even tried to "force" the command inserting "MITSUBISHI" instead of "UNKNOWN" but it did not work. Thank you in advance.
Federico

mdhiggins, 2020-03-09:
I think you'll have to refer to the IRremoteESP8266 library https://github.com/crankyoldgit/IRremoteESP8266

Lots of these air conditioners have trouble with the library and won't work sadly but you might have some luck over there

closed 2020-03-20 by mdhiggins