mdhiggins/ESP8266-HTTP-IR-Blaster #275: Control of a Light works only a part

hummeruli, 2020-03-19:
I've a Light with a IR-Remote. I connect it with the Skill for the functions "ON" and "OFF". But there is no function for dim up/down, warm and cold white. I try to use the light for my 91 years old Mum. She is nearly blind and don't see the buttons on the remote well. I flashed the WiFi-IR-Box with Tuya firmware in it to the ESP8266-HTTP-IR-Blaster. It works to switch on end off. I can use vol up/down for dim but it's not good for the old lady.
I hope you can help.

Thanks a lot
Uli from Germany

PicoMitchell, 2020-04-26:
Are you familiar with coding? If so, there is a way you can make this work yourself...

Instead of using the IR Controller Device Setup (https://tehpsyc.pythonanywhere.com) to get this device into Alexa, you could use Samsung SmartThings with webCoRE. A Samsung SmartThings Hub IS NOT required to do what I'm suggesting.

In the SmartThings Groovy IDE (https://graph.api.smartthings.com) you can create a Simulated RGB Bulb (or Simulated RGBW Bulb). Then, after connecting webCoRE to SmartThings (https://wiki.webcore.co/#GitHub_Installation_.28preferred_method.29), you can write code in webCoRE (https://dashboard.webcore.co) to respond to events send to the Simulated RGB(W) Bulb to send the correct commands to your ESP8266-HTTP-IR-Blaster. It's a bit roundabout, but Simulated devices in SmartThings work perfectly with Alexa and webCoRE is super easy to write code in. It's all a bit intimidating at first to set it up, but it really isn't that bad once you get going.

I just did this myself to control the bias light (which has an IR remote) that I added onto my living room TV. I haven't implemented dimming yet on the bias light, but I'm sure it's possible to get something working with the triggers available in webCoRE. The only thing that would be weird about implementing dimming via IR by any means is that you wouldn't be able to reliably know the current brightness level so you wouldn't be able to tell Alexa to correctly set it to a specific brightness. But, being able to tell Alexa to go up or down in brightness should be trivial.

giuliomoro, 2020-08-18:
@PicoMitchell that sounds interesting, could you please share your webCoRE code ?

closed 2021-05-26 by mdhiggins