mdhiggins/ESP8266-HTTP-IR-Blaster #276: Unable to do port forwarding

nairsatheesh, 2020-04-02:
Hi Michael,

IM facing this issue for quite sometime i have put the code everything is fine but the ESP 8266 is not communicating with the Alexa
When im trying to use port 667 Im unable to open the Gateway 192.168.0.107 which i had assigned to esp8266
It is only working on Port 80
Any tutorial for  Port forwarding will be great
16:06:52.930 -> *WM: IP Address:
16:06:52.930 -> *WM: 192.168.0.107
16:06:52.930 -> WiFi connected! User chose hostname 'Irblaster' passcode 'Pass' and port '667'
16:06:52.930 -> *WM: freeing allocated params!
16:06:52.930 -> WiFi configuration complete
16:06:52.930 -> Local IP: 192.168.0.107
16:06:52.930 -> URL to send commands: http://Irblaster.local:667
16:06:52.930 -> ArduinoOTA started
16:06:52.930 -> MDNS http service added. Hostname is set to Irblaster.local:667
16:06:52.930 -> HTTP Server started on port 667
16:06:52.930 -> Starting UDP
16:06:52.930 -> Local port: 8888
16:06:52.930 -> Waiting for sync
16:06:52.930 -> Transmit NTP Request
16:06:52.977 -> time.google.com: 216.239.35.12
16:06:53.071 -> Receive NTP Response
16:06:53.352 -> External IP: 
16:06:53.352 -> External IP address request took 267 ms
16:06:53.352 -> No errors detected with security configuration
16:06:53.352 -> EPOCH time obtained for security checks
16:06:53.352 -> Ready to send and receive IR signals
16:06:54.899 -> Turning off the LED to save power.
With port 667 not able to open dashboard after adding the dashboard

nairsatheesh, 2020-04-02:
and one more thing i would like to add the ESP8266 server only opens up on port 80 i tried on 443 or 667 im unable to open anything on that and  serial monitor doens't give any error aswell

TheMCMan, 2020-04-02:
What is the make and model of your internet router?

nairsatheesh, 2020-04-03:
Thank you so much for replying
It's TPLINK WR841N

TheMCMan, 2020-04-03:
Hi there,. Looks like we have a bit of a time difference : ). I'm in California..

Here is a port forwarding instructional for your router : )

https://www.cfos.de/en-us/cfos-personal-net/port-forwarding/tp-link-tl-wr841n.htm

nairsatheesh, 2020-04-03:
let me check that today

nairsatheesh, 2020-04-03:
sure I will

nairsatheesh, 2020-04-03:
And thank you for helping me

nairsatheesh, 2020-04-03:
Few question before I try again
Should I use ESP8266 board version 2.4.0 ?or should the later version?
Im only able to access the 192.168.0.107 on port 80 If add any other port for eg 667,443 on the WIFI setup page it shows its connected on serial monitor however unable to open the gateway(192.168.0.107) sorry for bothering you:)

mdhiggins, 2020-04-03:
You are specifying the custom port when you try to access it correct? 80 is the default port so if you don't specify the port but you're using 80 everything works, if you switch to anything else, you have to include that in your address

TheMCMan, 2020-04-03:
so in chrome surf over to http://192.168.0.107:667 and see if it works..

TheMCMan, 2020-04-03:
Hey there nairsatheesh,  no bother at all.  no worries.. hope you get it working : )

The Board Manager Library Installs the ESP8266 Arduino Board Core:

As of Feb 11, 2020, most current version is ESP8266 board v2.6.3.  
Upgraded to this version on Feb 11.
Did not work after ESP8266 rebooted

Trying ESP8266 board v2.6.2.  Upgraded to this version on Feb 11.
Did not work after ESP8266 rebooted

Trying ESP8266 board v2.6.1.  Upgraded to this version on Feb 11.
Worked after rebooting ESP8266

Hope this helps..

nairsatheesh, 2020-04-03:
Hi michael,
something weird is happening her on the board in spite of flashing the raw code which you have put on the pagenow its not taking me to to the network configuration page. its not allowing to enter the wifi set up page now when i put 192.168.0.107 its taking me directly there 
so ideally it should have taken me to the WIFI setup page it is not happeing this issue started happening since today and one more new thing i would like to show you 
[image]

[image]
[image]

nairsatheesh, 2020-04-03:
[image]

mdhiggins, 2020-04-03:
Your settings are stored in the SPIFFS filesystem and are not erased by a reflash. To get back to the configuration page you need to set the config pin to ground which is outline in the readme

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster#force-wifi-reconfiguration

mdhiggins, 2020-04-03:
Since I can see you're on windows, open command prompt and post your `ipconfig` output too so we can get an idea if you have your network setup correct for your router

nairsatheesh, 2020-04-03:
> 
> 
> Since I can see you're on windows, open command prompt and post your `ipconfig` output too so we can get an idea if you have your network setup correct for your router

[image]

mdhiggins, 2020-04-03:
Alright yeah so that's good
192.168.0.107 is fine for your ESP8266
Gateway and DNS should be 192.168.0.1
Subnet should be 255.255.255.0

Port should be fine. Did you clarify if you were specifying the port in the address when trying to access it when it was set to something other than 80?

nairsatheesh, 2020-04-03:
> 
> 
> Alright yeah so that's good
> 192.168.0.107 is fine for your ESP8266
> Gateway and DNS should be 192.168.0.1
> Subnet should be 255.255.255.0
> 
> Port should be fine. Did you clarify if you were specifying the port in the address when trying to access it when it was set to something other than 80?

im trying to reflash it to wifi mode forgive me im still in learning phase to my understanding "Set GPIO10 to ground to force a WiFi configuration reset" meaning to connect jumper from GND to GPIO10Correct?
[image]

mdhiggins, 2020-04-03:
Exactly. It has to be set to ground on boot.

nairsatheesh, 2020-04-03:
Nothing is happening after reboot
So should I upload the code again
In this setup

mdhiggins, 2020-04-03:
Uploading code again won’t help and isn’t required. Some unofficial NodeMCU models don’t support setting GPIO10 to ground so you might need to modify the configPin in the blueprint to another pin. 

The LianLi V3 for example is one that doesn’t support GPIO10 setting to ground

nairsatheesh, 2020-04-03:
I flashed using a blank bin files and node mcu pyflasher gone to the
configuration mode
😊

nairsatheesh, 2020-04-03:
I have open the interface on port 667 success now the next thing is port
forwarding 😵
For which I will refer to ur explanation

nairsatheesh, 2020-04-04:
I have still doing port forwarding once you are awake just message me if you can help me step by step on my router
Im able to open the page on 192.168.0.107:667 on local network

mdhiggins, 2020-04-04:
You just need to access your router settings now and forward port 667 to 192.168.0.107

https://www.cfos.de/en-us/cfos-personal-net/port-forwarding/tp-link-tl-wr841n.htm

Service Port and Internal Port - 667
IP Address - 192.168.0.107
Protocol - TCP
Status - Enabled

nairsatheesh, 2020-04-04:
I have done that now how to test this if it's working or not

nairsatheesh, 2020-04-04:
I have got this following error on the alexa skill

mdhiggins, 2020-04-04:
You didn't post an error or give any information to allow us to help you

nairsatheesh, 2020-04-04:
sure i had attached the error give me sometime i will send it

nairsatheesh, 2020-04-04:
[image]

mdhiggins, 2020-04-04:
Now it’s probably something to do with your amazon userID not being set correctly, the time code not working correctly. One thing you can do is post the home page screenshot of the irblaster including the footer text most importantly (blur your external IP). The other would be to privately email me your json output at the bottom of tehpsyc.pythonanywhere.com and let me see if you’ve formatted that correctly.

nairsatheesh, 2020-04-04:
if you don't mind can you provide me ur email address I will send you that details 
as I will send you the details of the serail monitor output

nairsatheesh, 2020-04-04:
Hi Mike

here is the serial monitor details
00:59:58.996 -> WiFi connected! User chose hostname 'IRblaster' passcode 'pass' and port '667'
00:59:58.996 -> *WM: freeing allocated params!
00:59:58.996 -> WiFi configuration complete
00:59:59.030 -> Local IP: 192.168.0.107
00:59:59.030 -> URL to send commands: http://IRblaster.local:667
00:59:59.030 -> ArduinoOTA started
00:59:59.030 -> MDNS http service added. Hostname is set to IRblaster.local:667
00:59:59.030 -> HTTP Server started on port 667
00:59:59.030 -> Starting UDP
00:59:59.030 -> Local port: 8888
00:59:59.030 -> Waiting for sync
00:59:59.030 -> Transmit NTP Request
00:59:59.030 -> time.google.com: 216.239.35.8
00:59:59.167 -> Receive NTP Response
00:59:59.613 -> External IP: 
00:59:59.613 -> External IP address request took 461 ms
00:59:59.613 -> No errors detected with security configuration
00:59:59.613 -> EPOCH time obtained for security checks
00:59:59.648 -> Ready to send and receive IR signals
01:00:00.992 -> Turning off the LED to save power.
[image]

mdhiggins, 2020-04-04:
mdhiggins23 @ gmail.com

closed 2020-04-24 by mdhiggins