mdhiggins/ESP8266-HTTP-IR-Blaster #277: Web UI is not available

Redwid, 2020-04-13:
After a while my IR Blaster Web UI stop responding. I've set up to control blaster over home-assistant. So, I'm still able to send commands to blaster, but UI is not loading.
That happened second time during 2w -1 month perioud.
Last time I've resolved that by re-flashing everything.
Now that's happened again.
Looks like something like controller running of space or memory. Might be so many logs?
How could I find out what is wrong?

I'm using Melife ESP8266 NodeMCU CP2102 ESP-12E Board.

mdhiggins, 2020-04-13:
A lot of this I think has to do with wifi manager. For my personal use I've moved to the beta version of WiFi Manager (which you can get on their github) and the dev branch of this project and its been much more stable. The current version of wifi manager on the Arduino libraries is very old and has memory leak issues with anything beyond ESP 8266 2.4.0 and lwIP 2.0

Redwid, 2020-04-13:
Thanks @mdhiggins, if you provide a little how-to that will be great :) 
Sorry, I'm new on arduino programming/setup.

Another observation. If my router disconnects from internet, than board switches to wifi configuration mode, after connecting router back, board connects to wifi and Web UI is available again.

mdhiggins, 2020-04-13:
https://github.com/tzapu/WiFiManager/tree/development

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/tree/dev

Those are the two branches

Instead of installing WM from the arduino library interface just download from github and put that in your library folder

closed 2020-04-13 by mdhiggins