mdhiggins/ESP8266-HTTP-IR-Blaster #278: JSON Parsing Failed

jaymzx, 2020-04-24:
Hi, there! I'm glad I found this project as I'd like to get Alexa to control my Honeywell tower fan via IR. They don't seem to re-use Sony codes like a lot of appliances do, so I need to send the raw command. 
I'm attempting to use the URLs generated on the received code screen and they return 'Error! JSON parsing failed'.
Until I get this up and running, I left the password and Amazon user ID fields empty.
Here's an example URL string: 
`http://192.168.1.50:80/json?plain=[{'data':[1278,438,1252,436,410,1280,1250,442,1276,418,410,1284,408,1284,408,1286,408,1284,1252,440,410,1284,408,8768,1278,436,1254,438,410,1280,1250,440,1252,414,436,1284,408,1286,408,1286,408,1284,1252,440,410,1284,408,8768,1278,438,1252,436,412,1280,1250,440,1252,440,412,1284,408,1286,408,1284,410,1284,1252,442,408,1286,410,8768,1278,406,1284,438,408,1280,1252,440,1252,442,408,1286,408,1284,408,1286,408,1286,1250,442,408,1286,408,8768,1278,436,1254,438,408,1280,1252,440,1250,444,408,1284,408,1284,408,1284,408,1286,1250,444,410,1284,408,8766,1278,438,1252,436,410,1280,1252,440,1252,442,408,1284,408,1284,408,1286,408,1286,1250,442,410,1284,408,8770,1278,438,1252,436,410],'type':'raw','khz':38}]`

The serial output looks like this (snipped the header stuff):
`New client request: GET /json?plain=[{'data':[1278,438,1252,436,410,1280,1250,442,1276,418,410,1284,408,1284,408,1286,408,1284,1252,440,410,1284,408,8768,1278,436,1254,438,410,1280,1250,440,1252,414,436,1284,408,1286,408,1286,408,1284,1252,440,410,1284,408,8768,1278,438,1252,436,412,1280,1250,440,1252,440,412,1284,408,1286,408,1284,410,1284,1252,442,408,1286,410,8768,1278,406,1284,438,408,1280,1252,440,1252,442,408,1286,408,1284,408,1286,408,1286,1250,442,408,1286,408,8768,1278,436,1254,438,408,1280,1252,440,1250,444,408,1284,408,1284,408,1284,408,1286,1250,444,410,1284,408,8766,1278,438,1252,436,410,1280,1252,440,1252,442,408,1284,408,1284,408,1286,408,1286,1250,442,410,1284,408,8770,1278,438,1252,436,410],'type':'raw','khz':38}] HTTP/1.1`
`method: GET url: /json search: plain=[{'data':[1278,438,1252,436,410,1280,1250,442,1276,418,410,1284,408,1284,408,1286,408,1284,1252,440,410,1284,408,8768,1278,436,1254,438,410,1280,1250,440,1252,414,436,1284,408,1286,408,1286,408,1284,1252,440,410,1284,408,8768,1278,438,1252,436,412,1280,1250,440,1252,440,412,1284,408,1286,408,1284,410,1284,1252,442,408,1286,410,8768,1278,406,1284,438,408,1280,1252,440,1252,442,408,1286,408,1284,408,1286,408,1286,1250,442,408,1286,408,8768,1278,436,1254,438,408,1280,1252,440,1250,444,408,1284,408,1284,408,1284,408,1286,1250,444,410,1284,408,8766,1278,438,1252,436,410,1280,1252,440,1252,442,408,1284,408,1284,408,1286,408,1286,1250,442,410,1284,408,8770,1278,438,1252,436,410],'type':'raw','khz':38}]
`
<...>
`args: plain=[{'data':[1278,438,1252,436,410,1280,1250,442,1276,418,410,1284,408,1284,408,1286,408,1284,1252,440,410,1284,408,8768,1278,436,1254,438,410,1280,1250,440,1252,414,436,1284,408,1286,408,1286,408,1284,1252,440,410,1284,408,8768,1278,438,1252,436,412,1280,1250,440,1252,440,412,1284,408,1286,408,1284,410,1284,1252,442,408,1286,410,8768,1278,406,1284,438,408,1280,1252,440,1252,442,408,1286,408,1284,408,1286,408,1286,1250,442,408,1286,408,8768,1278,436,1254,438,408,1280,1252,440,1250,444,408,1284,408,1284,408,1284,408,1286,1250,444,410,1284,408,8766,1278,438,1252,436,410,1280,1252,440,1252,442,408,1284,408,1284,408,1286,408,1286,1250,442,410,1284,408,8770,1278,438,1252,436,410],'type':'raw','khz':38}]
args count: 1
Request: /json`

`final list of key/value pairs:
  key:'plain' value:'[{'data':[1278,438,1252,436,410,1280,1250,442,1276,418,410,1284,408,1284,408,1286,408,1284,1252,440,410,1284,408,8768,1278,436,1254,438,410,1280,1250,440,1252,414,436,1284,408,1286,408,1286,408,1284,1252,440,410,1284,408,8768,1278,438,1252,436,412,1280,1250,440,1252,440,412,1284,408,1286,408,1284,410,1284,1252,442,408,1286,410,8768,1278,406,1284,438,408,1280,1252,440,1252,442,408,1286,408,1284,408,1286,408,1286,1250,442,408,1286,408,8768,1278,436,1254,438,408,1280,1252,440,1250,444,408,1284,408,1284,408,1284,408,1286,1250,444,410,1284,408,8766,1278,438,1252,436,410,1280,1252,440,1252,442,408,1284,408,1284,408,1286,408,1286,1250,442,410,1284,408,8770,1278,438,1252,436,410],'type':'raw','khz':38}]'`
`Connection received - JSON`
`JSON parsing failed
`
Any thoughts? The libraries are up-to-date and they compiled fine with just some nonfatal warnings about deprecated declarations. I'm using the /dev branch as I had the issue on my WEMOS D1 Mini where the web server was freezing after about 5 seconds. I also noticed that Chrome likes to be helpful and encode some of the characters in the URL, where Edge does not. So I've been using Edge to keep things simple.

Thanks for your help.
-James

mdhiggins, 2020-04-24:
Try removing the single quotes from around the keys (data, type, khz) and see if that helps
I think that was a change enforced by the newer versions of the json library

jaymzx, 2020-04-24:
Hey, alright. That got me somewhere. 
I was running the string through a json validator and what got it to work was to remove the single quotes on the keys, as you suggested, _and_ placing double quotes around the string after the _data_ key.
e.g.; 
`http://192.168.1.50/json?plain=[{data:"[1278,438,<...>1252,436,410],type:raw,khz:38"}]`

Thanks for the tip!

closed 2020-04-24 by mdhiggins

jaymzx, 2020-04-24:
Actually, I spoke too soon. The JSON parsing now works in the UI, however. the LED isn't transmitting. The new serial output shows the transmitting string is null, so it appears to me that the new JSON string is breaking somewhere else:

```
Connection received - JSON
Blasting off
Blocking incoming IR signals
0:null:0
Transmission complete
Reenabling receiving
Sending home page
Turning off the LED to save power
```
The LED transmits fine if I use one of the built-in codes, e.g.; `msg?code=A90:SONY:12`
Would the simplest solution for the time being be to revert to an earlier ArduinoJSON library? If so, which one do you suggest?

mdhiggins, 2020-04-24:
Remove the double quotes from the array

jaymzx, 2020-04-24:
That's the rub - it returns the parsing error without the double quotes around the array (even with no single quotes around the keys, as suggested). I tried replacing the doubles with single quotes as well and while it doesn't return the parsing error, it also returns null.

mdhiggins, 2020-04-24:
Does it work if you put the individual numbers in quotes?

jaymzx, 2020-04-24:
Negative.
This passes as 'successful' in the Web UI, but returns null. Note the double quotes after the data key
`192.168.1.50/json?plain=[{data:"['1278','438','1252',<...>'410'],type:raw,khz:38"}]`

These fail json validation, as do other various permutations of quotes around the keys and values:
`192.168.1.50/json?plain=[{data:"["1278","438","1252",<...>"410"],type:raw,khz:38"}]`
`192.168.1.50/json?plain=[{data:['1278','438','1252',<...>,'410'],type:raw,khz:38}]`
`192.168.1.50/json?plain=[{data:'['1278','438','1252',<...>'410'],type:raw,khz:38'}]`

It seems that the URL jason validation is contingent upon placing the 'data' values in double quotes.

mdhiggins, 2020-04-25:
Alright I was able to get it to work using this format

```
http://10.0.1.46:667/json?plain=[{data:[1278,438,1252,436,410,1280,1250,442,1276,418,410,1284,408,1284,408,1286,408,1284,1252,440,410,1284,408,8768,1278,436,1254,438,410,1280,1250,440,1252,414,436,1284,408,1286,408,1286,408,1284,1252,440,410,1284,408,8768,1278,438,1252,436,412,1280,1250,440,1252,440,412,1284,408,1286,408,1284,410,1284,1252,442,408,1286,410,8768,1278,406,1284,438,408,1280,1252,440,1252,442,408,1286,408,1284,408,1286,408,1286,1250,442,408,1286,408,8768,1278,436,1254,438,408,1280,1252,440,1250,444,408,1284,408,1284,408,1284,408,1286,1250,444,410,1284,408,8766,1278,438,1252,436,410,1280,1252,440,1252,442,408,1284,408,1284,408,1286,408,1286,1250,442,410,1284,408,8770,1278,438,1252,436,410],type:'raw',khz:38}]
```

But the issue is that the JSON parsing error was due to lack of memory. I just pushed an update that expands the allocated memory to accommodate raw commands of this size. I also improved the error logging to include why the json parsing error was flagged since that can be misleading

jaymzx, 2020-04-25:
You nailed it. Works like a champ now. Thanks for your help on this.