mdhiggins/ESP8266-HTTP-IR-Blaster #280: Web UI Still Failing to Load

PicoMitchell, 2020-04-25:
I'm using the very latest Dev versions from today mentioned in https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/277 and I'm still not able to load the Web UI.

I also modified my platformio.ini file to use espressif8266 platform version 2.4. I noticed issue https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/279 because the Dev version of WifiManager would not build when using espressif8266 version 1.8.

The most I get is the following HTML snippet and then nothing else seems to ever load
`<html xmlns='http://www.w3.org/1999/xhtml' xml:lang='en'>
  <head>
    <meta name='viewport' content='width=device-width, initial-scale=.75' />
    <link rel='stylesheet' href='https://stackpath.bootstrapcdn.com/bootstrap/3.4.1/css/bootstrap.min.css' />
    <style>@media (max-width: 991px) {.nav-pills>li {float: none; margin-left: 0; margin-top: 5px; text-align: center;}}</style>
    <title>Living Room - ESP8266 IR Controller (Living-Room-Blaster)</title>
  </head>
  <body>
    <div class='container'>
      <h1><a href='https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster'>Living Room - ESP8266 IR Controller</a></h1>
      <div class='row'>
        <div class='col-md-12'>
          <ul class='nav nav-pills'>
            <li class='active'>
              <a href='http://Living-Room-Blaster.local:4223'>Hostname <span class='badge'>Living-Room-Blaster.local:4223</span></a></li>
            <li class='active'>
              <a href='http://192.168.0.223:4223'>Local <span class='badge'>192.168.0.223:4223</span></a></li>`

mdhiggins, 2020-04-25:
Platform IO should be fixed now please try

PicoMitchell, 2020-04-25:
Awesome! The Web UI and everything appears to be working properly and quickly with the latest code from your Dev branch :-)

closed 2020-04-25 by PicoMitchell