mdhiggins/ESP8266-HTTP-IR-Blaster #281: Erreur lors de la récupération de l'IP externe #

slimlesaurien, 2020-04-26:
Salut, seriez-vous en mesure de nous aider avec ce qui suit?

Erreur lors de la récupération du
code HTTP IP externe : -1
connexion refusée
La demande d'adresse IP externe a pris 5002 ms
Aucune erreur détectée avec la configuration de sécurité L'
époque de timeServer est inattendue, probablement la connexion au serveur de temps a échoué. Vérifiez vos paramètres réseau
14400
Heure EPOCH non valide, les vérifications de sécurité peuvent échouer si vous ne pouvez pas vous synchroniser avec le serveur NTP

mdhiggins, 2020-04-27:
Don't speak French so using translator software

Make sure your network settings are correct (feel free to post your configuration)
Make sure if you're using the main branch that you're using lwIP v1.4 and ESP8266 Core 2.4.0

You can reference the platformio config to see what versions are recommended
```
[env:nodemcuv2]
platform = espressif8266@~2.4.0
board = nodemcuv2
framework = arduino
build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH # lwIP v1.4
board_build.ldscript = eagle.flash.4m2m.ld
lib_deps =
  IRremoteESP8266@~2.7.5
  WifiManager@~0.15
  ArduinoJson@~6.13.0
  Time@~1.5
  https://github.com/jjssoftware/Cryptosuite.git
monitor_speed = 115200
```

slimlesaurien, 2020-04-30:
Hello I work under Arduino IDE 1.6.5
i wanted to know if server name i but love or i leave like that

mdhiggins, 2020-04-30:
IDE should be fine (though it wouldn't hurt to update it). What matters is your library versions as above
Server name can be whatever you want

slimlesaurien, 2020-04-30:
and for ports and IP I have trouble.
I start on arduino.

slimlesaurien, 2020-04-30:
alexa answer me:
sorry the TV device is not responding

slimlesaurien, 2020-04-30:
already first point:
"check-up.dyndns.org"
I have to replace it or not in const char serverName

slimlesaurien, 2020-04-30:
can you tell me what i should modify on the program thank you

mdhiggins, 2020-05-01:
You should not need to modify the code at all. You need to post all the above information in full or I can't help you

- Serial logging data from the ESP8266 from boot to request
- Information about your local network settings
- Information about you tehpsyc.pythonanywhere.com settings (feel free to censor your external IP)
- Confirmation that you've set up port forwarding and are using your external IP
- What configuration you're putting into WiFiMananger

slimlesaurien, 2020-05-01:
[image]

mdhiggins, 2020-05-01:
That all looks great. Did you set up port forwarding on your router?

slimlesaurien, 2020-05-01:
no i tried but it didn't work can you help me with that

slimlesaurien, 2020-05-01:
I don't have a router
just fiber srf box

TheMCMan, 2020-05-01:
At my girlfriend's I had to login to her internet provider to setup port
forwarding.

The internet provider had port forwarding as a managed service.  It was
Comcast XFi service.

Then I had to turn off advanced monitoring to allow the port forwarding to
actually work...

slimlesaurien, 2020-05-01:
can you make an example of port forwarding with IP that would help me
thank you in advance

mdhiggins, 2020-05-01:
Port forwarding is a basic concept in networking not specific to this device/software

https://stevessmarthomeguide.com/understanding-port-forwarding/

You may be better served just googling a guide in your native language but that one may be a helpful starting point. Usually it is through a router but it may depend on how you connect to the internet

mdhiggins, 2020-05-01:
Going to close this since it's not a technical issue but just a request for help, feel free to keep discussion going

closed 2020-05-01 by mdhiggins

slimlesaurien, 2020-05-02:
it's good i succeeded 👍
however need help with channel numbers
i don't use roku TV
I saw that there was define chain and wanted to know how to proceed Thank you in advance

mdhiggins, 2020-05-02:
That just maps names to specific channel numbers

Like HBO -> 14

You still need to code in the individual number buttons in the channel controller (0 through 9) so it knows how to send the individual numbers

slimlesaurien, 2020-05-02:
I have to add what and where thank you in advance

slimlesaurien, 2020-05-02:
[image]

slimlesaurien, 2020-05-02:
[image]

mdhiggins, 2020-05-03:
That looks fine

mdhiggins, 2020-05-03:
You might need to spell out the number though if Alexa struggles with it

slimlesaurien, 2020-05-03:
how to specify the number seen Alexa tell me no skills active when I say TF1
or else when I say 1 Alexa tell me she can't find the channel

slimlesaurien, 2020-05-03:
[image]

mdhiggins, 2020-05-03:
Are you using your echo in French?
https://developer.amazon.com/en-US/docs/alexa/device-apis/list-of-interfaces.html

the ChannelController API isn't support in French

slimlesaurien, 2020-05-03:
hello everyone I have 2 questions
in 1 I have to put what the local address or not
 in 2 sound the code of the key 1 its such well filled for you (key of my TV
Thanks in advance 😉

mdhiggins, 2020-05-03:
1 needs to be your external IP address but formatted exactly how the placeholder text is

slimlesaurien, 2020-05-03:
😭😭😭😭 yes I use echo in France
is there a solution

slimlesaurien, 2020-05-03:
the link you sent me I don't understand anything can you help me

mdhiggins, 2020-05-03:
There is no solution for ChannelController until Alexa adds support for your language

slimlesaurien, 2020-05-04:
a pity for me that there is no French support
Thank you for your help
excellent project 😉.