mdhiggins/ESP8266-HTTP-IR-Blaster #286: wifi not showing up after turning it on later

riskyrun, 2020-06-06:
Initially when I completed the process ,I was able to configure it via the local ip.But when I tried to turn the device on after a while by connecting it to a power source,It is not generating a wifi signal so that I can finish the rest of configuration.
Also,when I press reset button on the chip,It shows all the credentials that I entered to setup the device previously(Like the name gave,etc which I linked below)
[image]

It also recieves the IR signals I give it and shows it in the serial monitor

mdhiggins, 2020-06-06:
You need to set your config pin to ground to initiate the configuration process during boot.  GPIO10 is your config pin. Connect that to ground and leave it that when while rebooting the device

riskyrun, 2020-06-06:
I tried the above method.While the chip seems to be entering the config mode,it still shows the same problem.After configuration, the "IR BLASTER CONFIGURATION" disappears and doesn't appear again until I have to do what you told above about entering config mode during boot and everything has to be done from fi
[image]
rst again

kirkpensalvo, 2020-06-06:
@riskyrun I just setup today and am seeing the same behavior. After initial Wifi configuration, web server works successfully, can send/receive IR blasts. However, next reboot and though reporting as you have screenshot above, web server does not respond.

riskyrun, 2020-06-07:
@kirkpensalvo .Any luck till now?

kirkpensalvo, 2020-06-07:
@riskyrun I went back to the 2.4 ESP8266 board version (which is called out in the readme) and 0.15 version of wifimanager (other comments suggested using the development branch, but it has issues). Anyway, is working now through resets and power cycles.

mdhiggins, 2020-06-07:
Yeah this is almost certainly the very well documented version issue. You either need to be on 2.4 with lwIP 1.4 or you need to use the development branch of WiFiManager and then you're free to use the latest versions. Its a bug and known issue with WiFiManager and there's no work around. 

I will say I've had all my personall devices on the development branch for about 3-4 months with continuous uptime and no issues.

mdhiggins, 2020-06-07:
If you search for 'lwIP' in the issues you'll see there's >20 issues of people with the same problem that didn't use the instructed versions.

PlatformIO is a good development environment that handles all the version control for people that don't want to micro manage that stuff

riskyrun, 2020-06-07:
Sigh.No luck with downgrading the versions.Tried my hand at platformio but I don't think I did a good job there.Same issue persists.

kirkpensalvo, 2020-06-07:
@mdhiggins Yup, saw those postings and downgraded. Are you using 2.03-alpha of Wifimanager? Won't build for me...

closed 2020-06-17 by riskyrun