mdhiggins/ESP8266-HTTP-IR-Blaster #291: JSON Parsing Failed No Memory - AC IR Raw code

digital-phantom, 2020-07-03:
Hi! Excellent work with this. Just running into an issue transmitting raw IR code from my Air conditioning remote.

I'm getting a "Error! JSON parsing failed" message when trying to send the following

http://192.168.4.1:80/json?plain=[{data:[432,440,432,438,432,438,430,438,432,436,432,25420,3476,1738,434,1306,432,438,432,442,432,430,430,1306,434,436,432,438,430,438,432,436,432,1308,432,438,432,1304,432,1306,432,440,432,1308,432,1302,434,1306,432,1304,434,1306,432,436,432,438,432,1306,434,436,430,438,432,440,432,436,432,436,432,438,432,436,432,438,430,436,432,440,430,436,432,436,432,438,432,438,432,438,432,434,432,436,432,436,432,1304,432,438,432,440,432,434,432,1308,434,1306,432,434,432,436,432,438,432,438,432,1304,432,438,432,1306,434,1304,434,438,432,436,432,436,432,438,432,436,432,436,432,438,430,436,432,436,432,438,432,1306,434,1304,434,1306,432,1306,434,436,432,1304,434,442,434,1302,432,438,432,436,432,438,432,436,432,438,432,438,432,436,432,436,432,438,432,434,432,446,432,430,432,436,432,436,432,436,432,438,432,436,432,436,432,436,432,438,432,438,432,436,432,434,432,440,430,438,432,444,432,430,432,438,432,438,432,436,432,436,432,438,432,1304,434,440,432,434,432,438,432,438,430,438,432,436,432,436,432,438,432,436,432,436,432,440,430,438,432,438,432,436,432,436,432,1306,434,438,432,1306,432,436,430,438,432,438,432,1304,434,1306,432,438,432,438,432,438,430,436,432,438,432,434,432,438,432,436,432,438,432,436,432,436,434,1304,434,436,430,440,432],type:'raw',khz:38}]

It seems this is quite a bit longer than regular IR code because of how AC remotes are.

Any chance this could work?

Thanks!

mdhiggins, 2020-07-03:
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/278

This may provide some insight

There was an update 90ca42b6bf8a24e80818de375285f7023d21fa7d that increased the buffer size and added some more error logging as to why the JSON parsing failed and your error message seems to not include that

Updating should at least improve the error message, and may fix by increasing the buffer

If that doesn't work try going up on the buffer more

digital-phantom, 2020-07-03:
Great! Thanks for the reply. I increased the buffer and it seems to parse fine now. The AC doesn't respond to the raw code so I'm not 100% sure my IR led is working. At any rate, this parsing issue is resolved for me now. Thanks again.

closed 2020-07-03 by digital-phantom