mdhiggins/ESP8266-HTTP-IR-Blaster #292: Skills

slimlesaurien, 2020-07-07:
Hello I would like to know if you can put the skills in French for Channel control
thank you in advance

mdhiggins, 2020-08-18:
[translate.pdf](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/5090677/translate.pdf)

I need translation of all these elements and I can try to submit to Amazon for approval

mdhiggins, 2020-08-18:
> IR Controller Smart Home Skill that allows registration of your TV / other infrared smart home controller devices with Alexa. Your http URLs are translated into natural language smart home operations.
> 
> Register your IR Controller smart devices at https://tehpsyc.pythonanywhere.com/
> 
> Skill is designed to work with the https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster schematic and software. It may be possible to use similar HTTP enabled devices as well. The IR Blaster can be easily assembled with parts under $10 and provides full home theater control.

For the part you can't fully read in the PDF

alka79, 2020-10-22:
Hi,
is this issue still open ?
if yes, I can provide french translation. Let me know the best way to send it to you.
Al

mdhiggins, 2020-10-22:
Yes I still need help with the translation. You can see the PDF from earlier. I got the other languages approved but didn't have anyone to do a native French translation.

alka79, 2020-10-22:
ok. I am on it.

alka79, 2020-11-11:
Hi, I sent a while ago the translation to you by email. Just wondering if you could submit the FR langage skill to Amazon ?

mdhiggins, 2020-11-13:
The language submission process is frustrating and has many hoops to jump through. I have to have a test device active and functional for amazon to use for every language included in the skill for a single language to be approved. Working on it but busy with work. I'll try to get to it soon.

closed 2022-01-25 by mdhiggins