mdhiggins/ESP8266-HTTP-IR-Blaster #294: Using Sparkfun WiFi IR Blaster WRL-15031

jacobhweeks, 2020-08-14:
After re-configuring the pin declarations and entering my network address info, the device seems to load everything accurately, but the IR receiver seems to receive superfluous data. Random data seems to be received even when taken to an "IR quiet" area. Additionally, the green LED isn't working either. But I confirmed it to work with the blinky example. Can someone please help with this?

mdhiggins, 2020-08-15:
Might be a hardware problem or a cheap receiver, or a receiver pin that has other uses. Might want to try a different pin and see if it persists. Ambient light and sunlight can occasionally trigger receives (sunlight more commonly so)

This would be something to pass on to the IRRemote8266 github though as this script just uses that library it doesn't handle any receiving directly on its own

closed 2020-08-15 by mdhiggins

jacobhweeks, 2020-08-15:
I also thought it was a hardware problem, so I replaced it... and the Beefy 3 FTDI that I used. I suppose that it's possible that it could be a pin issue since these boards are manufactured, but they are manufactured specifically for this purpose, so swapping pins will be significantly harder. I also thought that it could have been some of the code in the IRRemote8266 library, but then I would suspect that it would have caused problems when I tested the receive demo. But it works fine, so did the transmitter.
I've been trying this for some time now, and I can't figure out what could be causing this. It's been a headache. Trust me, I wouldn't have bothered you with this unless I have exhausted all other possibilities.
I have even considered re-writing all of this myself, but coding is time-consuming and I only wanted to do this if I couldn't get this to work at all.
I thought that it could be that the receiver is receiving what is being transmitted, but this is also not the case since it receives IR data when not transmitting. Another thing to note is that the type of code is not detected, where the receive demo did detect the type of signal (Sony, Samsung, etc.).

TheMCMan, 2020-08-15:
I experienced such erratic behavior when connecting to the node32 using an Android based browser on both my phone and a tablet.  Switched over to my Win10 laptop to browse the Node32 website for reliable results.  Hope this helps.

jacobhweeks, 2020-08-15:
I wish this was the case, but my problem is deeper than that. Thanks anyway, it's good to know.

mdhiggins, 2020-08-15:
Post a pic of your build

Random noise tends to not be in any readable format so you get an 'unknown' code so that part isn't surprising. Browser should have absolutely nothing to do with it, the code reading happens on a layer below the browser level functions and can be viewed from the serial output as well but its all independent of the browser and can be verified on the serial output

I will say I've never tried this on the sparkfun board so maybe worthwhile to try one of the more common nodemcu boards to rule out a hardware issue

Additionally the irremoteesp8266 library has some sample code projects that are very bare bones that you could upload to test the receiver that don't have all the other functionality of this project that you could use to try and debug with but honestly from a receiver side of things this blueprint doesn't add on any significant changes. 

Post some serial logs of the data you're receiving if you want more help but without any logs or pics of the build I won't be able to help

mdhiggins, 2020-08-15:
> Hey MDHIGGINS,
> 
> I don't know what happened to my problem report from before, but I found a possible solution. This is the one about Sparkfun WiFi IR Blaster WRL-15031
> In the IR rec demo, I found that I needed to adjust the timeout of UNKNOWN signals. I set it to basically ignore unknown signals (1000ms) and that seems to have solved the problem with receiving. I don't remember having that problem with this board, the first board I tried I did not.
> 
> The LED problem persists. This board used GPIO 5.

mdhiggins, 2020-08-15:
A code sample of what you changed is always helpful for others who have the same board/issue