mdhiggins/ESP8266-HTTP-IR-Blaster #297: Is the python code available anywhere?

giuliomoro, 2020-09-02:
Is the Python code that is running on https://tehpsyc.pythonanywhere.com available anywhere? I'd be curious to look at it and if possible contribute some of the missing Alexa Interfaces.

mdhiggins, 2020-09-02:
The server side code is not available, mostly just because I haven't had time to audit the code and I'm not actively developing the project at this time

TheMCMan, 2020-09-02:
I recommend using IFTTT if you want to customize your service beyond what is provided in the server side code.

I use IFTTT with this project rather than the python code.

giuliomoro, 2020-09-02:
thanks @TheMCMan , any code you'd be willing to share? I am not familiar with IFTTT, so any hint would be greatly appreciated.

mdhiggins, 2020-09-03:
IFTTT doesn't require any code you just share the URLs and link your account

You can accomplish basically everything IFTTT offers using the scene controller though

closed 2020-09-03 by mdhiggins